scikit-rf/scikit-rf #613: Suggestion to Add higher number of Ports example to se2gmm documentation

denzchoe, 2022-03-01:
At present, [se2gmm ](https://github.com/scikit-rf/scikit-rf/blob/3bb6cd24cad0d5d1ab291812c64707f8432ea391/skrf/network.py#L3407) example only shows 3 ports
https://github.com/scikit-rf/scikit-rf/blob/3bb6cd24cad0d5d1ab291812c64707f8432ea391/skrf/network.py#L3433-L3447

As cited over [here](https://github.com/scikit-rf/scikit-rf/issues/607#issuecomment-1051105613) 
and a frequent user who has to deal with high port counts (8 and more), ASLOBEAR is right. 
Need more example in the documentation to reflect the se2gmm return output.

right now, if SE network of such;
`8-Port Network: 'Diff_Pair_Ntwk13245768',  0.0-5.0 GHz, 501 pts, z0=[50.+0.j 50.+0.j 50.+0.j 50.+0.j 50.+0.j 50.+0.j 50.+0.j 50.+0.j]`

converts to MM network, the output Differential Modes (DM) are at the first 4 ports; and the Common Modes (CM) are the last 4 Ports. This isn't reflected in the documentation. ;
```
Diff_Pair_Ntwk13245768.se2gmm(p=4)
8-Port Network: 'Diff_Pair_Ntwk13245768',  0.0-5.0 GHz, 501 pts, z0=[100.+0.j 100.+0.j 100.+0.j 100.+0.j  25.+0.j  25.+0.j  25.+0.j  25.+0.j]
```
<br>
For beginners, this could be a tricky issue to grasp. So when plotting, the expected outcome may not be immediately understood. 
What do you guys think?

<br>
<br>

jhillairet, 2022-03-01:
Yes, an additional example could be useful to make it more clear.

mhuser, 2022-06-11:
Something like this?
```
numbering diagram::

      single-ended 2N-Port
           +-----------+ 
          -|s0       sN|-
          -|s2     sN+1|-
          ...         ...
          -|sN-2  s2N-2|-
          -|sN-1  s2N-1|-
           +-----------+ 

      mixed-mode 2N-Port
           +-----------+ 
         0-|d0       c0|-N
         1-|d1       c1|-N+1
          ...         ...
       N-2-|dN-2   cN-2|-2N-2
       N-1-|dN-1   cN-1|-2N-1
           +-----------+ 

    Where sx is a single-ended port, dx is a differential port, and cx is a common port.
```
and this?
```
numbering diagram::

      single-ended 5-Port
           +-----------+ 
          -|s0       s2|-
          -|s1       s3|-
           |         s4|-
           +-----------+ 

      mixed-mode 5-Port
           +-----------+ 
         0-|d0       c0|-2
         1-|d1       c1|-3
           |         s0|-4
           +-----------+ 

    Where sx is a single-ended port, dx is a differential port, and cx is a common port.
```
I am not 100% sure that this is correct 😅

denzchoe, 2022-06-13:
Hi @mhuser , 

<br>

This looks acceptable, I think. Suggestion to improve on the wording wrt to [IEEE 802.3 standards terminology](https://ieeexplore.ieee.org/document/8457469)
Where;

> Where sx is a single-ended port, dx is a differential port, and cx is a common port.

can be rephrased as :
`Where sx is a single-mode port, dx is a differential-mode port, and cx is a common-mode port.`

<br>

I think the single-ended part isn't correct.
It should be the below. But I need someone to verify..

```
           +-----------+ 
          -|s0      sN-3|-
          -|s2      sN-1|-
          ...         ...
          -|s2N-4  s2N-3|-
          -|s2N-2  s2N-1|-
           +-----------+ 
```

mhuser, 2022-06-13:
> Where sx is a single-mode port, dx is a differential-mode port, and cx is a common-mode port.

This make sense !
For the numbering I am a bit lost (I updated above replacing s2 that was a typo by s1). Is it that single-ended s0 + s1 make common-mode c0 and differential-mode d0?

mhuser, 2022-06-13:
I used purposely for single-ended the `skrf` numbering scheme that enable cascading 2N-ports:
https://github.com/scikit-rf/scikit-rf/blob/0cd53a0e598c2dfb802e1eeccf584b91833cd14e/skrf/network.py#L4311-L4318

mhuser, 2022-06-13:
This is usefull, e.g. to write 2N-port deembedding.
However after transforming to mixed-modes, direct cascading is no more possible using `**` operator unless some renumbering is done first.

mhuser, 2022-06-13:
Looks good!
Would it not be more simple with a continuous numbering on single-ended side 1 to N-1 (like one of IEEEP370 scheme 😄) if SIG_B1_T1 would be on Port[3] and SIG_B2_T1 on Port[4]?

Stacking two 4-port is of course also a valid point of view. I think it is up to the user at the end. But I like very much to write `network1 ** network2` with 2N-Ports.

mhuser, 2022-06-13:
if we start some ascii art in docstring with below and end it with a blank line, it will display nicely on RTD:
https://github.com/scikit-rf/scikit-rf/blob/0cd53a0e598c2dfb802e1eeccf584b91833cd14e/skrf/network.py#L4307-L4320
[image]

mhuser, 2022-06-13:
I do not know if it is possible to add a picture. But even so, the picture would not be visible in the source code docstring. Maybe it would be better placed into some notebook.

mhuser, 2022-06-14:
[First attempt](https://scikit-rf--707.org.readthedocs.build/en/707/api/generated/skrf.network.Network.se2gmm.html#skrf.network.Network.se2gmm)
Is 2N-Port as terrible as that?

mhuser, 2022-06-16:
- [se2gmm](https://scikit-rf--707.org.readthedocs.build/en/707/api/generated/skrf.network.Network.se2gmm.html#skrf.network.Network.se2gmm)
- [gmm2se](https://scikit-rf--707.org.readthedocs.build/en/707/api/generated/skrf.network.Network.gmm2se.html#skrf.network.Network.gmm2se)

mhuser, 2022-06-18:
@denzchoe do you see more examples to add, or shall we merge these ?

denzchoe, 2022-06-19:
Looks good now! Thank you for the ascii art. Perhaps @jhillairet should have a final say?

jhillairet, 2022-06-19:
very clear I think, it's nice. Thank you!

PR #707 (2022-06-19): Add higher number of ports examples to se2gmm
Add 2, 3, 4, 5, 8, and 2N-Port examples to se2gmm docstring.
Attempt to fix #613
closed 2022-06-19 by jhillairet