scikit-rf/scikit-rf #615: Support for old Python releases

Vinc0110, 2022-03-09:
The proposed changes in PR #614 require the latest major release of NumPy (>= v1.22.0), in which they dropped support for Python < 3.8. 
Scikit-rf is currently still supporting Python 3.6 and 3.7, but that would stop if we decide to depend on NumPy v1.22.

Python 3.6 is already at its end-of-life, but Python 3.7 is still a thing:

https://devguide.python.org/#status-of-python-branches

[image]

For PR #614, I can probably find a workaround and live with the oldish numpy, but is it really necessary?

What is the general opinion about keeping the support of Python 3.6 and 3.7? Does anyone care?

keikawa, 2022-03-10:
How about including a question about the version of Python in the next survey?, but It's still March now...

jhillairet, 2022-03-10:
Two years ago, in the 2020's survey, around 70% of the user were using Python 3.7. Since I didn't ask for the python version in the 2021's version of the survey, I can't compare...

However, as a lot (most?) of the users are occasional Python users, I'm not sure if they are using the last version of Python. In my work, for example, the default Python version, (which was upgraded last year), is still 3.7 (and I guess it will remain like this for a certain time...).

JAnderson419, 2022-03-10:
I definitely have some virtual environments on 3.6/3.7 with old versions of skrf that are "good enough" for opening and looking at s-parameters, maybe doing a few calculations, and some plotting. That being said, I think there is some advantage to not requiring support for 5 year old versions of Python, especially in the scientific domain when trying to provide a good (or good enough) alternative to commercial software that often has yearly release cycles.

Instead of supporting every Python version that isn't EOL, an intermediate approach would be to follow [NEP 29](https://numpy.org/neps/nep-0029-deprecation_policy.html) since it seems to be a common one among many of the scientific libraries (including skrf dependencies). This requires the greater of 42 month or two minor version support for Python and 24 month or 3 minor version support for numpy. Currently, this would mean Python 3.8+ and numpy 1.19+ (we are requiring 1.19.5 currently), with that changing to numpy 1.20 in June. 

Adopting NEP 29 still wouldn't help the VF improvements that rely on numpy 1.22. That could be solved by creating nightly/rc branches for future releases that advanced users could install, but that might be more work than it is worth unless there are a large number of corporate users who require support of old versions for a long time...

FranzForstmayr, 2022-03-10:
To be honest I would like to use scikit-rf on older python versions too, although I love new features like the improved typing system. However, I commented #614 to use an alternative implementation of `np.linalg.qr` which would support even python3.6.

Regarding the original question, I think we could drop the support of 3.6, as this version is out of maintenaince right now. 3.7 implements an improved typing support for example. However, if we do not make use of this new features, I don't see any point why we should drop the support of 3.6

Vinc0110, 2022-03-12:
> unless there are a large number of corporate users who require support of old versions for a long time...

It seems that is the case, so let's keep supporting the old versions of Python. In case of my PR #614, this is actually not a big deal. Numpy 1.22.0 only made things slightly more convenient.

closed 2022-03-12 by Vinc0110