scikit-rf/scikit-rf #617: CuPy support

nikosavola, 2022-03-15:
I wonder if https://cupy.dev/ could be used as a drop-in replacement to accelerate some of the matrix operations in skrf using a GPU.
For example, generating big networks at different frequencies makes many calls to  `connect_s` and `z2s`, which seem to take some time. CuPy appears to support most of the NumPy routines, such as, [cupy.linalg.inv](https://docs.cupy.dev/en/stable/reference/generated/cupy.linalg.inv.html)

FranzForstmayr, 2022-03-15:
Most `numpy` functions work fine with cupy arrays, so if you call `np.something` on a `cupy` array, there's a good chance that the function will be executed by the appropriate `cupy` function.

However, using your example `z2s` will have problems when creating new arrays.
Calling `np.zeros_like` will create a `numpy` array, you'd have to call `cp.zeros_like` instead.

I think it would be a great addition, but maybe will need some time. There's a lot going on, and there's a big chance, that the coexistence of `numpy` and `cupy` arrays will get easier, without manually copying data from/to cpu/gpu.

arsenovic, 2022-03-15:
it would be cool to do some benchmarks for a prototype solution.

jhillairet, 2022-03-15:
@nikosavola if you're connecting many big networks together, maybe  `Circuit` is faster, in particular since the latest version of scikit-rf. Did you try?

nikosavola, 2022-07-07:
Hello again! I experimented with using CuPy in some of the inner functions of `skrf.network` (z2s, s2z, connect_s) but at least for my case this was indeed a bit slower.

#### Couple of notes:
* Might be faster if everything was on the GPU, as this implementation transfers the matrices between RAM and VRAM when necessary
* CuPy does not currently support `np.linalg.eig`, see https://github.com/cupy/cupy/issues/3255

#### The code
```python
import cupy as xp

from skrf.network import fix_z0_shape, ZERO, S_DEF_DEFAULT

def connect_s(A: xp.ndarray, k: int, B: xp.ndarray, l: int):
    if k > A.shape[-1] - 1 or l > B.shape[-1] - 1:
        raise (ValueError('port indices are out of range'))

    nf = A.shape[0]  # num frequency points
    nA = A.shape[1]  # num ports on A
    nB = B.shape[1]  # num ports on B
    nC = nA + nB  # num ports on C

    # create composite matrix, appending each sub-matrix diagonally
    C = xp.zeros((nf, nC, nC), dtype='complex')
    C[:, :nA, :nA] = xp.array(A) #.copy()
    C[:, nA:, nA:] = xp.array(B) #.copy()

    # call innerconnect_s() on composit matrix C
    return innerconnect_s(C, k, nA + l)

def innerconnect_s(A: xp.ndarray, k: int, l: int):
    if k > A.shape[-1] - 1 or l > A.shape[-1] - 1:
        raise (ValueError('port indices are out of range'))

    nA = A.shape[1]  # num of ports on input s-matrix
    # create an empty s-matrix, to store the result
    C = xp.zeros(shape=A.shape, dtype='complex')

    # loop through ports and calculates resultant s-parameters
    for i in range(nA):
        for j in range(nA):
            C[:, i, j] = \
                A[:, i, j] + \
                (A[:, k, j] * A[:, i, l] * (1 - A[:, l, k]) + \
                A[:, l, j] * A[:, i, k] * (1 - A[:, k, l]) + \
                A[:, k, j] * A[:, l, l] * A[:, i, k] + \
                A[:, l, j] * A[:, k, k] * A[:, i, l]) / \
                ((1 - A[:, k, l]) * (1 - A[:, l, k]) - A[:, k, k] * A[:, l, l])

    # back to CPU
    C = C.get()

    # remove ports that were `connected`
    C = xp.delete(C, (k, l), 1)
    C = xp.delete(C, (k, l), 2)

    return C

def _rsolve(A: xp.ndarray, B: xp.ndarray):
    r"""Solves x @ A = B. Takes Numpy or CuPy array, returns CuPy."""
    # converts to cuda array (if needed) but returns np
    return xp.transpose(xp.linalg.solve(xp.transpose(xp.asarray(A), (0, 2, 1)).conj(),
            xp.transpose(xp.asarray(B), (0, 2, 1)).conj()), (0, 2, 1)).conj()

def _nudge_eig(mat: np.ndarray, cond: float = 1e-9, min_eig: float = 1e-12):
    r"""GPU copy of :func:`skrf.network.nudge_eig`. Takes Numpy or CuPy array, returns CuPy.
    """
    # Eigenvalues and vectors
    # mat_gpu = xp.asarray(mat)
    eigw, eigv = np.linalg.eig(xp.asnumpy(mat))
    eigw, eigv = xp.asarray(eigw), xp.asarray(eigv)
[... 90 lines cut ...]
        USU = xp.linalg.solve(U, s @ U)
        z = xp.linalg.solve(
            _nudge_eig(Id - USU),
            (Id + USU) @ ZR
        )

    elif s_def == 'traveling':
        # Traveling-waves definition. Cf.Wikipedia "Impedance parameters" page.
        # Creating diagonal matrices of shape (nports, nports) for each nfreqs
        sqrtz0 = xp.zeros_like(s)
        xp.einsum('ijj->ij', sqrtz0)[...] = xp.sqrt(z0)
        z = sqrtz0 @ xp.linalg.solve(
            _nudge_eig(Id - s),
            (Id + s) @ sqrtz0
        )
    else:
        raise ValueError('Unknown s_def: {}'.format(s_def))

    return z.get()

# GPU for selected skrf functions
skrf.network.z2s = z2s
skrf.network.s2z = s2z
skrf.network.connect_s = connect_s
```

jhillairet, 2025-08-08:
I'm totally agnostic (actually ignorant) between CuPy or JAX. I don't know if @FranzForstmayr has continued to investigate JAX or not.

FranzForstmayr, 2025-08-08:
I didn't work on that Jax branch for some time, the first results were also not that promising. 
So if cupy performs better, it's probably worth a try.