scikit-rf/scikit-rf #623: Calibration: TwoPortOnePath with S12=S22=0

Vinc0110, 2022-04-06:
Working on the instrument class for the NanoVNA V2 in PR #616, I'm trying to use the `TwoPortOnePath` calibration for an example notebook, but I'm running into two issues:

**Issue 1:**
`TwoPortOnePath` calls `TwelveTerm`, which expects non-zero data on all S-parameters.
In contrast to the [example with the frequency extenders connected to a PNA-X](https://scikit-rf.readthedocs.io/en/latest/examples/metrology/Calibration%20With%20Three%20Receivers.html), the NanoVNA is a true 1.5-port device and does not produce any data for S12 and S22, so I chose to set them 0 in the 2-port `Network`. See the [NanoVNA architecture](https://nanorfe.com/nanovna-v2-user-manual.html#__RefHeading___Toc1867_2953165397).

It seems that once the Twelve Term coefficients have been calculated, only the meaningful forward coefficients get used for `TwoPortOnePath`, but that fails with S12=S22=0 because these Networks cannot be inverted (`numpy.linalg.LinAlgError: Singular matrix`). However, it works in the PNA example, because S12 and S22 contain strong noise and are far from zero.

There is some old code commented out at the beginning of the `TwoPortOnePath.run()` that apparently got replaced with the call of `TwelveTerm.run()`, see below. Would that old implementation maybe work better in this case? It's hard for me to analyze that code, but maybe @arsenovic or @Ttl can tell.

I could also duplicate the data and set S12=S21 and S22=S11 to fill the 2-port, but why? Is that kind of fake TwelveTerm the only option?

https://github.com/scikit-rf/scikit-rf/blob/6316af7eb1172fe62664c766dacc5c5563ccd1c4/skrf/calibration/calibration.py#L1830-L1863

**Issue 2:**
Judging by the example 2-ports of the attenuator in `skrf/doc/source/examples/metrology/three_receiver_cal/data/` with the forward and reverse measurements, the results always need to be stored as S11 and S21. I initially assumed the reverse measurements need to be stored accordingly (as S12 and S22). It would be nice to include these kinds of quirks in the docs. 

Also, why passing both measurements as separate 2-ports when they can be combined so easily? Passing a single (fully populated) 2-port to `apply_cal()` would be consistent with the "real" 2-port implementations, such as SOLT.

Ttl, 2022-04-07:
I'm not too familiar with this calibration. However, looking at the code it looks like the issue is that `TwoPortOnePath` internally uses`TwelveTerm`which calculates the error parameters for both directions. `TwoPortOnePath` then discards the reverse path error terms and replaces them with copies of the forward error parameters. The issue in your case is that with zero S12 `TwelveTerm` calibration fails due to numerical issues when calculating the error parameters that would be discarded later. In the tests it looks like the reverse S-parameters are filled with random values.

jhillairet, 2022-04-08:
@Vinc0110 , does the PR provided by @Ttl in #625 fixes the issue?

PR #625 (2022-04-14): Fix numerical issue in TwoPortOnePath
Fixes #623. Make measured networks symmetrical for `TwelveTerm` to avoid numerical issues.
closed 2022-04-11 by Vinc0110