scikit-rf/scikit-rf #626: How to connect a port to ground?

jrmagers, 2022-04-08:
How do I attach ground to a port on network? I expected the following 4-port circuit to become a 3-port circuit after connecting ground to port 4:
```
import skrf
transformer = skrf.Network(r"TTC1-33W+_Plus25DegC_TB-1111+.s4p")
ground = skrf.Circuit.Ground(transformer.frequency, name='GND')
skrf.connect(transformer, 3, ground, 0)  # connect port 4 to ground
```

But it results in a 4-port network:
```4-Port Network: 'TTC1-33W+_Plus25DegC_TB-1111+',  1.0-4200.0 MHz, 217 pts, z0=[50.+0.j 50.+0.j 50.+0.j 50.+0.j]```

Ground appears to be a 2-port circuit which doesn't make that much sense to me. I must be missing something.

jhillairet, 2022-04-09:
A `Circuit.Ground` is indeed a 2-port Network, as it models an infinite shunt admittance. 

However, `Circuit.Ground` is actually made to be used within the `Circuit` class (and not with `connect`, which is a different connection algorithm), like: 

```python
import skrf
transformer = skrf.Network(r"TTC1-33W+_Plus25DegC_TB-1111+.s4p", name='my4port')
ground = skrf.Circuit.Ground(transformer.frequency, name='GND')
cnx = [
    [(transformer, 3), (ground, 0)],
]
cir = skrf.Circuit(cnx)
ntw = cir.network  # a 3-port Network as you expect
```

jrmagers, 2022-04-09:
```
import skrf

transformer = skrf.Network(r"TTC1-33W+_Plus25DegC_TB-1111+.s4p",name='my4port')
freq = transformer.frequency
ground = skrf.Circuit.Ground(freq, name='ground')
cnx = [[(transformer, 3),(ground, 0)]]
skrf.Circuit(cnx).network
```

This seems to return a 1-port circuit. Also, I noticed that if I name the transformer object "transformer", I get an error.
[TTC1-33W+_Plus25DegC_TB-1111+.zip](https://github.com/scikit-rf/scikit-rf/files/8457831/TTC1-33W%2B_Plus25DegC_TB-1111%2B.zip)

jhillairet, 2022-04-10:
You are right, I was too fast in my example. The `Circuit` class also expect some `Circuit.Port` to know calculate the expected circuit. Hence for example:

```python
import skrf as rf

transformer = rf.Network(r"TTC1-33W+_Plus25DegC_TB-1111+.s4p", name='transformer')
freq = transformer.frequency
port1 = rf.Circuit.Port(freq, name="Port1")
port2 = rf.Circuit.Port(freq, name="Port2")
port3 = rf.Circuit.Port(freq, name="Port3")
ground = rf.Circuit.Ground(freq, name='ground')

cnx = [
       [(port1, 0), (transformer, 0)],
       [(port2, 0), (transformer, 1)],
       [(port3, 0), (transformer, 2)],
       [(transformer, 3),(ground, 0)]
]
cir = rf.Circuit(cnx)

ntw = cir.network  # a 3-port

# you can check the graph of your circuit
cir.plot_graph(network_labels=True)

```

Depending on your final application, the previous example is probably a bit verbose. It gets more relevant if the number of connection is higher, as the Circuit description is easier to read than multiple calls to connect. So if you just want to short a single port, you can also create an ideal short and connect it directly to your port 3:

```python
#%% method 2
media = rf.DefinedGammaZ0(freq)
short = media.short()
ntw2 = rf.connect(transformer, 3, short, 0)

print(ntw == ntw2)  # True
```

jhillairet, 2022-04-13:
@jrmagers , does it solve your issues?

jrmagers, 2022-04-13:
Yes, this worked. I used method 2. Thanks!

closed 2022-04-13 by jrmagers