scikit-rf/scikit-rf #629: TwoPortOnePath Example Assumes the Availability of Four S-parameters In Measurement Data

biergaizi, 2022-04-13:
In the example *[Calibration With Three Receivers](https://scikit-rf.readthedocs.io/en/latest/examples/metrology/Calibration%20With%20Three%20Receivers.html)*, the Two Port One Path calibration algorithm is demonstrated, allowing full error correction on a VNA that can only measures `a1`, `b1` and `b2`. In other words, only S11 and S21 can be measured each time. The other half, S22 and S12 are unknown. Thus, the DUT is required to be measured twice in the forward and reverse direction.

However, in the subsequent demonstration, the example still implicitly assumes that all four S-parameters are available. The document says,

> The measurements of the calibration standards and DUT’s were downloaded from the VNA by saving touchstone files of the raw s-parameter data to disk.
> ```
>    $ ls three_receiver_cal/data/
>    'attenuator (forward).s2p'       simulation.s2p
>    'attenuator (reverse).s2p'       thru.s2p
>     load.s2p                       'wr15 shim and swg (forward).s2p'
>    'quarter wave delay short.s2p'  'wr15 shim and swg (reverse).s2p'
>     short.s2p
> ```

Here, all the measurement data, including the calibration data for Open and Short, are all given in s2p data. For example, this is the content of `attenuator (forward).s2p`, which has all the four S-parameters available. 

    # GHz S RI R 50.0 
    !freq ReS11 ImS11 ReS21 ImS21 ReS12 ImS12 ReS22 ImS22
    60.0 -0.883511900902 -1.44533252716 -8.34876118461e-06 1.33288849611e-05 -0.705963575399 -0.684623175984 0.916409479195 -0.800569661745

And more confusingly, S11/S22 and S21/S12 are not equal, suggesting they really are separate measurements made on a real 2-port VNA instead of a 1.5-port VNA. 

Now, I was trying to apply the Two Port One Path calibration to my real measurement data from a 1.5 port VNA, the data has the form of:

    # GHz S RI R 50.0 
    !freq ReS11 ImS11 ReS21 ImS21 ReS12 ImS12 ReS22 ImS22
    60.0 re lm re lm 0 0 0 0

Since S12 and S22 are unknown, they're represented as `0` in the data files.

This is confusing and raises several questions. These need to be addressed in the tutorial.

1. During the OSM calibration, a 1.5-port VNA can only give S11 and S21, S12 and S22 are unknown and cannot be measured. What is the correct way to provide the calibration data to the `TwoPortOnePath()` algorithm? Currently I'm using the following method,
    ```
    # convert one-port measurements to two-port data by faking data
    short_std = two_port_reflect(short_std, short_std)
    open_std = two_port_reflect(open_std, open_std)
    load_std = two_port_reflect(load_std, load_std)
    ```
    Is it correct?
2. Due to the same reason, when the `thru_std` is passed to `TwoPortOnePath()`, it creates a division-by-zero error due to the 0s in the data files for the unknown S22 and S12. 
    ```
    /home/user/.local/lib/python3.9/site-packages/skrf/calibration/calibration.py:4601: RuntimeWarning: divide by zero encountered in true_divide
      k_second =1/(Etr/(Erf + Edf *(Elr - Esf)))
    /home/user/.local/lib/python3.9/site-packages/skrf/calibration/calibration.py:4601: RuntimeWarning: invalid value encountered in true_divide
      k_second =1/(Etr/(Erf + Edf *(Elr - Esf)))
    ```
    Thus, I'm using the following workaround.
    ```
    # convert one-port measurements to two-port data by faking data
    def one_half_ports_to_two_ports(network):
        s11 = network.s11
        s22 = s11
        s12 = network.s21
        s21 = s12
        return four_oneports_2_twoport(s11, s12, s21, s22)

    thru_std = one_half_ports_to_two_ports(thru_std)
    ```
    Is it correct? Long story short, the correct way to perform a SOLT calibration on a 1.5-port VNA needs to be fully explained in the Calibration With Three Receivers tutorial.

3. When the measurement data, e.g. `attenuator (forward).s2p` is passed into `TwoPortOnePath()`, it is unclear whether the S11/S21 pair is used, or the S22/S12 pair is used. 
4. 
    By reading the documentation, I noticed the default argument `source_port=1`, this seems to suggest the S11/S21 pair is used. If it's the case, it should be explained in the tutorial.

biergaizi, 2022-04-13:
After read more the documentation and code, this is the calibration procedure I'm using.

1. Attach a Load at Port 2. Measure Short-Load, Open-Load, Load-Load, and Thru. Save four s2p files.
2. Call `one_half_ports_to_two_ports()` (see above) to copy S11 to S22, S21 to S12, thus avoiding the division-by-zero problems. I no longer use `two_port_reflect()` to fake the data, since I think S21 is still needed for reflective standards to correct port crosstalk.
3. Create ideal standards for Open, Short, and Loads.
4. Pass measured standard, DUT, and ideal response to `TwoPortOnePath()`.

I hope this is the correct procedure. If so, the documentation can be updated to include a remark.

Vinc0110, 2022-04-13:
The numeric issue with S12=S22=0 is known and will be solved in the upcoming release, see #623 and and fix in #625.
I'm currently preparing an instrument class and a calibration example specifically for the NanoVNA V2 in #616. The procedure is very similar to what you summerized above for using `TwoPortOnePath`. 
Keep in mind to call  `apply_cal()` with both the forward and the reverse measurements for a full correction, as that worked much better in my case. See also the [example in the docs](https://scikit-rf.readthedocs.io/en/latest/examples/metrology/TwoPortOnePath%2C%20EnhancedResponse%2C%20and%20FakeFlip.html)

Vinc0110, 2022-04-13:
By the way, I agree that the documentation needs a big update for the Calibration classes. A lot of these intricacies are not documentated at all, especially regarding the API. Currently, a few more new examples are added and I'm collecting ideas for improvements in #592. Feel free to add more things to the list.

biergaizi, 2022-04-13:
> I'm currently preparing an instrument class and a calibration example specifically for the NanoVNA V2 in https://github.com/scikit-rf/scikit-rf/pull/616. The procedure is very similar to what you summerized above for using TwoPortOnePath.
Keep in mind to call apply_cal() with both the forward and the reverse measurements for a full correction, as that worked much better in my case. See also the [example in the docs](https://scikit-rf.readthedocs.io/en/latest/examples/metrology/TwoPortOnePath%2C%20EnhancedResponse%2C%20and%20FakeFlip.html)

Indeed, I'm an owner of NanoVNA v2 and I was trying to verify the performance of a RF precision reference attenuator I got from the flea market. [NanoVNA-Saver](https://github.com/NanoVNA-Saver/nanovna-saver) software only has a basic calibration algorithm that still gives a residue error of around 0.15 dB. Thus I tried to calibrate it manually using the sophisticated TwoPortOnePath w/ FakeFlip method in scikit-rf and that was how I encountered these problems.

The final data looks pretty good after I figured everything out. Measured value is within 0.05 dB of the official calibration (curiously, FakeFlip and EnhancedResponse gives essentially the same data, probably because Port 2 is highly matched in this case). The TwoPortOnePath algorithm allowed me to make the conclusion that this old attenuator is still usable as a reference. Thanks for your contribution to the community.

jhillairet, 2022-04-13:
@biergaizi if you want to submit a jupyter notebook explaining your calibration to add an example, it is very welcome!

jhillairet, 2022-04-24:
[Documented ](https://scikit-rf.readthedocs.io/en/v0.22.0/examples/metrology/NanoVNA_V2_4port-splitter.html)by @Vinc0110  in #616. @biergaizi, please send us some feedback. I'm closing for now but feel free to reopen it needed.

biergaizi, 2022-05-04:
I found the tutorial is extremely confusing, and doesn't really fix the problem as I originally reported. It's an "application note" type tutorial on using the NanoVNA and several different concepts are mixed together:

1. How to use the new Instrument class to control the NanoVNA and acquire the data (which should be better explained by adding an article in the "Virtual Instruments" section).
4. How to use the TwoPortOnePath class.
5. How to apply SOLT calibration (and it totally failed to explain it, the code assumes the SOLT calibration already exists, without telling the reader how they're measured, the reader should be directed to read SOLT first, or better, update the article "Calibration With Three Receivers" to explain the similarities and differences of SOLT calibration for two-port and 1.5-port).
6. How to measuring a multi-port device and merge the ports.

This is certainly useful and needed, as a tutorial for using the NanoVNA via scikit-rf. But it has two problem, (1) it doesn't fix what I originally reported (request for a general calibration procedure for 1.5-port VNA), and (2) if and only if the reader is already familiar with individual concepts.

# 1. It doesn't fix what was reported

What I was reporting was the lack of an explanation of the basic concepts in a 1.5-port VNA calculation and what I requested was an update to the article "Calibration With Three Receivers", not a NanoVNA tutorial.

In my original issue, I only wanted to point out that the tutorial "Calibration With Three Receivers" doesn't say a single word about SOLT calibration and that needs some elaboration. I knew how to apply the calibration, what I didn't know is how the SOLT standards are supposed to be measured. The situation was exacerbated by the fact that v0.22.0 was unreleased and there was absolutely no information even about a normal SOLT calibration.

But coincidentally, now a regular SOLT calibration tutorial is already added in v0.22.0, so I think the problem is already largely addressed. I think just directing the reader of "Calibration With Three Receivers" to "SOLT" and adding a few sentences to inform the reader about the similarities and differences should be sufficient.

# 2. As a NanoVNA tutorial, it introduces too many concepts at once.

For new users, I think it's better to introduce each concept to the reader separately first, rather than writing a catch-all tutorial with everything included. The final tutorial can be left as a final tutorial after individual concepts are introduced.

For example, a potential NanoVNA user may want to do some data post-processing with existing s2p and not interested in instrument control, another reader may need to understand Two Port One Path in a general sense.

Thus, I think it's helpful to add a general overview in the beginning of the NanoVNA tutorial.

# Plan

I'm willing to contribute a tutorial to address all of these problems, but I'm busy working on something else...

biergaizi, 2022-05-04:
BTW, I'm not allowed to reopen this issue.

jhillairet, 2022-05-04:
Reopened.

Please remember that the project is made and improved with benevolent people only. They do what they can and with their own understanding of what the issues are.

Vinc0110, 2022-05-05:
Wow. I think your message is rather rude, but let's not get into that too much.

I guess most of your points are valid and I do agree that the (my) NanoVNA example turned out more like a 3-in-1 tutorial (data acquisition, calibration, network assembly and plotting). However, I don't see a big problem there because readers can always skip parts of it and only use what they need.

In my opinion, the scientists and engineers using this package can/should handle the challenge of finding and selecting the information they need from all the different examples and tutorials available. It's not ideal, but it's not too bad either.

A more detailed description of the measurement of the calibration standards is definitely a good idea. In my demonstration, I did not pay much attention to the isolation and I honestly don't even know how the TwoPortOnePath algorithm handles S21 of the reflection standards (SHORT and OPEN). The unused port 2 was not even terminated in those measurements (which might have been a bad idea for the higher frequency range >1 GHz).

biergaizi, 2022-05-08:
@Vinc0110 It's not meant to be a criticism of your tutorial *per se*, by itself it provides useful information to NanoVNA users who are already familiar with how the scikit-rf calibration works, it's good for this particular purpose, and I don't think your current tutorial needs any change for this intention. Instead, The main issue I was reporting happens to be a problem in the general documentation of Three Receiver calibration (irrelevant to any particular hardware implementation). I think the cause of this misunderstanding was an overlap of scope - the NanoVNA happens to be a 3-receiver VNA, and there happens to be a need for a NanoVNA tutorial, what was complaining is that a NanoVNA tutorial is not to be confused with the documentation issue in Three Receiver calibration.

Back to the calibration problem. I took a look again at the Two Port One Path calibration, and I found I was mistaken, in fact, it does not perform any isolation calibration by the default (I incorrectly assumed it did that during the "load" calibration), the isolation error terms are assumed to be zero. However, all the optional parameters taken by `TwoPortOnePath` is passed directly into its parent class `TwelveTerm`, including `isolation`, thus an isolation calibration should still be possible. On the other hand, the `TwoPortOnePath` does not mention the `isolation` parameter, so I'm not sure if it's actually unsupported or just needs an update. Three Receiver calibration documentation needs an update to explain it.

Vinc0110, 2022-05-22:
Right, there is a way to pass an additional isolation measurement to TwelveTerm (through `**kwargs` of TwoPortOnePath). However, it does not do much, at least on the level of accuracy I'm dealing with on the NanoVNA V2.

Here is what I did:

Regular TwoPortOnePath:
```
cal1 = TwoPortOnePath(ideals=[line.short(nports=2), line.open(nports=2), line.match(nports=2), line.thru()],
                     measured=[short_raw, open_raw, match_raw, thru_raw],
                     n_thrus=1, source_port=1)
```

TwoPortOnePath with additional isolation (recycled from *match* measurement):
```
nw_iso = match_raw.copy()
nw_iso.s[:, 0, 1] = nw_iso.s[:, 1, 0]
nw_iso.s[:, 1, 1] = nw_iso.s[:, 0, 0]
cal2 = TwoPortOnePath(ideals=[line.short(nports=2), line.open(nports=2), line.match(nports=2), line.thru()],
                     measured=[short_raw, open_raw, match_raw, thru_raw],
                     n_thrus=1, source_port=1, isolation=nw_iso)
```

Applying both of them to the [4-port splitter example](https://scikit-rf.readthedocs.io/en/latest/examples/metrology/NanoVNA_V2_4port-splitter.html), the difference is tiny:
1) Results (raw, cal1, cal2)
[image]
2) Difference (cal1 - cal2)
[image]

biergaizi, 2022-06-08:
Another issue I found confusing is this statement in the documentation *[Calibration](https://scikit-rf.readthedocs.io/en/latest/tutorials/Calibration.html)*

> Two-port calibration is accomplished in an identical way to one-port, except all the standards are two-port networks. This is even true of reflective standards (S21=S12=0). So if you measure reflective standards you must measure two of them simultaneously, and store information in a two-port. For example, connect a short to port-1 and a load to port-2, and save a two-port measurement as ‘short,load.s2p’ or similar.

This is the source of my mistaken belief that a load at port 2 is always necessary since scikit-rf performs isolation calibration by default - which is not correct. In reality, I think this is unnecessary. It's my understanding that the 2-port calibration procedure is:

1. Perform SOL calibration at both port 1 and port 2, using the one-port reflectometer calibration procedure.
2. Perform Isolation Calibration by terminating port 1 and port 2 with a load. (Optional).
3. Perform 2-port Thru Calibration.

In the [SOLT example](https://scikit-rf.readthedocs.io/en/latest/examples/metrology/SOLT.html) this fact is made clear by this statement:

> If you don’t have the two same reflective standard, no worries! You can forge a two-port Network from two one-port Network using the function `skrf.network.two_port_reflect`.

So I think during a SOLT calibration, it's entirely acceptable to calibrate port 1 when nothing is connected to port 2, and then calibrate port 2 when nothing is connected to port 1, then do an isolation calibration if you wish, and finally perform a Thru. And in a three-receiver calibration, since port 2 cannot transmit, the only difference between 1-port and 2-port calibration is essentially an extra Thru step (ignoring isolation).

If what I've written is correct, I'd like to submit a patch to clarify the documentation as well.

Vinc0110, 2022-06-09:
It's always good to add comments on these intricacies. What you have summarized is also my understanding, but maybe someone else can confirm this as well? @Ttl @jhillairet @arsenovic .

Ttl, 2022-06-09:
It's ok to measure reflective standards as one-ports and then make a two-ports of two one-port measurements for the calibration. Reflective standard S12 and S21 are not used. The exception is 16-term calibrations which do require measuring two reflects simultaneously.

biergaizi, 2022-06-09:
Thanks, good to know. I didn't realize that scikit-rf in fact has a [16-term implementation](https://scikit-rf.readthedocs.io/en/latest/api/calibration/generated/skrf.calibration.calibration.SixteenTerm.html). I'll open a PR to include this clarification into the doc.

Vinc0110, 2022-06-11:
Sounds good. We the contributions you already made and maybe going to make, do you think we can close this issue?

biergaizi, 2022-06-18:
I still want a more detailed NanoVNA example in the context of three-receiver calibration and instrument control. But I also do realize that it's counterproductive to leave an issue open for too long.

I just opened three PRs. They only include minimal changes of clarifying the language and adding reference links to other guides. With these changes are in place, I believe the original issue as I reported can already be solved satisfactorily and the issue can be closed.

closed 2022-06-19 by jhillairet