scikit-rf/scikit-rf #631: Suggestion to add a Q-factor fitting algorithm to SKRF

andrewg22, 2022-04-18:
I recently published a report that contains a previously-unpublished algorithm for fitting Q-factor and resonant frequency to swept-frequency complex S-parameter data. In the case of S21 data, the leakage vector (which has the effect of making resonant peaks on a magnitude display appear asymmetric) is also fitted.

[Report MAT 58](https://www.researchgate.net/profile/Andrew-Gregory-11/publication/357556968_Q-factor_Measurement_by_using_a_Vector_Network_Analyser/)

The algorithm works very well in automated experiments, and could be a useful addition to SKRF. Demonstration Python code can be downloaded. What do other people think about the possibiity of adding it to SKRF? Would this be this useful, or is Q-factor fitting too remote from SKRF's purpose? Comments on how it could be implemented would also be welcome. I have not studied the SKRF codebase in detail, so I may not be able to give an immediate response to detailed comments on coding.

The SKRF implementation would require six coefficient (normally used for S21 or S12 Q-circles) and seven coefficient (normally used for S11 or S22 Q-circles) algorithms to be implemented. These actually determine the indicated (loaded) Q-factor. The unloaded Q-factor, which is the Q-factor of the resonator uncoupled from the 50-Ohm VNA ports, can then be estimated. The best experimental method is to first trigger a sweep with a wide sweep to obtain provisional values of Q and resonant frequency, then repeat the measurement with a narrower sweep centred on the resonance.

Andrew.

jhillairet, 2022-04-18:
Dear Andrew,

Thank you for your proposal! Yes, I think this is definitely of great interest and fits very well the scikit-rf package purposes.

Concerning the implementation, I see few possibilities we can discuss here, depending on the complexity/functionalities  of your code:
- adding a new function into `skrf/utils.py` that would return Q for a given `skrf.Network` object
- adding a new dedicated class if more functionalities could be provided than just the Q value (is it the case?)

In both case, we can add a new method to `Network` to make the Q-value calculation. 

Any other opinion?

andrewg22, 2022-04-18:
There is some modest additional complexity to determine the "unloaded" Q from the fitted "loaded" Q. The formula used is different for transmission and reflection resonances. For calculating unloaded Q, there are a few options regarding the 'scaling" of the data if, as is usually the case, the VNA is uncalibrated.

Instead of using a dedicated class, a function (added to Network) that returns a result object could be used (similar to how scipy.minimize operates). The result object could have methods to return the fitted parameters and calculate unloaded Q-factor. Do you think that this approach might be easier?

Experimenters would also find a method to return only the resonant frequency and optimum span useful - so that the VNA sweep parameters can be changed as shown in Figure 6.

jhillairet, 2022-04-21:
> Instead of using a dedicated class, a function (added to Network) that returns a result object could be used (similar to how scipy.minimize operates). The result object could have methods to return [...]

Eventually. The `Network.py` is already quite crowded, so I suggest adding things in this file as little as possible.
The resulting object class could be implemented in another file: either in a dedicated file (preferred) is your code is long (>100 lines), either in `util.py` if it is short. 
 
> Experimenters would also find a method to return only the resonant frequency and optimum span useful - so that the VNA sweep parameters can be changed as shown in Figure 6.

OK. I've seen some measurements in a kind of resonator as illustrated in your report to measure the permittivity and loss tangent of a dielectric sample. There were a huge number of peaks, and finding and fitting the correct ones is a kind of art. I guess that these methods would be definitely useful in such case. Such method could return a `Network` for a sub-frequency range directly.

arsenovic, 2022-04-21:
in my opinion a dedicated file is really nice and clean. easy to debug too.

andrewg22, 2022-04-21:
That sounds like a good plan. If you have time, might I suggest that you take a look at the [demonstration code](https://doi.org/10.47120/npl.MAT58) so you get a better idea of what is involved? This code (a few hundred lines) is not written in an object-orientated way. The files of test data are not in touchstone format (which I think is what SKRF uses), but it would be easy enough to convert them (I could do that).

I presume that the Network class will provide data and frequency arrays, and identify the S-parameter (it is important to know whether the data is reflection or transmission).
The main steps involved are:
   a. De-embed uncalibrated lines if necessary.
   b. Assign estimates for frequency and Q-factor if not already known.
   c. Perform linear fit (Step 1 in MAT 58) to get approximations for fit parameters.
   d. Perform non-linear optimisation (Step 2 in MAT 58). This gives resonant frequency and loaded Q-factor. Normally, six terms are fitted for transmission, and seven for reflection.
   e. Calculate unloaded Q-factor from the loaded Q-factor if required. There are a few possibilities here.

I think that a, b, c and d could all be done in one function. The calculation of unloaded Q-factor would probably be best implemented as a separate function.

jhillairet, 2022-04-22:
I will have a look. If you are not much comfortable yet with the `Network` class, I can try to make a first version for you to comment it.

jhillairet, 2022-04-22:
Another thing: the report and the code are distributed with a [(CC BY-ND 3.0) ](https://creativecommons.org/licenses/by-nd/3.0/), which means that we follow:

> Attribution — You must give [appropriate credit](https://creativecommons.org/licenses/by-nd/3.0/#), provide a link to the license, and [indicate if changes were made](https://creativecommons.org/licenses/by-nd/3.0/#). You may do so in any reasonable manner, but not in any way that suggests the licensor endorses you or your use.
> 
> NoDerivatives — If you [remix, transform, or build upon](https://creativecommons.org/licenses/by-nd/3.0/#) the material, you may not distribute the modified material. 

While it is of course perfectly OK to cite you and the DOI of the report in the code documentation, how should we interpret de NoDerivatives specification?

andrewg22, 2022-04-22:
The report text is released under the CC BY-ND 3.0 licence,  but  the computer code is released under the Creative Commons CC0 Public Domain licence - this is stated on the second page of the report and in the downloaded scripts. There should therefore be no problem creating derivatives of the computer code, and these can be re-issued under the licence used by SKRF if desired. Do you see any problems here? 

Demonstration code showing how to use the internals of the Network class would be appreciated .. if you have the time to create a first version that would be even better.

closed 2022-06-12 by jhillairet