scikit-rf/scikit-rf #640: NameError: name 'Frequancy' is not defined

j0hu3ttl, 2022-04-26:
Use of the newly implemented IEE370 deembedding methods #628 throws an error if using non-uniform frequency vectors.
It seems there are two typos in [deembedding.py](https://github.com/scikit-rf/scikit-rf/blob/master/skrf/calibration/deembedding.py)

Line 1860: `sfix_dut_fix.interpolate(Frequancy.from_f(fnew))`
Line 1864: `s2xthru.interpolate(Frequancy.from_f(fnew))`

Error Message: NameError: name 'Frequancy' is not defined

jhillairet, 2022-04-26:
Thanks for reporting this bug @j0hu3ttl (and welcome!). Note even with the typo fixed, this new feature does not support yet non-uniform frequency vectors.

closed 2022-04-27 by jhillairet

jhillairet, 2022-04-28:
The release v0.22.1 corrects this bug.

@j0hu3ttl, note that you can use the `interpolate` method using a uniform frequency vector to make it work.

commit 8067898d (2022-04-27): Fix #640 and add tests for non-uniform frequency array

quick fix, non-uniform frequency array is not yet implemented
commit 81dc0c21 (2022-04-27): Merge pull request #641 from jhillairet/fix/640

fix #640 and add tests for non-uniform frequency array