scikit-rf/scikit-rf #647: Formula in media.mline to calculate delta_w1 incorrect

rfbug, 2022-04-28:
I believe the formula used in `media.mline` to calculate `delta_w1` is not correct:
Hammerstad and Jensen define `t` to be the normalized strip thickness, while here `t` is given in meters. So the last multiplicaton by h is incorrect in my opinion. 

```python
def delta_w1(self) -> NumberLike:
    """
    Intermediary parameter. see qucs docs on microstrip lines.
    """
    w, h, t = self.w, self.h, self.t
    if t > 0.:
        # Qucs formula 11.22 is wrong, normalized w has to be used instead (see Hammerstad and Jensen Article)
        return t/pi * log(1. + 4*exp(1.)*tanh(sqrt(6.517*w/h))**2/t) * h
    else:
        return 0.
```
(As an aside, just because I always wondered and just finally checked: formula 11.22 in the Qucs documentation is indeed wrong as stated. Qucsator uses the original Wheeler formula in its implementation, though. )

Greetings, Michael
.

jhillairet, 2022-04-29:
Thanks for reporting @rfbug 

Is there a way to compare to other sources or results?

By the way, `mline.py` is poorly covered by the tests, so this is maybe an opportunity to improve things.

mhuser, 2022-04-29:
Hello @rfbug, in Hammerstad and Jensen paper the equations for \delta_u1 and \delta_ur is:

[image]

 Which are  later used here:

[image]

mhuser, 2022-04-29:
For scikit-rf implementation I followed (as far as my memory admit it) [qucs version of equations](http://qucs.sourceforge.net/tech/node75.html) that compute delta_w1 and delta_wr instead of u. For this reason t and w are not normalized here.

mhuser, 2022-04-29:
I think this is because in qucs, unormalized width is used here with w1 = w + delta_w1

[image]

but indeed the formula for `delta_w1` now looks strange to me. If we want unormalized delta_w1 it is enough to remove h from t/(pi * h) and there should be no need to multiply by h.

mhuser, 2022-04-29:
@jhillairet in #225 there was a comparison with a qucs generated line (qucs has option to pick impedance and dispersion model). Maybe a test case could be built comparing a generated .s2p and the results from `Media.mline` ?
The question is of course if we can trust qucs implementation 😄 
I have no access to other simulator like ADS that should be able to do it too with a choice of model.

mhuser, 2022-04-29:
Inside [889f4f8](https://github.com/scikit-rf/scikit-rf/commit/889f4f80cdb6a7572695710cabd8b78ddfbf9201) there was this change from:
```Python
return (t/h)/pi * log(1. + 4*exp(1.)/((t/h)/(tanh(sqrt(6.517*(w/h)))**2))) * h
```
To:
```Python
return t/pi * log(1. + 4*exp(1.)*tanh(sqrt(6.517*w/h))**2/t) * h
```
The first equation looks close to the paper and qucs **but** `tanh` is used instead of `cotanh`. `cotanh = 1/tanh`, which is correct.

Both t have lost their division by h, wich is wrong if the whole is multiplied by h.

rfbug, 2022-04-29:
Oh I totally missed the other t there... good catch;)

@jhillairet @mhuser Sorry, no access to ADS or alike here either. Maybe use some published data as test cases? I could do some literature browsing maybe...

mhuser, 2022-04-29:
Good idea @rfbug. Meanwhile I run a small test that I stored [in this branch](https://github.com/mhuser/scikit-rf/tree/fix-mline/examples/mline_test) against this qucs circuit with or without t divided by h in delta_w1:
[image]
This show me that while there is a good agreement with insertion loss, there is a slight deviation with return loss.
[image]
[image]
Maybe we should check against [qucsator source](https://github.com/Qucs/qucsator/blob/develop/src/components/microstrip/msline.cpp). I still remember quite well what the code is doing but I cannot do it right now. What is the pressure of time for that ?

JAnderson419, 2022-04-29:
I deal more with IC component modeling than circuit level stuff, so feel
free to let me know if some of my parameters are not matching yours, but
here is an output file from the shown ADS schematic for comparison.

[image: image.png]

*Jackson Anderson*
*Graduate Research Assistant*
*Electrical and Computer Engineering*
***@***.*** ***@***.***>*
*https://engineering.purdue.edu/hybridmems/
<https://engineering.purdue.edu/hybridmems/>*

JAnderson419, 2022-04-29:
(edit: repost to fix image and s2p upload as txt since email replies don't support markdown)

I deal more with IC component modeling than circuit level stuff, so feel
free to let me know if some of my parameters are not matching yours, but
here is an output file from the shown ADS schematic for comparison.

[image]

Jackson
[tline_ads.txt](https://github.com/scikit-rf/scikit-rf/files/8591363/tline_ads.txt)

mhuser, 2022-04-30:
> `delta_w1` and `delta_wr` magnitudes don't look correct right now. They are so small that changing `t` barely changes anything.
> 
> Here's one example with thin substrate microstrip line where thickness should have measurable effect:
> 
> ```
> m = MLine(frequency=skrf.F(10,10,1), w=100e-6, h=100e-6, t=30e-6, ep_r=3, rho=2e-8)
> m0 = MLine(frequency=skrf.F(10,10,1), w=100e-6, h=100e-6, t=0, ep_r=3, rho=2e-8)
> 
> print('t=30 um')
> print('z0', m.z0)
> print('gamma', m.gamma)
> print('w1', m.delta_w1)
> print('wr', m.delta_wr)
> 
> print()
> print('t=0')
> print('z0', m0.z0)
> print('gamma', m0.gamma)
> print('w1', m0.delta_w1)
> print('wr', m0.delta_wr)
> ```
> 
> ```
> t=30 um
> z0 [83.7079855]
> gamma [2.29635364+316.66950244j]
> w1 1.2200526316282287e-08
> wr [8.90088257e-09]
> 
> t=0
> z0 [83.71029485]
> gamma [2.296476+316.67338239j]
> w1 0.0
> wr 0.0
> ```

I am moving from w to normalized u and it indeed give sligthly more change:
t=30 um
z0 [77.58574475]
gamma [2.02436782+307.82088967j]
u1 0.3431916068243951
ur [0.25037512]

t=0
z0 [83.68830449]
gamma [2.296476+316.67338239j]
u1 0.0
ur [0.] 

I need to fix MLine line to take by default the frequency dependant Z0 computed by the equations instead of z0 passed as a media parameter (which obviously we only need if we renormalize to that impedance using embed = True).
I am always confused with that because when z0 = Z0, the S11 is constant 0 (-infinite dB) and we need to renormalize or connect with 50 ohm thru() on each side to plot the usual dB curve with hills and valleys.

mhuser, 2022-04-30:
I started an attempt in #650 to refactor `MLine` to follow qucsator implementation, which seems to follow the Hammerstad and Jensen paper well.
Agreement between qucs and skrf is superb with this version, even for S11 !
[image]
[image]
I still need to check against the ADS file and see what notebooks are broken and add some test cases.

mhuser, 2022-05-03:
[image]
color lines : qucs .s2p
color 'x' : new skrf mline

mhuser, 2022-05-07:
Hello @JAnderson419,
Could you please re-run you ADS simulation but with following changes?
This would enable me to test consistency with ADS results in the same conditions as my other tests.

- Frequency from 0.2 to 20 GHz, 100 points linear sweep
- Length 25mm 

I currently have kind of an issue skrf vs ADS I am working on 😅 :
[image]

mhuser, 2022-05-07:
Could you check too that following parameters are set into MSUB block ?
|  |  |
|---|---|
| FreqForEpsrTanD | 1.0e9 |
| LowFreqForTanD | 1.0e3 |
| HighFreqForTanD | 1.0e12 |

mhuser, 2022-05-08:
By the way, it would be interesting to also have a second dataset with the same parameters as abow but with `DielectricLossModel` set to frequency invariant (maybe a string or value 0?). This would enable a check Qucs vs ADS, because Qucs implement the Kirschning model but not the Svensson/Djordjevic, that skrf does.
I know this start to appear as a long wishlist 😅

mhuser, 2022-05-10:
Thank you so much @JAnderson419 ! I am trying them at the moment and the results are of highest interest !

mhuser, 2022-05-10:
If comparing Qucs against ADS and close our eyes about S11 as for now, there is a small difference between insertion loss S21.
[image]

mhuser, 2022-05-10:
Then I use frequency-dispersed impedance and dielectric constant to compute the losses and ...
[image]

But now skrf agree with ADS for S21 and no more with Qucs !

mhuser, 2022-05-10:
The original Hammerstad and Wheeler is ambiguous because it use the same variable names for frequency-dispersed and quasi-static values. However, it probably make sense to compute the losses with the same frequency-dispersed values that are used to compute the propagation constants.

mhuser, 2022-05-10:
While agreeing with qucs, the S11 looks bad if compared to ADS in region where the magnitude is small (qucs only has frequinvariant).
[image]

mhuser, 2022-05-11:
@JAnderson419 could you run the same two simulation but once with Kobayashi and once with Yamashita instead of Kirschning ?
Do you agree if I use these files as test-cases in skrf ?

mhuser, 2022-05-11:
@rfbug do you know where to find R. H. Jansen and M. Kirschning, ``Arguments and an accurate Model for the Power-Current Formulation of Microstrip Characteristic Impedance,'' Archiv für Elektronik und Übertragungstechnik (AEÜ), vol. 37, pp. 108-112, 1983 ?

mhuser, 2022-05-11:
I think ADS is probably using the modified Kirschning Jansen equation, were u is w_eff(t, f)/h, while in current qucs and skrf implementation w_eff is a constant, mostly equivalent to w_eff(t, f = 0). I will try to test this hypothesis when time will allow for it.

jhillairet, 2022-05-11:
> Now I face the terrible situation were all my tests agains Qucs fails if I fix mline to follow ADS losses.

Will you submit an issue to Qucs ? They would be probably interested to know about these differences.

mhuser, 2022-05-11:
> > Now I face the terrible situation were all my tests agains Qucs fails if I fix mline to follow ADS losses.
> 
> Will you submit an issue to Qucs ? They would be probably interested to know about these differences.

I have ! Would it be appropriated to mention our issue inside theyr for reference ?

mhuser, 2022-05-11:
> I think ADS is probably using the modified Kirschning Jansen equation, were u is the frequency-dependant thickness corrected width w_eff(t, f)/h, while in current qucs and skrf implementation w_eff is a constant, mostly equivalent to w_eff(t, f = 0). I will try to test this hypothesis when time will allow for it.

I found a description of Modifier Kirschning Jansen into Verma A. K., "[Introduction To Modern Planar Transmission Lines](https://books.google.ch/books?id=-pMvEAAAQBAJ&pg=PA284&lpg=PA284&dq=improved+kirschning+impedance+dispersion+formula&source=bl&ots=sVGiyPZyM9&sig=ACfU3U3tAmpJjJjc5O6_Wr7zihVV_3TTAQ&hl=fr&sa=X&ved=2ahUKEwj5-Je3htf3AhUQ6qQKHXsiANAQ6AF6BAghEAM#v=onepage&q=improved%20kirschning%20impedance%20dispersion%20formula&f=false)", Wiley & Sons, Wiley-IEEE Press, 2021, p. 284 But have not found a related paper yet.

JAnderson419, 2022-05-11:
ADS references the following work in the MLIN man page in case it is helpful:

M. Kirschning and R.H. Jansen, "Accurate Model for Effective Dielectric Constant of Microstrip and Validity up in Millimeter-Wave Frequencies," Electron . Lett, Vol. 18 March 18, 1982, pp. 272-273.

[data (1).zip](https://github.com/scikit-rf/scikit-rf/files/8669717/data.1.zip)

mhuser, 2022-05-11:
> ADS references the following work in the MLIN man page in case it is helpful:
>
> M. Kirschning and R.H. Jansen, "Accurate Model for Effective Dielectric Constant of Microstrip and Validity up in [...]

Thank you so much for new dataset and hint @JAnderson419 
Unfortunately, I already have the paper but the MLIN man page say _Dispersion effects are included using [...] the improved version of the Kirschning and Jansen model_ while the article is only the "regular" Kirschning and Jansen model. Furthermore, the paper contain only the dispersion for dielectric and not for the impedance. There is another paper with impedance dispersion, but not cited in the page, but the equations are referenced in qucs and in a textbook.
However, I will try with the other models and maybe I will understand more, the S11 difference does not seems random so it seems worth to continue the efforts to some extents.

rfbug, 2022-05-11:
First of all let me say a word of appreciation at @mhuser and all other contributors: thank for your energy and effort - it is much appreciated.

I remember coming across that AEÜ paper when I was looking at coupled microstrip lines. It is cited all over the place but nowhere to find. After some digging around I finally found a library that has the issue on file and will order a copy of the paper. I will keep you posted...

As far as I remember the relevant expressions are reproduced in *Reinmut K. Hoffman.,“Handbook of Microwave Integrated Circuits,”. Artech House, Inc. 1987.* pp. 177. 

[image]
[image]
[image]
[image]

mhuser, 2022-05-11:
Thank you a lot @rfbug ! I verified the code thoroughly and qucs and skrf seems to stick to these equations. Beside I also had the impression this article is a lot cited but it is very difficult to read it !

This is the current state :
[image]

From the perspective of the original issue, the refactoring of mline will make it follow the same implementation as QUCS.

[image]

If compared to ADS thanks to @JAnderson419, the S11 and S21 deviate slightly

[image]

The S21 can be almost fixed by using frequency-dispersed impedance and dielectric constant to compute it, but the S11 need to be further investigated (seems from the ADS plot that perhaps the same dispersion equation is used for impedance with only the dielectric constant equation that change).

I added a `compatibility_mode = 'qucs'` or `= 'ads'` in the `MLine` media to switch between both behaviour without having to change the class for testing purpose.

mhuser, 2022-05-12:
For these courageous ones interested in the Modified Kirschning Jansen model with thickness correction, I found this paper https://doi.org/10.1049/joe.2017.0078 which is openly accessible and cite the model with the related equations. It also embeds some plots for loss and effective permittivity.

mhuser, 2022-05-12:
And finally the strip thickness correction used in Modified Kirschning Jansen is from [I. J. Bahl and R. Garg, ‘Simple and accurate formulas for microstrip with finite strip thickness,” Proc. IEEE, vol. 65, pp. 1611–1612, Nov. 1977](https://doi.org/10.1109/PROC.1977.10783), (require IEEE access).
I will give a try to this during next sprint.

mhuser, 2022-05-12:
[image]

[image]

- ADS seems to use the Kirschning Jansen impedance dispersion in the case the permittivity dispersion is Kobayashi or Yamashita. QUCS on the other hand use constant impedance with quasi-static value in these cases.
- skrf Djordjevic/Svensson wideband Debye frequency-varying permittivity agree whith ADS (not implemented in Qucs) 
- There is still an unexplained feature in return loss phase and magnitude. I tried Modified Kirschning Jansen but it has almost no effect at these frequencies. Any ideas welcome !

By default `Media.MLine` follow QUCS implementation unless `compatibility_mode = 'ads'` is specified in its constructor. This should not break already existing scripts.

I will show an example plot in the QUCS issue and look if they are interested into correcting the insertion loss behaviour and eventually the impedance dispersion.

As for now, I think that I need to go once trough the docstrings, make #650 ready and maybe create a new issue dedicated to the S11 issue with ADS-like behaviour.

mhuser, 2022-05-14:
> I found a description of Modifier Kirschning Jansen into Verma A. K., "[Introduction To Modern Planar Transmission [...]

Ok, I found a local library selling this textbook (which also cover other planar medias such as CPW, CPWG and coupled microstrips 😄 ). I will make a last attempt to rewrite `Modified Kirschning Jansen` from blank to compute a frequency-dependant equivalent width to account for strip thickness. Perhaps it will by interesting to leave it in the media as `ModifiedKirschingJansen` if the results are closes.

mhuser, 2022-05-14:
[image]

Narrowing, but this is not what I tough.

mhuser, 2022-05-14:
Thanks to @JAnderson419 ADS files I made following finding: After looking at the S11 in the smith chart, I noticed there was a small imaginary offest in quasi-static impedance used by ADS data.

This come from ADS using a complex definition of the relative permittivity by incorporating the tand loss factor like `ep_r_cpmplex =  ep_r - 1j * ep_r * tand`. I already knew this was the case because it is used into the Svensson/Djodjevic wideband Debye dispersion model. However, I took only the real part of this value to be compatible with QUCS.

It will now be possible to pick the behaviour using `compatibility_mode = 'qucs'` or `compatibility_mode = 'ads'` when initializing an instance of `mline`.

ADS behaviour is still a bit questionable because it lead to a complex characteristic impedance and I am not sure the initial equations were made with complex permittivity and impedance in mind.

@rfbug I noticed the example values I took, has a quite small thickness (35um) compared to width (3mm) and height (1.55mm). Do you need a test with a thicker line to have more effect of width correction (initial issue) or are you satisfied with current attempts ? I am still interested if you get the Kirschning Jansen paper which is missing my collection 😄 

[image]

[image]

rfbug, 2022-05-15:
> @rfbug I noticed the example values I took, has a quite small thickness (35um) compared to width (3mm) and height (1.55mm). Do you need a test with a thicker line to have more effect of width [...]

Oh no, anything you are ok with is totally fine with me :+1: ... it was really just by chance I stumbled across the initial issue;) 

I have acquired copy of the AEÜ Kirschning Jansen paper by now. Due to copyright issues I would rather not attach the whole paper in this place. The formulas I quoted above from the Hoffmann book do however match the ones in the paper.

Ttl, 2022-05-15:
>ADS behaviour is still a bit questionable because it lead to a complex characteristic impedance and I am not sure the initial equations were made with complex permittivity and impedance in mind.

Complex Z0 with lossy dielectric sounds correct. In RLGC transmission line the dielectric loss is modeled with the G element as `G = tand * 2 * pi * f * C`. If I set the conductor as lossless and calculate C and L with [online calculator](https://www.emisoftware.com/calculator/microstrip-capacitance/). I get very similar complex Z0 to what ADS model gives.

```python
freq = skrf.F(10, 10, 1, unit='GHz')
tand = 0.01
C = 143e-12
G = tand*freq.f*2*np.pi*C
md = DistributedCircuit(frequency = freq,
                          z0 = 50,
                          C = C,
                          L = 230e-9,
                          R = 0,
                          G = G)
print('RLGC Z0', md.Z0)
print('RLGC gamma', md.gamma)

m_ads = MLine(frequency=freq, z0=50, w=300e-6, h=100e-6, t=0, ep_r=3, tand=tand, rho=0, compatibility_mode='ads')
print('ADS Z0', m_ads.Z0)
print('ADS gamma', m_ads.gamma)

m_qucs = MLine(frequency=freq, z0=50, w=300e-6, h=100e-6, t=0, ep_r=3, tand=tand, rho=0, compatibility_mode='qucs')
print('qucs z0', m_qucs.Z0)
print('qucs gamma', m_qucs.gamma)
```

```
RLGC Z0 [40.10325411+0.20051126j]
RLGC gamma [1.8016747+360.34394893j]
ADS Z0 [44.79617423+0.1991969j]
ADS gamma [1.46557043+326.67485775j]
qucs z0 [44.79751115]
qucs gamma [1.46270228+326.67483055j]
```

mhuser, 2022-05-15:
This make sense !
And it give us another example of these networks with complex characteristic impedance for the connexion of which @Ttl made recent development. Unless we renormalize it to some real ort impedance of course.

closed 2022-05-21 by jhillairet

JAnderson419, 2022-10-11:
If it is helpful, the ADS man page for MLIN references the following work:

M. Kirschning and R.H. Jansen, "Accurate Model for Effective Dielectric
Constant of Microstrip and Validity up in Millimeter-Wave Frequencies,"
Electron . Lett, Vol. 18 March 18, 1982, pp. 272-273

*Jackson Anderson*
*Graduate Research Assistant*
*Electrical and Computer Engineering*
***@***.*** ***@***.***>*
*https://engineering.purdue.edu/hybridmems/
<https://engineering.purdue.edu/hybridmems/>*