scikit-rf/scikit-rf #648: CPW, Coaxial and MLine low frequency conductor loss is incorrect

Ttl, 2022-04-29:
CPW, Coaxial and MLine media calculate the conductor loss using `surface_resistivity` which calculates it as `rho/skin_depth`. At low frequencies where skin depth is greater than conductor thickness this gives overly optimistic results. Currently at DC the lines are lossless.

mhuser, 2022-04-30:
Well spotted @Ttl.
Do you tink it is better to:

- saturate the ouput of this function inside `Media`
- modify `surface_resistivity`  with a `max_depth = None` parameter that if not None take `depth = minimum(skin_depth(), max_depth)` as effective depth (would not change existing codes)
- other solution ?

Ttl, 2022-04-30:
I think that for coaxial cable analytic solution for the current density can be found. Not sure about microstrip and CPW, maybe there are some approximations?

Saturating or taking a min/max has a problem that it creates a discontinuity and looks very unnatural.

mhuser, 2022-04-30:
In the Hammerstad and Jensen paper for microstripline it is stated that the current distribution proposed is meant to be _[...] a very good approximation to the current distribution factor [...] provided that the strip thickness exceeds three skin depths._ They are referencing R. A. Pucel et al. "Losses in Micostrip", _IEEE-trans._, 1968.

In qucs, they solved this issue by sending a warning when the condition is not met.

Obviously we now that at DC the microstrip will be a full surface resistance but we are missing the portion between that point and the skin_depth = t/3.

Ttl, 2022-04-30:
I changed the Coaxial resistance calculation so that it works at DC in #649. However S-parameters are not calculated correctly at DC and the lines are still lossless. I guess the same issue applies to other media too.

A more accurate model would be similar to the coaxial cable case where it's assumed that current is uniform in horizontal direction and decreases exponentially going upwards. But maybe minimum might be fine for most of the cases after all and is at least more accurate than the current model. In reality microstrip current density is quite complex and depends on the width and height due to current crowding at the corners.

mhuser, 2022-04-30:
Indeed, currently MLine alpha_conductor is defined as such, with `skin_depth` and `surface_resistance` at two place:
```Python
def alpha_conductor(self) -> NumberLike:
        """
        Losses due to conductor resistivity.

        Returns
        -------
        alpha_conductor : array-like
                lossyness due to conductor losses
        See Also
        --------
        surface_resistivity : calculates surface resistivity
        """
        if self.rho is None or self.t is None:
            raise(AttributeError('must provide values conductivity to calculate this. see initializer help'))
        else:
            f = self.frequency.f
            ZF0 = npy.sqrt(mu_0/epsilon_0)
            ZL, rho, mu_r  = real(self.Z0_f), self.rho, self.mu_r
            w = self.w + self.delta_ur * self.h
            rough = self.rough
            Kr  = 1 + 2/pi*arctan(1.4*(rough/skin_depth(f, rho, mu_r))**2)
            Ki  = exp(-1.2*(ZL/ZF0)**0.7)
            Rs  = surface_resistivity(f=f, rho=rho, mu_r=1)
        return  Rs*Kr*Ki/(ZL*w)
```
I will try to look at MLine #647 first

mhuser, 2022-05-04:
With #650 `MLine` will warns when outside the domain of validity of conductor loss computation (t < 3 * skin_depth).
This is the same behavious as Qucs simulator.

mhuser, 2022-05-15:
With #665 CPW warns when outside the domain of validity of conductor loss computation (t < 3 * skin_depth).
Qucs send no warnings for this components, but we can probably deduce empirically that we can apply the same conditions than for microstripline.

closed 2022-05-21 by jhillairet

Ttl, 2022-05-22:
The loss is still calculated incorrectly though. Coaxial is calculated correctly now but Mline and CPW only warn that the results are incorrect.

mhuser, 2022-05-22:
Indeed, however at least the user will now know it.
For PCB, most planar conductor are somewhat shorts so their DC resistance is quite small however. There are not such lengths that could be seen with coax lines. Maybe this could be an issue for MMIC where everything is small so resistances could be biggers.

I heard about Holloway and Kuester method to compute the loss that could solve the thickness vs skin_depth issue of Wheeler's incremental inductance rule, but I had no occasion to read them yet.  [A quasi- closed-form expression for the conductor loss of CPW lines with [...] ](https://doi.org/10.1109/22.477846) and [Expression for the conductor loss of stripline and coplanar strip [...]](https://doi.org/10.1002/(SICI)1098-2760(20000505)25:3%3C162::AID-MOP3%3E3.0.CO;2-W)