scikit-rf/scikit-rf #651: Line S-parameters are incorrect if characteristic impedance is complex

Ttl, 2022-05-01:
At very low frequencies characteristic impedance of transmission line is complex. skrf doesn't seem to handle it correctly because mixing of port reference impedance `z0` and characteristic impedance of line `Z0` in calculating the S-parameters and use of powerwave S-parameter definition.

Below is example of coaxial line with 1 cm length and very low conductivity resulting in high loss. DC resistance is about 50 ohms and the frequency is 1 MHz so the line is electrically very short and could be modeled as series resistor. Requires #649 for resistance to be calculated correctly.

`media_z0` is `z0` given to the media. Currently this means setting the port impedance `z0` to this value and characteristic impedance `Z0` is left unchanged. None sets port impedance to characteristic impedance.

`line_z0` is the characteristic and port impedance to use when creating a line.

If `embed` is True then the port impedances are renormalized to `z0`.

Given the above definitions I would assume that setting `media_z0=50, line_z0=None, embed=True` should make a line with the complex characteristic impedance and renormalize the ports to 50 ohms matching the series resistor calculation. However this is not the case.

```python
import numpy as np
import skrf
from skrf.media import Coaxial, Freespace

freq = skrf.Frequency(1, 1, 1, unit='MHz')
line_l = 1e-2 # meters

for media_z0 in [50, None]:
    for line_z0 in [None, 50]:
        for line_embed in [False, True]:
            print(f'media_z0: {media_z0}, line_z0: {line_z0}, Embed: {line_embed}')
            media = Coaxial(freq, z0=media_z0, Dint=0.5e-3, Dout=1.0e-3, sigma=1e3)
            print(' z0: {:.2f}, Z0: {:.2f}, gamma: {:.2f}'.format(media.z0[0], media.Z0[0], media.gamma[0]))
            line = media.line(line_l, 'm', embed=line_embed, z0=line_z0)
            print(' line S11', line.s[0,0,0])
            print(' line S21', line.s[0,1,0])
            print(' line z0', line.z0[0,0])
            if line.z0[0,0] != 50:
                line.renormalize(50, s_def='traveling')
                print(' line renorm S11', line.s[0,0,0])
                print(' line renorm S21', line.s[0,1,0])
            print()

print('S-parameters calculated as series resistor')
r_dc = line_l*(1/media.sigma)/(np.pi*media.a**2)
print('dc R', r_dc)

if 0:
    # Manually assemble series resistor S-parameters
    y = np.zeros((1,2,2), dtype=complex)
    y[:,0,0] = 1/r_dc
    y[:,1,1] = 1/r_dc
    y[:,0,1] = -1/r_dc
    y[:,1,0] = -1/r_dc

    # This crashes
    net_r = skrf.Network(y=y, frequency=freq, z0=50)
else:
    media = Freespace(frequency=freq, z0=50)
    net_r = media.resistor(r_dc)

print('Resistor S11', net_r.s[0,0,0])
print('Resistor S21', net_r.s[0,1,0])
```

```
media_z0: 50, line_z0: None, Embed: False
 z0: 50.00, Z0: 2257.60-2257.22j, gamma: 1.14+1.14j
 line S11 0j
 line S21 (0.988617463159112-0.011255830151580522j)
 line z0 (50+0j)

media_z0: 50, line_z0: None, Embed: True
 z0: 50.00, Z0: 2257.60-2257.22j, gamma: 1.14+1.14j
 line S11 0j
 line S21 (0.988617463159112-0.011255830151580522j)
 line z0 (50+0j)

media_z0: 50, line_z0: 50, Embed: False
 z0: 50.00, Z0: 2257.60-2257.22j, gamma: 1.14+1.14j
 line S11 0j
 line S21 (0.988617463159112-0.011255830151580522j)
 line z0 (50+0j)

media_z0: 50, line_z0: 50, Embed: True
 z0: 50.00, Z0: 2257.60-2257.22j, gamma: 1.14+1.14j
 line S11 0j
 line S21 (0.988617463159112-0.011255830151580522j)
 line z0 (50+0j)

media_z0: None, line_z0: None, Embed: False
 z0: 2257.60-2257.22j, Z0: 2257.60-2257.22j, gamma: 1.14+1.14j
 line S11 0j
 line S21 (0.988617463159112-0.011255830151580522j)
 line z0 (2257.6025298862933-2257.2199592444235j)
 line renorm S11 (0.3394831573782906-9.291482605883979e-05j)
 line renorm S21 (0.6605168053866275-0.00015923170778380224j)

media_z0: None, line_z0: None, Embed: True
 z0: 2257.60-2257.22j, Z0: 2257.60-2257.22j, gamma: 1.14+1.14j
 line S11 0j
 line S21 (0.988617463159112-0.011255830151580522j)
 line z0 (2257.6025298862933-2257.2199592444235j)
 line renorm S11 (0.3394831573782906-9.291482605883979e-05j)
 line renorm S21 (0.6605168053866275-0.00015923170778380224j)

media_z0: None, line_z0: 50, Embed: False
 z0: 2257.60-2257.22j, Z0: 2257.60-2257.22j, gamma: 1.14+1.14j
 line S11 0j
 line S21 (0.988617463159112-0.011255830151580522j)
 line z0 (50+0j)

media_z0: None, line_z0: 50, Embed: True
 z0: 2257.60-2257.22j, Z0: 2257.60-2257.22j, gamma: 1.14+1.14j
 line S11 (-0.3394831573782804+9.291482605806173e-05j)
 line S21 (0.6605168053866376-0.00015923170763578625j)
 line z0 (2257.6025298862933-2257.2199592444235j)
 line renorm S11 (1.1379786002407855e-15-9.992007221626409e-15j)
 line renorm S21 (0.9886174631591118-0.011255830151576995j)

S-parameters calculated as series resistor
dc R 50.92958178940652
Resistor S11 (0.3374393620229403+0j)
Resistor S21 (0.6625606379770599+0j)
```

By default there is no way to get S-parameters that match the series resistor calculation. `media_z0: None, line_z0: 50, Embed: True` is close but S11 is negative which means negative series resistor and the port impedance is not 50 ohms. I assumed that setting `line_z0=50` would use 50 ohm characteristic impedance but I'm not sure that's the case since port impedance is left as characteristic impedance?

The first two results is with 50 ohm port impedance, but skrf seems to calculate the S-parameters as if the characteristic impedance of line was 50 ohms. When reference impedance is set to characteristic impedance the S-parameters are calculated correctly, however the port impedance is left as characteristic impedance. It can be renormalized to 50 ohms but either `traveling` or `pseudo` definition must be used as `power` gives incorrect results.

This is the renormalization result if `power` definition is used:
```
 line renorm S11 (0.9992631627440348+0.021882842141062064j)
 line renorm S21 (0.0004818427027186568-0.022134902799545834j)
```

`pseudo` renormalization gives results very close to `traveling`.

I think the first step to fixing this would be to decide what the various `z0` arguments should do. I think `traveling` renormalization should be used in line and it should give correct results when the starting impedance is the characteristic impedance, but I'm not completely sure about this. `power` however seems to not work correctly.

mhuser, 2022-05-01:
Dear @Ttl, please mind I am not the author of `Media` or `Coax` so my understanding may be incomplete. However, I stumbled on this behaviour with `MLine` and `CPW`.
Here is what is happening with current `scikit-rf`:
When `Coax.line()` is called, it execute `Media.line` inherited from `Media`.
```Python
def line(self, d: NumberLike, unit: str = 'deg',
             z0: Union[NumberLike, str, None] = None, embed: bool = False, **kwargs) -> Network:
        r"""
        [...]
        """

        if isinstance(z0,str):
            z0 = parse_z0(z0)* self.z0

        kwargs.update({'z0':z0})
        result = self.match(nports=2,**kwargs)

        theta = self.electrical_length(self.to_meters(d=d, unit=unit))

        s11 = npy.zeros(self.frequency.npoints, dtype=complex)
        s21 = npy.exp(-1*theta)
        result.s = \
                npy.array([[s11, s21],[s21,s11]]).transpose().reshape(-1,2,2)

        if  embed:
            result = self.thru()**result**self.thru()

        return result
```
If `line` is invoked without `embed = True` it is only a perfect losseless line with a length. The impedance is set by the `match` method. If `embed = True` then the network is connected to `thru()` on both side which are only zero length `line` of same media invoked without `z0` parameter (default to `None`).
```Python
def match(self, nports: int = 1, z0: Union[NumberLike, None] = None,
              z0_norm: bool = False, **kwargs) -> Network:
        r"""
        [...]
        """
        result = Network(**kwargs)
        result.frequency = self.frequency
        result.s = npy.zeros((self.frequency.npoints, nports, nports),\
                dtype=complex)
        if z0 is None:
            z0 = self.z0
        elif isinstance(z0, str):
            z0 = parse_z0(z0)*self.z0

        if z0_norm:
            z0 = z0*self.z0

        result.z0 = z0
        return result
```
The attribut `z0` of `Media` is like this:
```Python
@property
    def z0(self) -> npy.ndarray:
        """
        [...]
        """
        if self._z0 is None:
            return self.Z0
        return self._z0*ones(len(self))
```

If line_z0 = none, and media_z0 set, take media_z0 wich is also z0 for thru so embed has no effect. Z0 is ignored.
```
media_z0: 50, line_z0: None, Embed: False
 z0: 50.00, Z0: 2257.60-2257.22j, gamma: 1.14+1.14j
 line S11 0j
 line S21 (0.988617463159112-0.011255830151580522j)
 line z0 (50+0j)

media_z0: 50, line_z0: None, Embed: True
 z0: 50.00, Z0: 2257.60-2257.22j, gamma: 1.14+1.14j
 line S11 0j
 line S21 (0.988617463159112-0.011255830151580522j)
 line z0 (50+0j)
```
If line_z0 = media_z0 is set, thru and line have same z0 so embed has no effect. Z0 is ignored.
```
media_z0: 50, line_z0: 50, Embed: False
 z0: 50.00, Z0: 2257.60-2257.22j, gamma: 1.14+1.14j
 line S11 0j
 line S21 (0.988617463159112-0.011255830151580522j)
 line z0 (50+0j)

media_z0: 50, line_z0: 50, Embed: True
 z0: 50.00, Z0: 2257.60-2257.22j, gamma: 1.14+1.14j
 line S11 0j
 line S21 (0.988617463159112-0.011255830151580522j)
 line z0 (50+0j)
```
If line_z0 = media_z0= None, Z0 is taken for z0 for line and thru so embed has no effect.
```
media_z0: None, line_z0: None, Embed: False
 z0: 2257.60-2257.22j, Z0: 2257.60-2257.22j, gamma: 1.14+1.14j
 line S11 0j
 line S21 (0.988617463159112-0.011255830151580522j)
 line z0 (2257.6025298862933-2257.2199592444235j)

media_z0: None, line_z0: None, Embed: True
 z0: 2257.60-2257.22j, Z0: 2257.60-2257.22j, gamma: 1.14+1.14j
 line S11 0j
 line S21 (0.988617463159112-0.011255830151580522j)
 line z0 (2257.6025298862933-2257.2199592444235j)
```
If line_z0 = 50 and media_z0= None, 50 is taken for line and Z0 is taken for thru so embed renormalize to Z0 which is kind the inverse as targeted.
```
media_z0: None, line_z0: 50, Embed: False
 z0: 2257.60-2257.22j, Z0: 2257.60-2257.22j, gamma: 1.14+1.14j
 line S11 0j
 line S21 (0.988617463159112-0.011255830151580522j)
 line z0 (50+0j)

media_z0: None, line_z0: 50, Embed: True
 z0: 2257.60-2257.22j, Z0: 2257.60-2257.22j, gamma: 1.14+1.14j
 line S11 (-0.3394831573782804+9.291482605806173e-05j)
 line S21 (0.6605168053866376-0.00015923170763578625j)
 line z0 (2257.6025298862933-2257.2199592444235j)
```
Have you tried this ?
```Python
media_z0 = 50.
line_embed = True
media = Coaxial(freq, z0=media_z0, Dint=0.5e-3, Dout=1.0e-3, sigma=1e3)
line = media.line(line_l, 'm', embed=line_embed, z0=media.Z0)
```
Which give
```
z0: 50.00, Z0: 245.68-242.14j, gamma: 0.12+0.12j
 line S11 (0.005964206282295659-4.000722794175737e-05j)
 line S21 (0.9940357618651884-0.0002121393071130903j)
 line z0 (50+0j)
```
This should be results which are expected if a transmission line bloc was drawn on a simulator surrounded by two 50-ohm ports and you enforce line being computed with z0=Z0 and later renormalized to 50 ohm (media_zo) by the embedding thrus.
Strange the cascading of thru does not give the value of the ideal resistor here. Maybe I am still confused with all that.
Renormalization with the travelling waves
```Python
line = media.line(line_l, 'm', embed=False, z0=media.Z0)
line.renormalize(50, s_def='traveling')
```
Give the same:
```
line S11 (0.005964206282198348-4.0007228174301636e-05j)
line S21 (0.9940357618652855-0.00021213930668231555j)
line z0 (50+0j)
```
And same as
```
media_z0: None, line_z0: 50, Embed: True
 z0: 245.68-242.14j, Z0: 245.68-242.14j, gamma: 0.12+0.12j
 line S11 (-0.005964206282295548+4.000722794164635e-05j)
 line S21 (0.9940357618651884-0.0002121393071130903j)
 line z0 (245.68089902901662-242.14017760014647j)
```
Maybe you are using the code of your latest PR which give different results for Z0 from master ?
Could you test the below solution ?
```Python
media = Coaxial(freq, z0=50, Dint=0.5e-3, Dout=1.0e-3, sigma=1e3)
line = media.line(line_l, 'm', embed=True, z0=media.Z0)
```

mhuser, 2022-05-01:
I always feel writing ```z0=media.Z0``` is kind of a hack and reduce code clarity.

Ttl, 2022-05-01:
Okay, that works and now it makes more sense. So in summary `media.z0` is the port reference impedance, `media.Z0` is the characteristic impedance of the media, `line_z0` is the characteristic impedance of the line to create and `embed=True` embeds it to `media.z0`. I think the confusion is because by default the line method creates a line with characteristic impedance of `z0`, but I would have expected it to use the characteristic impedance of the media.

`media_z0: None, line_z0: 50, Embed: True` case gives close but slightly incorrect results because it embeds 50 ohm line to Z0, which is the opposite I would have wanted.

Overall this is really confusing. I really feel like the default behaviour should be to create line with the characteristic impedance of the media.

mhuser, 2022-05-01:
It is also quite typo-prone to have a single capital as difference between z0 and Z0.
Of course it is always complicated to change something in the case many are already using in their scripts.
In my wildest dreams I would see such behaviour of `Media` and `line`:

- `Media` would have `zp` or `z_ports` and `zc` or `z_characteristic` as parameters
- `Media.line` would have `zp` or `z_ports` parameter
- `zc` would be computed by `Media` from geometry or parameters at initialization
- If both `zp` are `None`, the line would have `zc` as `zp` else the line would be embedded such as `zc` renormalized to `zp`
- `zp` of `line` method (if not None) would have priority on `zp` of `Media` which would default to `None` or `50.`

It would not be possible any more to build lines with different characteristic impedances from a single `Media` (but why should we want that?)  And maybe we could keep `z0` for `zp` to keep consistent with `Network` and others. There would be no need to the `embed` parameter any more.

But that would be a lot of changes if `Media` is already widely used as such.

jhillairet, 2022-05-01:
Thank you @mhuser for having clarified all this.

> I think the confusion is because by default the line method creates a line with characteristic impedance of `z0`, but I would have expected it to use the characteristic impedance of the media.

> It is also quite typo-prone to have a single capital as difference between z0 and Z0.

I agree that this is confusing. 

> Of course it is always complicated to change something in the case many are already using in their scripts.

Changing the API could be made in two steps: next release will raise a warning, and release n+2 implements the change. I think it's better to keep the characteristic impedance as `z0`. But I'm ok to use `zp` or `z_ports` for the port impedance. Probably more clear.

Vinc0110, 2022-05-02:
> It is also quite typo-prone to have a single capital as difference between z0 and Z0.

That has already caused me at least 1 hour!

Also in terms of general coding style, having only `z0` with lowercase z would be better (see [PEP8](https://peps.python.org/pep-0008/#function-and-variable-names)).

mhuser, 2022-05-02:
> Changing the API could be made in two steps: next release will raise a warning, and release n+2 implements the change. I think it's better to keep the characteristic impedance as `z0`. But I'm ok to use `zp` or `z_ports` for the port impedance. Probably more clear.

Excellent idea ! Something like that into `Media` initialization ?
```Python
import warnings
warnings.warn('In a future version, `Media` characteristic impedance `Z0` will be renamed to `z0`.\n'
              '`Media` and `Media.line` port impedance `z0` will be renamed to `z_ports`.\n'
              'If `z_ports` is `None` lines will have `z0` port impedance ; else lines will be renormalized to `z_ports`.\n'
              '`Media.line` parameter `embed` will be deprecated.'
              FutureWarning, stacklevel = 2)
```

jhillairet, 2022-05-02:
Yes, something like that. But before implementing things, beware that many things in Media are defined like that, like `match`, etc

Ttl, 2022-05-02:
>It would not be possible any more to build lines with different characteristic impedances from a single Media (but why should we want that?)

It is useful if you just want to quickly have a line with a known z0 and don't want to spend time building a media for it. It does break the model though since in general characteristic impedance is frequency dependent and I feel like it shouldn't be as visible argument as it is now. Having `z0` in `line` is a potentially a source of modeling bugs since the generated line doesn't match the media (unless it's `DefinedAEpTandZ0`).

While the proposed API is much better than the current one, the backward compatibility breakage is quite big and would require lot of changes to existing scripts. `Z0 -> z0` change could be especially confusing since it changes the functionality of existing syntax.

I would personally like that `line` would create characteristic impedance line with 50 ohm ports by default to match what would be obtained by measuring a real line on VNA and also because working with complex port impedances is tricky.

mhuser, 2022-05-02:
Thanks a lot everyone, it's interesting to share our opinions on this matter.
Maybe a smaller step could be to keep `z0` and `Z0` for the time being but switch to the above behaviour with default `z0 = 50.` and renormalize the `Z0` line  by two `z0` thru, unless `z0 is None`.
Optionnaly we could add aliases attributes to the class like `z_ports` and `z_characteristic`, but it does sounds dangerous to also add them as a parameter of functions as someone would be at risks inputing different value for the parameter and its alias.
We could even have `Z0 = None` parameter of the `Media.line` to allow arbitrary impedance being embedded in `z0` (but discouraged)

mhuser, 2022-05-07:
After spending some time thinking about it, it comes to me that the confusion arise of two separated factor. Hopefully, these could be addressed separately:

1. names of parameters `z0` and `Z0` are confusion or typo prone (and not pep8 compliant for `Z0`)
2. behaviour and default of parameter `embed` of `mline` is not expected and this parameter is not required for `mline` use 

For 1. I feel there is not a 100% agreement to an API change, but the issue is dangerous in case you type your code very fast

For 2.  there is only 4 meaningfuls cases, with `z1` and `z2` being impedance arrays:

| | Media port impedance |  Line port impedance | Expected resulting network port impedance and normalization |
|---|---|---|---|
| a | `media.z0 == None` | `line.z0 == None` | `z0 = media.Z0`, no renormalization required |
| b | `media.z0 == None` | `line.z0 == z2` | `z0 = line.z0 = z2`, renormalization media.Z0 to z2 |
| c | `media.z0 == z1` | `line.z0 == None` | `z0 = media.z0 = z1`, renormalization media.Z0 to z1 |
| d | `media.z0 == z1` | `line.z0 == z2` | `z0 = line.z0 = z2`, renormalization media.Z0 to z2 |

Currently default is (a), the only way to achieve (c) is the `media.line(line_l, 'm', embed=True, z0=media.Z0)` hack. (b) and (d) give unexpected results whatever value of `embed` parameter.

I think we could enforce the a, b, c and d behaviour by fixing only `Media.line` method like this: We could make `embed` parameter having no more effect than emitting a deprecation warning if not default and use instead the values of `z0` to switch to the case we are.

jhillairet, 2022-05-07:
Nice summary

> I think we could enforce the a, b, c and d behaviour by fixing only `Media.line` method like this: We could make `embed` parameter having no more effect than emitting a deprecation warning if not default and use instead the values of `z0` to switch to the case we are.

seems nice to me.

mhuser, 2022-06-19:
#701  is intended as an exploratory work, to see how scikit-rf would be affected by the changes proposed in the table above. It will probably never be merged but enable others to chime in and provide feedback.

I am running into an issue which could be well described by an example:

In test_calibration.py. A waveguide media is defined here with the parameter `z0`.
https://github.com/scikit-rf/scikit-rf/blob/ac3aedfe0228615ef10da3e25212c7908059e6e0/skrf/calibration/tests/test_calibration.py#L25

Although a physical dimension `a` is specified, the physical characteristic impedance is never used (no renormalization or mismatch, the `Z0` is just like overwritten by parameter `z0`.

This media (or a lossless variant of it) is later used to define calibration standards like
https://github.com/scikit-rf/scikit-rf/blob/ac3aedfe0228615ef10da3e25212c7908059e6e0/skrf/calibration/tests/test_calibration.py#L389-L394

Here media `z0` is used for port and characteristic impedance. and the physical dimensions are only used to compute eventually gamma and losses.

Should we change the `line` behaviour only, it will not change the `match`, `short`, `open`, etc.

Only `thru` or `attenuator` or `delay_something` which relies on `line` would be affected, which I think will increase confusion. `match` has even a `z0_norm` boolean which is like `embed` for `line`!

Should we apply the table described in #651, I am afraid we would have to specify `Z0=None` at `line`, `thru`, `attenuator` and `delay_something` to keep test_calibration working, which seems ugly.

I am now starting to consider there is multiple use case for media:

- Users from finite element simulation, they are used to have port impedance being the characteristic impedance of the line, even if complex, they renormalize if required (see #717).
- Users from measurements or bloc simulation that await the mismatch in S11 caused by the impedance of measurement/simulation port being 50 ohm when they plot it, and eventually would wish discontinuities like open and short being implemented with the specific media models.
- Users from skrf that already have an extensive code base and are using any media to define a line with an arbitrary impedance (I did this, many times 😄 )

I am somehow lost on what move to do next. I think this is most confusing for new users, but any change will also impact these use cases. A patch could be to only fix the line `embed` behaviour to have it renormalize the line from `Z0` to `z0` properly, or to add `embed` at the media level so it switches behaviour for every method, but it would not really fix the confusion.

mhuser, 2022-06-20:
After a night on this, I have got this idea:

- The impedance behaviour (renormalized to a port impedance or physical characteristic impedance) could be set by a parameter in the media initialization e.g `z_port: Union[NumberLike, None] = None`. From this point, the impedance of all methods should be `z0`, either renormalized to `z_port` or not. This will enable to keep the call to various method such as `short`, `thru`, `line` clean and neat. Most of them relies on `match` and `line` behind the hood.
- Creating a line of arbitrary impedance could be done using a new method e.g. `Media.line_defined_z0` or, preferably, using ad-hoc media such as `DefinedGammaZ0`, or in the worst case we could keep a `z0` parameter in `line` and the other methods that overwrite media `z0`, but is not used by most users.

In such a way the `embed` parameter would disappear, and the choice is made once for all at the media initialization, which would ensure coherent behaviour.

It would hopefully both keep `z0` notation consistent with the rest of scikit-rf (`Z0` getter will still exists but remain unused for most users) and move the confusion at a single place in media initialization where we may take care it is better understood and carefully pick a default value.

What do you think of this? Is there a need be able to overwrite `z0` at arbitrary impedance in `Media` init? Any other ideas are welcome!

arsenovic, 2022-06-20:
i think this is a great idea. the confusion of `z0` and `Z0` are a big regret. `z0_port`  is unambiguous.  and i like the `line_defined_z0`.

jhillairet, 2022-06-20:
For, creating a line of arbitrary impedance, what if we cascade the operations:
```
my_media = MyMedia(physical_parameters and frequency)
my_line = my_media.line(d) # z0 is the charact impedance from the media
my_line = my_media.line(d).renormalize(z0)  # z0 becomes an arbitrary value
```

arsenovic, 2022-06-20:
that seems like it should work to me.   do you mean do this internally or expect user to use this method? 
i think  its nice  to have a single call  of `line_defined_z0` or `line_of_z0` or however its named .

jhillairet, 2022-06-20:
> do you mean do this internally or expect user to use this method?

I would expect users to use this method, as an explicit workflow, which keeps the original characteristic impedance value of the Media untouched, being eventually complex.

In fact, I'm a little biased as I'm more a full-wave user than a circuit simulator users, which is why I liked the comment made by @ZiadHatab in https://github.com/scikit-rf/scikit-rf/issues/717#issuecomment-1159564970. When one exports a Touchstone with HFSS, the interface asks if it should be converted to 50 ohm (default) or if it should use the characteristic impedance as port impedance. So it requires a deliberate choice by the user, and the simulated structure behind keeps its own characteristic impedance, as physical evaluated.
 
Of course, an alias like `line_...z0` could be created to speed up this workflow.

> Any time I have used the `.line` method I have wanted a line with 50 ohm port impedance and I don't think that is very unusual opinion.

I totally understand as well. I'm even still not sure if using complex characteristic impedance as port impedance as some practical interest... so far, I never received clear answer on that topic.

mhuser, 2022-06-20:
What about
```
m = Media(f, ...physical..., z0_port=50) # default z0_port = 50
m.match() # Z0 renorm to z0_port
m.line(d, unit) # Z0 renorm to z0_port
m.line(d, unit, defined_z0=48) # z0 = 48 without renorm

m = Media(f, ...physical..., z0_port=None) 
m.match() # Z0 without renorm
m.line(d, unit) # Z0 without renorm

m = Media(f, ...physical..., defined_z0=48) 
m.match() # z0 = 48 without renorm
m.line(d, unit) # z0 = 48 without 
```

ZiadHatab, 2022-06-20:
Since I was mentioned, let me share my opinion on this.

@Ttl , I understand the convenience of having the S-parameters referenced to a constant impedance, especially with smith charts. However, I still don't see why you want to define the port impedance when creating the media. I think the Media object should only take physical parameters as input (dimensions, dielectric constant...). If using renormalize is inconvenient, then just add a parameter in the line() method to renormalize to a different impedance. For example:
```python
m = Media(f, ...physical...)
matched_line = m.line(l, 'm')
mismatched_line = m.line(l, 'm', z0_port=z0_new) # the s-parameters are internally renormalized to z0_new
```
 This way you only introduce a single new parameter `z0_port`, without needing to modify the Media class. Though, I still think `z0` parameter should be removed from the main Media class ;)

mhuser, 2022-06-20:
I understand these points of views that have proper arguments, but it is also correct that when simulating measurements (e.g. optimisation of model or building artificially calibration standard) it is handy not having to renormalize or write the z0_port in each call to line, thru, open, short, etc.

At least we all agree we have to change something 😅

mhuser, 2022-06-20:
I think we also usually only wish only one of the behaviour when we use the media in a script: raw Z0, measured-like, defined impedance but not a mix of these

mhuser, 2022-06-20:
Following solutions could be explored: parameter in initialisation, a class MeasuredMedia extending a raw Media class, etc.

ZiadHatab, 2022-06-20:
Ok, if most people use the Media object with a constant impedance, then I will go with your suggestion, with a slight modification:
```python
m = Media(f, ...physical..., z0_port=50) # default z0_port = None (I think default should be the characteristic impedance)
m.line(d, unit) # renorm to z0_port
m.line(d, unit, z0_port=48) # renorm to another impedance 48
```
Then, if we want the network renormed to the characteristic impedance, we just define it either from the beginning or in the line method:
```python
# case 1
m = Media(f, ...physical...)
m.line(d, unit) # renorm to characteristic impedance

# case 2
m = Media(f, ...physical..., z0_port=50)
m.line(d, unit, z0_port=m.Z0) # renorm to characteristic impedance
```

I don't think you can please everyone. Just go with the most logical answer that maximizes the convenience of most users.

mhuser, 2022-06-20:
Thanks @ZiadHatab , this looks good to me!
Let's wait the feedback of the others courageous `Media` users.

jhillairet, 2022-06-21:
OK too. So `Media.z0` would be an immutable object, defined by the physical properties and frequency of the Media, while `Media.z0_port` is the port reference impedance passed to the Networks created from the Media, am I right?

mhuser, 2022-06-21:
> OK too. So `Media.z0` would be an immutable object, defined by the physical properties and frequency of the Media, while `Media.z0_port` is the port reference impedance passed to the Networks created from the Media (and could be modified), am I right?

I think yes. `z0` and `Z0` could be aliases to the same thing.
Overvritting `z0` at line creation could still be tolerated to create arbitrary impedance things, but it would not affect media `z0`.

mhuser, 2022-06-21:
A small summary (and to be sure of my understanding):
```Python
#In all cases, m.z0 == m.Z0

# Media with physical z0 impedance (default)
m = Media(f, ...physical...) 
m.match() # z0 without renorm
m.line(d, unit) # z0 without renorm
m.line(d, unit, z0=48) # arbitrary z0 of 48 ohm (does not affect media)
m.line(d, unit, z0_port=50) # renorm to 50 ohm (does not affect media)
m.line(d, unit, z0=48, z0_port=50) # arbitrary z0 = 48 ohm renorm to 50 ohm or raise ValueError ?

# Media renormalized to 50 ohm port impedance
m = Media(f, ...physical..., z0_port=50) 
m.match() # z0 renorm to 50 ohm
m.line(d, unit) # z0 renorm to 50 ohm
m.line(d, unit, z0=48) # arbitrary z0 of 48 ohm (does not affect media)
m.line(d, unit, z0_port=m.z0) # z0 renorm to 50 ohm renorm to z0 (does not affect media)
m.line(d, unit, z0_port=None) # z0 without renorm
m.line(d, unit, z0=48, z0_port=50) # arbitrary z0 = 48 ohm renorm to 50 ohm or raise ValueError ?
```
I think we should not implement this case (use DefinedGammaZ0 or DefinedAEpTanDZ0 instead), but of course I am open to reconsider:
```Python
m = Media(f, ...physical..., defined_z0=48) 
m.match() # z0 = 48 ohm without renorm
m.line(d, unit) # z0 = 48 ohm without renorm 
```

ZiadHatab, 2022-06-21:
@mhuser 

looks good. What do mean in the line before last?
```python
m.line(d, unit, z0_port=None) # z0 without renorm
```
you mean that there will be no additional renormalization? The same case as `m.line(d, unit)` ?

mhuser, 2022-06-21:
> you mean that there will be no additional renormalization? The same case as `m.line(d, unit)` ?

Yes, same as `m.line(d, unit)` with no `z0_port` in media initialization

arsenovic, 2022-06-21:
@jhillairet, i think allowing for complex port impedance is important. i also like a sloppier. concise workflow, but that is me. 
@Ttl , i think you are correct in that most people want 50ohm ports. should this be the default? i am unsure. (i use rf.air50 as a generic media)
@mhuser, your suggested usage makes sense to me. it will be a challenge to teach people all the different uses. but i think this is a because  concept  of impedance is confusing

 to me the `port_z0`'s are a way to implicitly generate impedance steps, and somtimes you want this , other times you dont.  conceptually  i like to think of `z0_renorm` more as `z0_embed`, ie the circuit is placed in a fixture of different z0.  obviously this could be done with cascading impedance steps also. many times  i like to simulate some  mismatches quickly  using normalized impedance in the line constructor (which was a feature taken from puff [really old circuit simulator from caltech]). just some thoughts about z0's

jhillairet, 2022-06-21:
and we definitely need drawings ;)

mhuser, 2022-06-28:
Ok, I guess it is [waveguide-to-coax-transitions](https://www.microwaves101.com/encyclopedias/waveguide-to-coax-transitions), but how does this work from the media z0, and VNA z0 point of view, e.g. to get measurement into a touchstone file?

ZiadHatab, 2022-06-28:
@mhuser 

I think you are confusing yourself here. The waveguide and the transitions are separate devices. The transition is a wideband mode/impedance transformer. The waveguide is just a rectangular tube. 

You are right about it makes no sense to embed waveguide S-parameters in 50ohm impedance, because nobody does that. From a measurement point-of-view, and if you care about accuracy, you will most likely calibrate using waveguide standards. Therefore, the reference impedance after calibration is the characteristic impedance of the waveguide itself. That is why you don't see companies like [Maury Microwave](https://www.maurymw.com/) selling their waveguide cal kit by specifying the impedance, because, most often, every waveguide structure has a different impedance (which you already observed yourself).

Going back to your concern. No, just performing an impedance transformation to 50ohm on a waveguide media makes no sense (unless this is intentionally wanted). Making the port 50 ohm without actually preforming a transformation is the easiest thing you could do, because this is equivalent to attaching the waveguide to an ideal 50ohm coaxial-to-waveguide transition (no loss; no reflection). 

You could introduce an option to wrap the waveguide around some transition to TEM mode (e.g., coaxial). I have no knowledge how people design such transitions. You might find some general equations that approximate such transitions, but most often every company has its own design. Think about this as PCB SMA connecters... do you think all companies produce same connectors? 

I'm no waveguide expert, but let me know if something is still not clear.

mhuser, 2022-06-29:
@ZiadHatab 
Thank you so much for your answer!

I think it is much more clear for me now, thanks to the explanations. Characteristic impedance of the waveguides is a complex thing, but once the calibration has been done with the waveguides standards, the reference is its own impedance (let say it is like normalized to 1.0).

If the user want a touchstone in ohm, the VNA will multiply 1.0 by its z0, but this has no physical meaning in term of the waveguide impedance and the coaxial-to-waveguide transition is "transparent" thanks to calibration.

Maybe the confusion in skrf media even originated with this fact. Later the non-waveguide media have somehow inherited this behaviour.

I could add a parameter `z0_transition = None` **only in waveguides media**. If not None, it would be taken as `z0`, acting like an ideal transition (no renormalization, just overwrites and shadows the waveguide impedance).

mhuser, 2022-11-09:
[image]

Some draft for documentation to explain differences between simulated/measured media and with/without waveguide-to-coax transition. The PR is very slow paced but not fully stalled 😅

mhuser, 2022-11-10:
[image]
Here it is!

jhillairet, 2022-11-10:
Super nice drawings !

Eventually, if you want to share it, I've already put a .svg file in `doc/source` for the network symbols. So, this place could accept new ones !

mhuser, 2022-11-10:
> Eventually, if you want to share it, I've already put a .svg file in `doc/source` for the network symbols. So, this place could accept new ones !

Sounds like a great place for it!
I will refine it a bit and use it to expand the media notebook in my next sprint 😄

mhuser, 2023-04-10:
I have started a sprint on this topic.
The media notebook is being updated in a separate PR to explain the different cases to get EM simulation or measurement-like s-parameters from media. 
[Media notebook preview](https://scikit-rf--888.org.readthedocs.build/en/888/tutorials/Media.html)

mhuser, 2023-04-15:
I have now started tedious the process of checking each notebook for media usage.
This means that the PR #727  is for the first time almost review-ready!
@jhillairet your taper example agreement is even better of a bit with ANSYS Circuit now:
Before
[image]
After `z0_port=50`
[image]

jhillairet, 2023-04-16:
Ho great! I never understood from where the difference could come.!

mhuser, 2023-04-17:
Ok, #727 should be review ready 🎉 
There is so many files modified I am a bit worried. But if everything does not change simultaneously, a lot of tests would not pass. Also, the notebook diffs are not very meaningful. The best is to look at the [readthedocs](https://scikit-rf--727.org.readthedocs.build/en/727/) of the PR.
All comments are of course welcome!

PR #727 (2023-05-07): Refactor media to simplify line creation
This PR is the second attempt to fix #651 

Ready for review.

Todo
===

- [x] Refactor all `media` for `z0_port`, `z0` scheme
- [x] Refactor waveguides `media` for `z0_override`, `z0` scheme
- [x] Make all tests pass and refine
- [x] Make all notebooks build
- [x] Update notebooks (check contents)
- [x] Update documentation
- [x] Extend documentation (e.g. by passing `z0` to `Media.line())`
closed 2023-05-07 by jhillairet