scikit-rf/scikit-rf #652: Writing touchstone with complex port impedances writes incorrect reference impedance

Ttl, 2022-05-02:
`write_touchstone` writes absolute value of `z0` of the first port first frequency point as the reference impedance, but doesn't seem to normalize the written S-parameters if other ports or frequency points are different.

Example:
```python
import numpy as np
import skrf
from skrf.media import DefinedAEpTandZ0

freq = skrf.Frequency(100, 200, 3, unit='MHz')
line_l = 0.1

media = DefinedAEpTandZ0(freq, A=0.1, ep_r=1, Z0=[50-50j, 50-25j, 50-15j], z0=50)
line_z0port = media.line(line_l, 'm', z0=media.Z0)

line_z0port.write_touchstone('line.s2p')
line_z0port_s2p = skrf.Network('line.s2p')

print('Written z0', line_z0port.z0)
print('Read z0', line_z0port_s2p.z0)

assert (line_z0port.s == line_z0port_s2p.s).all()
```

```
Written z0 [[50.-50.j 50.-50.j]
 [50.-25.j 50.-25.j]
 [50.-15.j 50.-15.j]]
Read z0 [[70.71067812+0.j 70.71067812+0.j]
 [70.71067812+0.j 70.71067812+0.j]
 [70.71067812+0.j 70.71067812+0.j]]
```

There's a question what should happen in this case? For one it seems that officially touchstone format doesn't support complex reference (page 7 in specification: http://ibis.org/touchstone_ver2.0/touchstone_ver2_0.pdf) although skrf can read them. Should reference always be 50 unless specified otherwise? There is `write_z0` option, but that's not standard.

mhuser, 2022-05-03:
Intuitively, if z0 in touchstone is not the same as network and `write_z0 == False`, shouldn't the output be renormalized to the real-valued reference impedance of touchstone file, which can be the lowest frequency absolute magnitude of first port impedance?
Obviously it is not he case here.
In network.py line 2158:
```Python
            # [HZ/KHZ/MHZ/GHZ] [S/Y/Z/G/H] [MA/DB/RI] [R n]
            output.write('# {} S {} R {} \n'.format(self.frequency.unit, form, str(abs(self.z0[0, 0]))))`
```
But there seems to be nothing done if any element of z0 is different of abs(self.z0[0, 0]), which lead to a slightly different network.

jhillairet, 2022-06-11:
What should we do about  this issue?

Ttl, 2022-06-11:
fixed in #672

closed 2022-06-11 by Ttl