scikit-rf/scikit-rf #653: Error using rf.io.general.network_2_spreadsheet function

ArturoLlorente, 2022-05-04:
Hi, I am trying to create a spreadsheet of a given .snp file from a Network using the **rf.io.general.network_2_spreadsheet()** function.

I  python 3.9.12 I have the following error:
[image]

In case of python 3.10.4, the error is not the same:
[image]

Can someone help me with this topic?
Thanks in advance

jhillairet, 2022-05-04:
The following ways should work:

`net.write_spreadsheet('test.xlsx')  # object oriented way`

or

`rf.network_2_spreadsheet(net, 'test.xlsx')  # function way`

jhillairet, 2022-05-04:
To complement: the functions in `skrf/io/general.py` are directly imported in the base scope.