scikit-rf/scikit-rf #656: Adopt 8-Term (Error Box) Calibration for 1.5-port VNAs

biergaizi, 2022-05-08:
Many useful calibration algorithms are based on the 8-term Error Box model, such as the *Unknown Thru* calibration or the *TRL* calibration. If these algorithms can be applied to a 1.5-port VNA as well, they will have great practical values. But currently in scikit-rf, the only calibration available for 1.5-port VNA is the Two Port One Path, which is a variant of the 12-term SOLT calibration.

Is it possible to adopt these 8-term Error Box models to 1.5-port VNAs? If my understanding is correct:

1. The 8-term error model is basically mathematical equivalent to the 12-term error model, with the exception of the switch term. 
2. The switch term represents the physical port match at the VNA, and it's highly stable and it's independent from the external cabling.

If this is the case, is it possible to find the switch term via a full Two Port One Path SOLT calibration first, then use a 8-term Error Model and apply the same trick of copying the forward error coefficients to the backward direction, and finally apply the switch term?

biergaizi, 2022-05-12:
This is my first attempt on solving this problem.

It seems to be an impractical (or at least not elegant) solution to copy all 8-term calibration, 2-port calibration classes and redefine them just for a 1.5-port VNA. Instead, I think we can have more flexibility if an arbitrary 2-port, 2-path calibration class can be automatically adopted for a 2-port, 1-path VNA dynamically, on the fly. I think this is possible using Pythos's metaprogramming capabilities like metaclasses, but for simplicity, I only used a function here to generate classes.

```
import pathlib
import skrf
from skrf.network import four_oneports_2_twoport
from skrf.calibration import TwoPortOnePath, UnknownThru
from skrf.media import DefinedGammaZ0

def ThreeReceiverCal(calibration):
    """
    Generate a new calibration class dynamically suitable for use with a
    1.5-port VNA from an existing 2-port calibration class.

    This is implemented in the following way. First, before the calibration
    measurements are passed to the underlying class, 2-port, 2-path measurements
    are forged by copying all forward S-parameters (S11, S21) to the backward
    direction (S22, S12).

    Next, the `apply_cal` function is modified to take two sets of measurements,
    one in the forward direction, and another one in the backward direction
    with the DUT physically flipped. These two sets of measurements is again
    combined, forging full 2-port, 2-path measurements before it's passed into
    the underlying calibration class.
    """
    def single_one_half_ports_to_two_ports(network):
        s11 = network.s11
        s21 = network.s21
        s22 = s11
        s12 = s21
        return four_oneports_2_twoport(s11, s12, s21, s22)

    def two_one_half_ports_to_two_ports(dut_forward, dut_backward):
        s11 = dut_forward.s11
        s21 = dut_forward.s21
        s22 = dut_backward.s11
        s12 = dut_backward.s21
        return four_oneports_2_twoport(s11, s12, s21, s22)

    class ThreeReceiverCal(calibration):
        def __init__(self, measured, ideals, *args, **kwargs):
            for idx, network in enumerate(measured):
                measured[idx] = single_one_half_ports_to_two_ports(network)
            super().__init__(measured, ideals, *args, **kwargs)

        def apply_cal(self, ntwk_tuple):
            if not isinstance(ntwk_tuple, tuple):
                raise NotImplementedError
            if not len(ntwk_tuple) == 2:
                raise ValueError

            dut = two_one_half_ports_to_two_ports(ntwk_tuple[0], ntwk_tuple[1])
            return super().apply_cal(dut)

    return ThreeReceiverCal

def main(solt_dir, unknown_thru_dir, measurement_dir, output_dir):
    # Perform Two Port One Path 12-term calibration to extract the switch term
    solt_short_std = skrf.Network(solt_dir / "short.s2p")
    solt_open_std  = skrf.Network(solt_dir / "open.s2p")
    solt_load_std  = skrf.Network(solt_dir / "load.s2p")
    solt_thru_std  = skrf.Network(solt_dir / "thru.s2p")

    frequency = solt_open_std.frequency
    line = DefinedGammaZ0(frequency=frequency)
    ideals = [
        line.short(nports=2), line.open(nports=2),
        line.match(nports=2), line.thru()
    ]
    measured = [
        solt_short_std, solt_open_std,
        solt_load_std, solt_thru_std
    ]
    cal = TwoPortOnePath(measured=measured, ideals=ideals, n_thrus=1)
    switch_terms = [
        cal.coefs_8term_ntwks["forward switch term"],
        cal.coefs_8term_ntwks["reverse switch term"]
    ]
    assert switch_terms[0] == switch_terms[1]

    # Perform Unknown Thru 8-term calibration, with the switch term added
    short_std = skrf.Network(unknown_thru_dir / "short.s2p")
    open_std  = skrf.Network(unknown_thru_dir / "open.s2p")
    load_std  = skrf.Network(unknown_thru_dir / "load.s2p")
    thru_std  = skrf.Network(unknown_thru_dir / "thru.s2p")

    frequency = open_std.frequency
    line = DefinedGammaZ0(frequency=frequency)
    ideals = [
        line.short(nports=2), line.open(nports=2),
        line.match(nports=2), line.line(d=1, unit='mm')
    ]
    measured = [
        short_std, open_std,
        load_std, thru_std
    ]
    cal_class = ThreeReceiverCal(UnknownThru)
    cal = cal_class(measured=measured, ideals=ideals, n_thrus=1, switch_terms=switch_terms)

    dut_forward = skrf.Network(measurement_dir / "coupler_forward.s2p")
    dut_backward = skrf.Network(measurement_dir / "coupler_backward.s2p")
    dut = cal.apply_cal((dut_forward, dut_backward))
    dut.write_touchstone(output_dir / "coupler.s2p")

solt_dir = pathlib.Path("./cal/solt/")
unknown_thru_dir = pathlib.Path("./cal/unknown-thru/")
measurement_dir = pathlib.Path("./raw/")
output_dir = pathlib.Path("./data/")
main(solt_dir, unknown_thru_dir, measurement_dir, output_dir)
```

The code is untested, currently I'm not sure if the calibration is really functional. Nevertheless, I think it's self-explanatory for my ideas...

arsenovic, 2022-05-12:
i am pretty sure that  not having full recievers makes it not posibble to perfectly cal. you will have an approximate call .  you would need to setup a test suit to ensure its working probably by making some simplificiations.

biergaizi, 2022-05-12:
> @arsenovic: i am pretty sure that not having 4 receivers makes it not possible to perfectly cal. (i think keysight had a TRL* algorithm with tried this and failed, but i cant remember exactly) . so you will have an approximate call.

I don't know much about network theory or calibration, so correct me if I get anything wrong.

But... My impression is that the TRL* algorithm's biggest deficiency is neglecting the switch terms (or *load reflection errors*). On the other hand, if you can determine the switch terms in a SOLT calibration before doing the second-tier TRL calibration (and applying the previously-obtained switch terms on top of the TRL result), this error can be removed, and the calibration should be equivalent (?) to an ideal one.

In the NIST paper [*Two-Tier Multiline TRL for Calibration of Low-Cost Network Analyzers*](https://zero.sci-hub.se/3094/7de434f51fc2380b56fe59015d7ec8cd/jargon1995.pdf), the authors said,

> The three-sampler architecture precludes straightforward TRL because it provides no direct mechanism for assessing the load reflection errors due to nonideal terminations on the port opposite the one to which the incident signal is applied.
>
> Since error occurs only if the switchablesource and load have nonidentical reflection coefficients, the effect is generally known as "switching error." Users of three-sampler VNA find themselves forced to use lumped-element methods such as OSLT (open-short-load-thr). Recently, an approximate version of TRL intended for use on a three-sampler VNA has been introduced commercially [Note: this is a reference to Keysight's TRL*]. However, it simply neglects the load reflection errors and therefore may be unsuitably inaccurate [7,8].
>
> Another way to overcome the limitations of a three-sampler system is to apply a two-tier calibration. In this case, the first tier may be an OSLT calibration in which only the load reflection terms will affect the measurements, since the second tier calibration will effectively overwrite all of the other calibration coefficients. Since the switch terms are determined in the first tier, the second tier can use TRL or a derivative. Two-tier calibrations in which the second tier is a TRL-based method have been used for many years [9]. More recently, this approach has been applied specifically to three-sampler VNAs [8,10]. The former documents that the accuracy of this method may be superior to TRL without load reflection correction.

Conclusion:

> Multiline TRL can be easily implemented on a three-sampler VNA using external software. The accuracy can be nearly as good as a four-sampler VNA. Although an additional first-tier calibration using coaxial standards is required, this first tier need not be repeated frequently as long as the calibration factors are retained in the VNA. Alternatively, the load reflection coefficients could be extracted from the first-tier calibration and stored externally for later use.

This is basically what I'm trying do to here, as seen in the code snippet I posted in the previous comment.

Also, this statement is repeated in the NIST paper [*Formulations of the Basic Vector Network Analyzer Error Model including Switch-Terms*](https://moscow.sci-hub.se/1854/e8dffcc675139eb08a3c02c07b1eba6a/marks1997.pdf), the author made this statement.

> Equations (30) and (31) [the two equations for calculating the switch term from the 12-term error coefficients] have been used [9] to demonstratethat a TRL calibration could be implemented on a three-samplerVNA once r, and r R were determined from a separate calibration. These switch terms were shown to be stable with respect to time, as we would expect of an electronic switch.

ZiadHatab, 2025-02-02:
Just a response for future users reading this thread.

If you don't have a full four-receiver VNA, you won't be able to measure the switch terms. Some VNAs multiplex the a-wave receiver to be used at the terminated port while the other port is sourcing, enabling switch term measurements.

To clarify about switch terms: we only need them when we define S-parameters as ratios, i.e., S = b/a. In multiport systems, S-parameters are defined as matrix multiplication of the wave-parameters:
[image]

By extracting a11 and a22 as diagonal matrices and inverting to the left side, we can simplify the equation in terms of ratios:
[image]

Where the ratio definitions are:
[image]

The Gamma_ij terms represent the switch terms. The ratio S-parameters can be measured with a 3-receiver VNA (in both directions).

For more detailed information about switch terms and their computation, even on 3-receiver VNAs, visit my blog:
https://ziadhatab.github.io/posts/vna-switch-terms/

I've implemented the function for computing switch terms on 3-receiver VNAs in scikit-rf. See the `rf.compute_switch_terms()` function in the Calibration class:
https://scikit-rf.readthedocs.io/en/v1.3.0/_modules/skrf/calibration/calibration.html#Calibration

To learn more about Two-port/One-path Calibration, check my other post:
https://ziadhatab.github.io/posts/two-port-one-path-calibration/

biergaizi, 2025-05-01:
In the previous discussion, it's assumed that the 8-term calibration family including Unknown Thru and TRL can't be used for Three Receiver (1.5-port VNA) because we can't determine the switch term. But my further investigation found it's the actual problem is much greater than that. Even if we know the switch term, the 8-term calibration classes still can't be used directly (the SOLT trick of copying forward and reverse coefficient doesn't work for either EightTerm, Unknown Thru, or TRL).

Does any experts out there have any idea about using EightTerm, Unknown Thru, or TRL with Three Receiver VNAs? According to NIST, it's definitely possible as a second-tier calibration, but how can it be implemented in scikit-rf remains unclear. I suspect the existing classes can't be reused, and we must work from the equations and rederived the algorithm from scratch?

In the following code, I created a numerical simulation of VNA calibration with completely random error boxes to test which calibration works and which doesn't. 

```python
import numpy as np
import skrf
import skrf.calibration
from skrf.media import Coaxial
from matplotlib import pyplot as plt
skrf.stylely()

freq = skrf.F(1, 100, 100, unit='GHz')
nports = 2
coax = Coaxial(freq, z0_port=50, Dint=0.44e-3, Dout=1.0e-3, sigma=1e8)

class UncalibratedNetworkAnalyzer():
    """
    Simulate a Vector Network Analyzer with a random n-port error box
    connected between all ideal VNA ports and DUT ports. The errors are
    so large that they make all measurements results meaningless. But
    they're linear, so a suitable calibration algorithm should be able
    to remowe them exactly - as a severe stress test.

    - switch_errors: Many error models (like the 8-term family) assume
    the VNA itself has near-perfect port termination, as a result the
    calibration is inexact. This termination error is known as the switch
    term, as the internal port termination is implemented by an RF switch.
    As a result, calibration may be less accurate than the traditional
    12-term SOLT model, which models this error implicitly. Since the
    switch term is a stable error that can be pre-determined, Many
    algorithms apply it as an additional correction for exact calibration.

    - onepath: Low-cost VNAs can only measure S11 and S21, without a
    dedicated receiver at Port 2 to measure the reverse direction (S12
    and S22).
    """
    def __init__(self, freq, nports, switch_errors=True, onepath=False):
        self.freq = freq
        self.nports = nports
        self.create_error_data(freq, nports, switch_errors)
        self.onepath = onepath

    def create_error_data(self, freq, nports, switch_term_error=True):
        # Create random error network
        # [0, nports) ports will connect to an ideal VNA
        # and the [nports, 2 * nports) ports will connect to the DUT.
        Z = coax.random(n_ports=2*nports, name='Z')

        # Zero leakage terms in the error network.
        port_type = lambda n: 'VNA' if n < nports else 'DUT'
        port_number = lambda n: n if n < nports else n - nports
        for i in range(2*nports):
            for j in range(i+1, 2*nports):
                # No connection between different VNA/DUT ports.
                # No connection between VNA and DUT ports with different number.
                if port_type(i) == port_type(j) or port_number(i) != port_number(j):
                    Z.s[:,i,j] = 0
                    Z.s[:,j,i] = 0

        # VNA switch terms. This is the termination impedance of each port
        # when the source is not connected to that port.
        gammas = []
        for i in range(nports):
            if switch_term_error:
                gammas.append(coax.random(n_ports=1, name=f'gamma_{i}'))
            else:
                gammas.append(coax.match())
[... 320 lines cut ...]
        gammas = list(reversed(vna.gammas))
        cal = skrf.calibration.EightTerm(ideals=ideal_list, measured=meas_list, switch_terms=gammas)

    cal.run()

    coefs = cal.coefs_12term
    out_coefs = coefs.copy()
    for k in coefs:
        if k.startswith("forward"):
            k_out = k.replace("forward", "reverse")
            out_coefs[k_out] = coefs[k]
    cal.coefs = skrf.calibration.convert_12term_2_8term(out_coefs)

    eight_term_coefs = skrf.calibration.convert_12term_2_8term(out_coefs)
    out_coefs.update({l: eight_term_coefs[l] for l in \
            ['forward switch term','reverse switch term','k'] })
    cal.coefs = out_coefs

    dut.s[:,0,1] = dut.s[:,1,0]
    dut.s[:,1,1] = dut.s[:,0,0]

    return cal.apply_cal(dut)
```

The calibration result is still random noise.

biergaizi, 2025-05-01:
In the previous discussion, it's assumed that the 8-term calibration family including Unknown Thru and TRL can't be used for Three Receiver (1.5-port VNA) because we can't determine the switch term. But my further investigation found it's the actual problem is much greater than that. Even if we know the switch term, the 8-term calibration classes still can't be used directly (the SOLT trick of copying forward and reverse coefficient doesn't work for either EightTerm, Unknown Thru, or TRL).

Does any experts out there have any idea about using EightTerm, Unknown Thru, or TRL with Three Receiver VNAs? According to NIST, it's definitely possible as a second-tier calibration, but how can it be implemented in scikit-rf remains unclear. I suspect the existing classes can't be reused, and we must work from the equations and rederived the algorithm from scratch?

In the following code, I created a numerical simulation of VNA calibration with completely random error boxes to test which calibration works and which doesn't. 

```python
import numpy as np
import skrf
import skrf.calibration
from skrf.media import Coaxial
from matplotlib import pyplot as plt
skrf.stylely()

freq = skrf.F(1, 100, 100, unit='GHz')
nports = 2
coax = Coaxial(freq, z0_port=50, Dint=0.44e-3, Dout=1.0e-3, sigma=1e8)

class UncalibratedNetworkAnalyzer():
    """
    Simulate a Vector Network Analyzer with a random n-port error box
    connected between all ideal VNA ports and DUT ports. The errors are
    so large that they make all measurements results meaningless. But
    they're linear, so a suitable calibration algorithm should be able
    to remowe them exactly - as a severe stress test.

    - switch_errors: Many error models (like the 8-term family) assume
    the VNA itself has near-perfect port termination, as a result the
    calibration is inexact. This termination error is known as the switch
    term, as the internal port termination is implemented by an RF switch.
    As a result, calibration may be less accurate than the traditional
    12-term SOLT model, which models this error implicitly. Since the
    switch term is a stable error that can be pre-determined, Many
    algorithms apply it as an additional correction for exact calibration.

    - onepath: Low-cost VNAs can only measure S11 and S21, without a
    dedicated receiver at Port 2 to measure the reverse direction (S12
    and S22).
    """
    def __init__(self, freq, nports, switch_errors=True, onepath=False):
        self.freq = freq
        self.nports = nports
        self.create_error_data(freq, nports, switch_errors)
        self.onepath = onepath

    def create_error_data(self, freq, nports, switch_term_error=True):
        # Create random error network
        # [0, nports) ports will connect to an ideal VNA
        # and the [nports, 2 * nports) ports will connect to the DUT.
        Z = coax.random(n_ports=2*nports, name='Z')

        # Zero leakage terms in the error network.
        port_type = lambda n: 'VNA' if n < nports else 'DUT'
        port_number = lambda n: n if n < nports else n - nports
        for i in range(2*nports):
            for j in range(i+1, 2*nports):
                # No connection between different VNA/DUT ports.
                # No connection between VNA and DUT ports with different number.
                if port_type(i) == port_type(j) or port_number(i) != port_number(j):
                    Z.s[:,i,j] = 0
                    Z.s[:,j,i] = 0

        # VNA switch terms. This is the termination impedance of each port
        # when the source is not connected to that port.
        gammas = []
        for i in range(nports):
            if switch_term_error:
                gammas.append(coax.random(n_ports=1, name=f'gamma_{i}'))
            else:
                gammas.append(coax.match())
[... 391 lines cut ...]
  File "/home/user/code/skrf-tutorials/calibration.py", line 370, in test_3receiver
    dut_corrected_trl = trl_calibration(vna, dut_meas, known_switch_term=True)
  File "/home/user/code/skrf-tutorials/calibration.py", line 274, in trl_calibration
    cal.run()
  File "/home/user/.local/lib/python3.10/site-packages/skrf/calibration/calibration.py", line 2583, in run
    self.ideals[k] = determine_line(m_ut[0], m_ut[k], line_approx) # find line
  File "/home/user/.local/lib/python3.10/site-packages/skrf/calibration/calibration.py", line 6359, in determine_line
    C = thru_m.inv**line_m
  File "/home/user/.local/lib/python3.10/site-packages/skrf/network.py", line 1416, in inv
    out.s = inv(self.s)
  File "/home/user/.local/lib/python3.10/site-packages/skrf/network.py", line 8109, in inv
    tinv = np.linalg.inv(t)
  File "<__array_function__ internals>", line 200, in inv
  File "/home/user/.local/lib/python3.10/site-packages/numpy/linalg/linalg.py", line 538, in inv
    ainv = _umath_linalg.inv(a, signature=signature, extobj=extobj)
  File "/home/user/.local/lib/python3.10/site-packages/numpy/linalg/linalg.py", line 89, in _raise_linalgerror_singular
    raise LinAlgError("Singular matrix")
numpy.linalg.LinAlgError: Singular matrix
```

## Question

What is the correct implementation of `onepath_eightterm_calibration()`, `onepath_unknown_thru_calibration()` or `onepath_trl_calibration()`? Is the generic Eight Term or its Unknown Thru variant even possible with 3-receiver VNAs? A clarification would be appreciated. It was previously assumed that it's not possible due to the missing switch term, but it seems impossible not because of that reason, but because we can't determine the right error box if we don't have a transmitter at port 2?

Also, as documented by NIST, `onepath_trl_calibration()` is definitely possible to implement, but perhaps a separation implementation must be derived from the original equations in the paper.

miek, 2025-05-01:
One thing I notice that might be causing some confusion is the use of the terms "1.5 port/one-path" and "three-receiver" interchangeably, but they're not always the same thing. It's quite common to have a two-path three-receiver VNA that can measure S12/S22, but can't directly measure switch terms. So, when literature mentions three-receiver VNAs, it's likely to be referring to a two-path configuration and indeed, looking at Figure 4 in that NIST paper, that's what they're describing.

ZiadHatab, 2025-05-03:
@biergaizi 

what NIST mentioned isn't a one-path measurement, it is two-path, but not all 4-waves are measured at once, instead only 3-waves are measured in single sweep (in both directions, you get 6 waves).

This is the VNA structure they are talking about (don't mind the flipped coupler direction 😅):
[image]

If you want a single-path and two-port, then you don't have the switch to flip to reverse measurement. Your VNA structure looks as below (again don't mind the flipped coupler):
[image]

If you want to learn more about single-path/two-port calibration, I have a blog on that: https://ziadhatab.github.io/posts/two-port-one-path-calibration/

You can make it work as two-path if you manually flip the standards and measure in reverse. In terms how to compute switch terms for 3-receivers design, you can either do SOLT or using several reciprocal devices: https://ziadhatab.github.io/posts/vna-switch-terms/

hope this helps!

biergaizi, 2025-05-08:
> One thing I notice that might be causing some confusion is the use of the terms "1.5 port/one-path" and "three-receiver" interchangeably, but they're not always the same thing. It's quite common to [...]

Thanks for the tip, this is indeed the source of confusion.

> If you want to learn more about single-path/two-port calibration, I have a blog on that: https://ziadhatab.github.io/posts/two-port-one-path-calibration/

I think I've understood it now.

1. The problem is simple: To calibrate a 2-port DUT with 4 S-parameters, we must have 4 equations to solve 4 unknowns. But if the VNA uses a one-path test set, we can also measure 2. As a result, exact calibration is impossible even if all error terms are perfectly known.
2. To get additional equations we must find a way to increase the number of measurements, but without increasing the number of unknown variables. This can be achieved using two methods. You can eliminate two variables by assuming the DUT is perfectly symmetric and reciprocal, or you can flip the DUT itself, since it's still the same DUT, you get two more measurements without more unknowns (assuming all connectors and cables have perfectly repeatability). 
3. The trick of copying the forward error coefficients into the backward error coefficients works for `skrf.calibration.SOLT` because it exploits the fact that the forward and backward calibrations are mirror images of each other. During forward calibration, only S11 and S21 are used, and vice versa, so the forward-backward-copying trick works - it's how `skrf.calibration.TwoPortOnePath` implements it internally. But I guess `skrf.calibration.EightTerm` uses all 4 S-parameters simultaneously to compute the error boxes, so the result is obviously incorrect.
4. A standard SOLT via `skrf.calibration.SOLT` absorbs the switch term into the error boxes, while the SOLT via `skrf.calibration.EightTerm` does not. This is true. But the inability to use `skrf.calibration.EightTerm` (or `skrf.calibration.UnknownThru`, or `skrf.calibration.TRL`) on a One Path VNA has nothing to do with the fact. This has nothing to do with the switch terms, but about which S-parameters are used during calibration.

If my reasoning is correct, another conclusion is that theoretically, one can probably achieve calibrated two-path measurements even on a one-path analyzer by the method of introducing a "perturbation network". That is, if there's a fully characterized two-port network connected between the 2-port DUT and the 2-port VNA that can be enabled and disabled in two measurement pairs, we also get enough equations to solve for all four unknowns, assuming a dissimilar frequency response (in practice one probably needs more perturbations and solve it by numerical optimization). I've previously experimented with this technique to solve for complex impedance using scalar power measurements, as published by METAS, interesting to see it again.

This can also be generalized from 2-port error boxes to the case of a 4-port error network including all crosstalk terms, which would be useful in the full calibration of measured two-path S-parameters by a one-path through an external and fully-characterized crossbar switch (the forward and backward configurations can be seen as two separate error networks). But the equations would be much more complicated in that case if crosstalks must be calibrated out.

biergaizi, 2025-05-10:
After clarifying the misunderstandings of the "switch terms" thanks to @ZiadHatab, I realized a curious fact that is probably obvious and trivial to everyone here, but rarely mentioned: the forward and reverse calibrations are completely independent in the traditional 12-term SOLT error model. There are actually 4 error boxes. You determine 2 of them via S11/S21, then you determine the other two via S22/S12. They do not interact, no time-invariance required. People say it's immune from "switch terms" errors, but it goes way beyond that. You can change the measurement setup and signal path arbitrarily between forward and backward modes, you don't even have to use the same VNA. You can do forward measurements with one VNA, and do backward measurements with another, the calibration is still exact.

Here's a numerical simulation of an amplifier measurement made by two different VNAs. The S11 and S21 of the DUT is measured by VNA No. 1 in forward mode, then the DUT is flipped, and its S22 and S21 is measured by VNA No. 2 in forward mode, so the readings from the VNA are also called S11 and S21. Finally we forge a new two-port network by copying the S11 and S21 from the backward measurement into the S22 and S12 of the forward measurement. We use `skrf.calibration.SOLT` to calibrate the VNA, and do the same for all SOLT calibration standards.

What we find is that exact calibration is achieved from a pair of one-path measurements with different VNAs or different signal path. I would name this technique Double One-Path SOLT calibration, both *Flip* and *FakeFlip* are special cases of it. An obvious practical application is calibrating the two-path measurements made by a one-path VNA across external crossbar switches. This is a more straightforward solution to the practical measurements without using the Rube Goldberg method of "perturbation" I proposed before. This seems useful enough to deserve its own tutorial, which I plan to submit.

The only thing I'm not sure about so far is whether the optional crosstalk terms are also exact, if the VNA is time-variant to the extreme between both modes.

In the following simulation, I measured a two-port amplifier using both the standard SOLT on a two-port, two-path analyzer, and also measured it using the Double One-Path SOLT on two one-path analyzer. The calibration is exact.

[image]

[image]

[image]

[image]

```
12-term SOLT Full max magnitude S11 error: 0.000000 at 5
12-term SOLT Full max phase S11 error: 0.000000 at 5
12-term SOLT Full max magnitude S12 error: 0.000000 at 94
12-term SOLT Full max phase S12 error: 0.000000 at 94
12-term SOLT Full max magnitude S21 error: 0.000000 at 74
12-term SOLT Full max phase S21 error: 0.000000 at 94
12-term SOLT Full max magnitude S22 error: 0.000000 at 74
12-term SOLT Full max phase S22 error: 0.000000 at 12
Double One-Path SOLT Full max magnitude S11 error: 0.000000 at 28
Double One-Path SOLT Full max phase S11 error: 0.000000 at 6
Double One-Path SOLT Full max magnitude S12 error: 0.000000 at 16
Double One-Path SOLT Full max phase S12 error: 0.000000 at 16
Double One-Path SOLT Full max magnitude S21 error: 0.000000 at 58
Double One-Path SOLT Full max phase S21 error: 0.000000 at 63
Double One-Path SOLT Full max magnitude S22 error: 0.000000 at 12
Double One-Path SOLT Full max phase S22 error: 0.000000 at 8
```

```
import numpy as np
import skrf
import skrf.calibration
from skrf.media import Coaxial
from matplotlib import pyplot as plt
skrf.stylely()

freq = skrf.F(1, 50, 100, unit='GHz')
nports = 2
coax = Coaxial(freq, z0_port=50, Dint=0.44e-3, Dout=1.0e-3, sigma=1e8)

class UncalibratedNetworkAnalyzer():
    """
    Simulate a 2-port Vector Network Analyzer with a random left error
    box at Port 1, and a right error box at Port 2. The Device-Under-Test
    is connected between both error boxes. The errors are so large that
    they make all measurements results meaningless. But they're linear,
    so a suitable calibration algorithm should be able to remowe them
    exactly - as a severe stress test.
    """
    def __init__(self, freq, onepath=False):
        self.freq = freq
        self.nports = 2
        self.onepath = onepath
        self.create_error_data()

    def create_error_data(self):
        # create random error network
        self.err_left = coax.random(n_ports=2, name='S_errL')
        self.err_right = coax.random(n_ports=2, name='S_errR')

    def measure(self, dut):
[... 131 lines cut ...]
    # I name this application "double one-path calibration". Practical
    # applications include calibration two one-path measurements by
    # one-path VNAs via external crossbar switches.

    # Measure S11 and S21 with one-path VNA No.1, the readings
    # are named S11 and S21.
    onepath_vna_fwd = UncalibratedNetworkAnalyzer(freq, onepath=True)
    dut_meas_fwd = onepath_vna_fwd.measure(dut_ideal)

    # Flip the DUT, measure S22 and S12 with one-path VNA No.2, the readings
    # are still named S11 and S21.
    dut_ideal_back = dut_ideal.copy()
    dut_ideal_back.flip()
    onepath_vna_back = UncalibratedNetworkAnalyzer(freq, onepath=True)
    dut_meas_back = onepath_vna_back.measure(dut_ideal_back)

    # Now perform double one-path calibration.
    dut_corrected_double_onepath = double_onepath_solt_calibration(
        onepath_vna_fwd, onepath_vna_back,
        dut_meas_fwd, dut_meas_back
    )

    compare("12-term SOLT Full", dut_ideal, dut_corrected_full)
    compare("Double One-Path SOLT Full", dut_ideal, dut_corrected_double_onepath)
```

ZiadHatab, 2025-05-10:
@biergaizi 

Glad things became clearer to you. Just one comment, if you start using two one-path VNAs in your "double" procedure, you will need to solve for the 5 error terms for both paths. Hence in total you'r solving 10 error terms, same as in standard SOLT with switches (ignoring cross-talk terms). 

Historically, the whole reason people even do one-path measurements, because they don't have the hardware to create the other path measurement, whether by switches or another VNA. The easiest way to recover the full S-parameter was to flip the DUT manually. Again, it's all about how far you willing to go with your hardware. You can do calibration in none conventional way based on what hardware you have.

biergaizi, 2025-05-14:
> I also noticed that the this technique doesn't work if the perturbation is made at Port 1, but it does work at Port 2 or both ports.

Of course it doesn't work by perturbing port 1. SOL is already an exact calibration, so little new information can be gained by perturbing it.

biergaizi, 2025-08-06:
To summarize, the reason that Unknown Thru and single-tier TRL works is the following assumption:

* It's possible to calibrate each port of the VNA as a one-port reflectometer, using only reflective standards.

The Unknown Thru works because the left and right error boxes associated at each port are already nearly-determined, the only missing step is to obtain an unknown constant factor. Because the error box's S12 and S21 is indistinguishable by one-port calibration, one is normalized to 1 - this is the only problem that prevents us from using two one-port calibrations to perfectly measure two-port DUTs directly. If we know all four terms there's no need to do any Thru calibration at all. All Unknown Thru does is to find a constant term.

However, the fundamental limitation of a 1-path VNA is that:

* It's impossible to calibrate the second port using reflective standards only. A transmissive standard *must* be used. 

You can simulate a full 2-path VNA using many methods, like two methods that I've shown here. A crossbar switch with S-parameters pre-measured, or an two-state perturbation Thru network (with both states calibrated by SOLT). But both methods require a known Thru, either implicitly by a known crossbar or explicitly by full SOLT calibrations.

So a true Unknown Thru calibration is impossible on 1-path VNAs. The second-tier can have an Unknown Thru, but only because a flush or known Thru has already been used in the first tier. The first-tier error can't be self-calibrated out because it's not possible to transmit a signal from Port 2 (without changing the error box).

The two-path 3-receiver VNA is a curious case in comparison to the one-path VNA: both instruments use crossbar switches, but at different locations. To simulate a 2-path measurement using a 1-path VNA, you place a switch after the coupler, so the signal sampled by the coupler is modified both reflective and transmissive terms when the switch changes its state, the switch must be characterized as two two-port network separately for two states, creating 4 combinations: Err_State1_Left, Err_State1_Right, Err_State2_Left, Err_State2_Right. But in a two-path 3-receiver VNA, the switch is placed after the coupler, acting effectively as a termination load, so it only needs to be characterized as two one-port termination networks known as the "switch terms".

Conclusion: T/R VNAs can't be self-calibrated.