scikit-rf/scikit-rf #659: [BUG?] Cannot reproduce the results from the "De-embedding with Scikit-RF" section in the online documentation

afvincent, 2022-05-11:
Python and relevant librairies versions
--------------------------------------------------

* skrf: 0.22.1 (from conda-forge)
* python: 3.7.13 (from conda)
* matplotlib: 3.5.1 (from conda)

Description
---------------

I tried to reproduce the example section [“De-embedding with Scikit-RF”](https://scikit-rf.readthedocs.io/en/latest/tutorials/Deembedding.html#De-embedding-with-Scikit-RF) in the scikit-rf online documentation but the results that I get after de-embedding look weird and do not agree with with is shown in the documentation. Not sure if this is a bug or if I am simply doing something wrong.

I generate the (synthetic) required data as Touchstone files using Qucs (0.0.19) and following the schematics shown in the documentation. Here is the Qucs *.sch file that I used in practice: [deembedding_example.sch.txt](https://github.com/scikit-rf/scikit-rf/files/8672073/deembedding_example.sch.txt) (*remove the `.txt` extension before use*).

Screenshots of the simulated setups:

**DUT**
[image]

**open**
[image]

**short**
[image]

Here are the output Touchstone files just in case (*remove the `.txt` extension before use*):
* [ind.s2p.txt](https://github.com/scikit-rf/scikit-rf/files/8672103/ind.s2p.txt)
* [open.s2p.txt](https://github.com/scikit-rf/scikit-rf/files/8672106/open.s2p.txt)
* [short.s2p.txt](https://github.com/scikit-rf/scikit-rf/files/8672108/short.s2p.txt)

Once this is done, I simply use the code provided in the documentation (with only minor tweaks):
```python
# Copy-pasted (with only very minor changes mostly related to style) from:
# https://scikit-rf.readthedocs.io/en/latest/tutorials/Deembedding.html#De-embedding-with-Scikit-RF

# Look at the raw inductor measurement with parasitics included
# From S11/S22, it is clear that it is not a pure inductance.

import skrf as rf
import numpy as np
import matplotlib.pyplot as plt

from skrf.calibration import OpenShort

plt.close("all")

fig0, ax0 = plt.subplots(1, 1, num="raw_inductance_s-parameters")
raw_ind = rf.Network("ind.s2p")
raw_ind.plot_s_smith(ax=ax0)

# Plot the inductance and q-factor of the raw inductor measurement.
Lraw_nH = 1e9 * np.imag(1/raw_ind.y[:,0,0]) / 2 / np.pi / raw_ind.f
Qraw = np.abs(np.imag(1 / raw_ind.y[:,0,0]) / np.real(1 / raw_ind.y[:,0,0]))

fig1, (ax1, ax2) = plt.subplots(1, 2, num="raw_inductance_measurements")
ax1.plot(raw_ind.f*1e-9, Lraw_nH)
ax1.grid(True)
ax1.set_ylabel("Inductance (nH)")
ax1.set_xlabel("Frequency (GHz)")
ax2.plot(raw_ind.f*1e-9, Qraw)
ax2.grid(True)
ax2.set_ylabel("Q-factor")
ax2.set_xlabel("Frequency (GHz)")
fig1.tight_layout()  # might rather rely on `plt.subplots(..., 'layout'='constrained')` since Matplotlib 3.5

# Load in short/open dummies as networks.
fig2, ax3 = plt.subplots(1, 1, num="actual_inductance_s-parameters")
open_nw = rf.Network("open.s2p")
short_nw = rf.Network("short.s2p")

dm = OpenShort(dummy_open=open_nw, dummy_short=short_nw, name='tutorial')

actual_ind = dm.deembed(raw_ind)
actual_ind.plot_s_smith(ax=ax3)

# Plot the inductance of the de-embedded measurement.
# We ignore plotting Q-factor here, because an ideal lossless inductor has infinite Q.

Lactual_nH = 1e9 * np.imag(1 / actual_ind.y[:,0,0]) / 2 / np.pi / actual_ind.f

fig3, ax4 = plt.subplots(1, 1, num="actual_inductance_measurements")
ax4.plot(actual_ind.f*1e-9, Lactual_nH)
ax4.grid(True)
#ax4.set_ylim(0.95, 1.1)
ax4.set_ylabel("Inductance (nH)")
ax4.set_xlabel("Frequency (GHz)")
fig3.tight_layout()

# Uncomment the 2 following lines if you want to export the figures

#for fig in (fig0, fig1, fig2, fig3):
#    fig.savefig(fig.get_label() + ".png", dpi=100)
```

The output figures are (in production order):

**fig0** (`raw_inductance_s-parameters`)
[image]

**fig1** (`raw_inductance_measurements`)
[image]

**fig2** (`actual_inductance_s-parameters`)
[image]

**fig3** (`actual_inductance_measurements`)
[image]

If both fig0 and fig1 seems to be OK, I would expect fig2 to look like
[image]
and fig3 like
[image]
while they clearly do not. Did I do something wrong or is there some kind of an issue with the `OpenShort.deembed` method (or in the documentation)?

Expected outcome
-------------------------

Getting all the four same figures as in the  [“De-embedding with Scikit-RF”](https://scikit-rf.readthedocs.io/en/latest/tutorials/Deembedding.html#De-embedding-with-Scikit-RF) section of the online documentation.

Ttl, 2022-05-11:
`OpenShort` first subtracts Y-parameters of the open and then Z-parameters of the short. In this case after subtracting the open Y-parameters, Z-parameters of the circuit have a very large values because there is no shunt impedance left. The difference compared to the example is that the your input S-parameters have so much precision that the cancellation in the open subtraction is so perfect that the method runs out of precision in the Z-parameter calculation.

By nudging the input S-parameters very slightly for example with `raw_ind.s[:,0,0] += 1e-10`, the de-embedding succeeds.

afvincent, 2022-05-11:
Indeed
```python
raw_ind = rf.Network("ind.s2p")
raw_ind.nudge(1e-9)
```
does the trick. Thank for the quick tip!

I wonder if it might be a good idea to include such a step in the documentation example (or at least mention it along the steps). I guess that I am not the only one, who tries to reproduce the example in simulation, when I see a screenshot of an easily available software and with all the relevant information to do so.

As far as I am concerned, my problem is solved. So feel free to close the issue if you think that no further action is actually required.

jhillairet, 2022-05-11:
Thank you @Ttl! Indeed, eventually a mention of this could be added in the documentation.

vik-s, 2022-05-12:
Thanks @Ttl! I recall now that I needed to do something like this when I wrote these functions, but didnt implement it in the documentation. I'll make an update shortly to the docs. Thanks for pointing this out @afvincent!

PR #661 (2022-05-15): Improve s2z, z2s, s2y and y2s accuracy
Slightly nudge the S-parameters if Z-parameters would be too close to infinity by changing any eigenvalues that are too small.
The previous method of adding a small value if S-parameter was one did not fix the issue except for in special cases (open and short).

Also improves the numerical accuracy in calculation by avoiding directly calculating the inverse of a matrix.

Fixes #659.

Example:
```python
net = skrf.Network(s=[[0.5, 0.5],[0.5, 0.5]])
print(net.z)
print()
print(skrf.z2s(skrf.s2z(net.s, net.z0), net.z0))

```

```
Previously:
numpy.linalg.LinA [... cut]
closed 2022-05-15 by Ttl