scikit-rf/scikit-rf #660: Errerneous Demo in "Measuring a Mutiport Device with a 2-Port Network Analyzer"

biergaizi, 2022-05-13:
In the tutorial *Measuring a Mutiport Device with a 2-Port Network Analyzer*, the section *[Test For Accuracy](https://scikit-rf.readthedocs.io/en/latest/examples/metrology/Measuring%20a%20Mutiport%20Device%20with%20a%202-Port%20Network%20Analyzer.html)* appears erroneous, it has probably been broken by a regression.

Two years ago, this section reads:

> Making sure our composite network is the same as our DUT
>
>      [10]: composite == dut
>      [10]: True
>
> Nice!. How close ?
>
>      [11]: sum((composite - dut).s_mag)
>      [11]: 9.917536367984054e-13

But it now reads:

> Making sure our composite network is the same as our DUT
>
>      [10]: composite == dut
>      [10]: False
>
> Nice!. How close ?
>
>      [11]: sum((composite - dut).s_mag)
>      [11]: 880.7934663060666

jhillairet, 2022-05-13:
Thank you for reporting. Indeed, seems less close ;)

Ttl, 2022-05-13:
Looks like it broke in #448. Port numbering should probably be changed.

PR #664 (2022-05-15): Z0 getter fix
Z0 getter must return a reference to the `Network._z0` so that code like `net.z0[0, 0] = 50` can work. Fixes #660.

I really don't like this code. I tried to refactor it but every time something broke in the tests. At least this keeps the current functionality.
closed 2022-05-15 by Ttl