scikit-rf/scikit-rf #662: Unable to Create Frequency Dependent Resistors from a Media

biergaizi, 2022-05-13:
The official documentation suggests that a frequency-dependent resistor can be attached to a `Media` by using an array of resistances as input to `Media.resistor()`, the behavior should be similar to `Media.inductor()` or `Media.capacitor()`.

> Media.resistor(R, *args, **kwargs)
> Resistor.
>
> Parameters:
> * R (number, array) – Resistance , in Ohms.
>    If this is an array, must be of same length as frequency vector.

However, it appears to be not really possible in practice. 

```
>>> import skrf
>>> from skrf.media import Coaxial
>>> freq = skrf.Frequency(1000, 2000, 2, "MHz")

>>> medium = Coaxial(frequency=freq)
>>> medium.inductor([1, 2])
2-Port Network: '',  1000.0-2000.0 MHz, 2 pts, z0=[109.1339848-2.16452170e-47j 109.1339848-1.53054797e-47j]
>>> medium.resistor([1, 2])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/user/.local/lib/python3.9/site-packages/skrf/media/media.py", line 534, in resistor
    y[:,0,0] = 1./R
TypeError: unsupported operand type(s) for /: 'float' and 'list'

>>> medium2 = skrf.DefinedGammaZ0(frequency=freq)
>>> medium2.inductor([1, 2])
2-Port Network: '',  1000.0-2000.0 MHz, 2 pts, z0=[50.+0.j 50.+0.j]
>>> medium2.resistor([1, 2])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/user/.local/lib/python3.9/site-packages/skrf/media/media.py", line 534, in resistor
    y[:,0,0] = 1./R
TypeError: unsupported operand type(s) for /: 'float' and 'list'
```

Vinc0110, 2022-05-13:
It seems the input array should have been converted to a numpy array first (implementation error):

It works for the inductor because it multiplies `L` with `w`, which already is a numpy array. So, the multiplication works thanks to broadcasting:
https://github.com/scikit-rf/scikit-rf/blob/139fccaa72600b65846ad0b44dd503cc10171f75/skrf/media/media.py#L603-L611

However, for the resistor, there is no frequency array to mulitply `R` with, so there is no broadcasting:
https://github.com/scikit-rf/scikit-rf/blob/139fccaa72600b65846ad0b44dd503cc10171f75/skrf/media/media.py#L532-L539

The code should convert `R` first: 
`y[:,0,0] = 1./npy.array(R)` 
and so on

biergaizi, 2022-05-14:
I found `Media.attenuator()` has the same problem. It seems that a part portion of the code are written with the assumption of numpy array inputs, not Python list inputs. If so, perhaps the documentation needs an update to mention this gotcha.

Vinc0110, 2022-05-14:
Of course. Instead of a warning in the docs, I think the code should be improved so that it can handle Python lists without a problem. I'll probably create a PR with the fix later.

closed 2022-05-20 by Vinc0110