scikit-rf/scikit-rf #667: CPW Media impedance does not  for srip thickness

mhuser, 2022-05-17:
When writing #665 , we observed that `Media.CPW` strip thickness `t` is not used to correct the characteristic impedance.
Currently, it only affect the conductor loss.
Qucs project provide equations and references to correct the impedance and effective permittivity to account for strip thickness.

mhuser, 2022-05-17:
After looking on [Qucs CPW page](http://qucs.sourceforge.net/tech/node86.html) and [Qucsator CPW source](https://github.com/Qucs/qucsator/blob/develop/src/components/microstrip/cpwline.cpp), there is a big discrepancy between them.

Qusator use equations that account from the substrate height `h` and provide a boolean to compute the case were the coplanar waveguide is backed by metal under the substrate (also known as grounded coplanar waveguide coplanar with lower ground plane or CPWG) that is often seen on printed circuit boards 😄 

In ADS simulator there is even two distinct components : [CPW](https://edadocs.software.keysight.com/pages/viewpage.action?pageId=5928535) and [CPWG](https://edadocs.software.keysight.com/pages/viewpage.action?pageId=5928544)

Both Qucs and ADS correct impedance and effective permittivity for strip thickness.

mhuser, 2022-05-21:
More comparison between Qucs and current `Media.CPW`

- red CPW with zero thickness, agreement of return loss phase and magnitude (thus of characteristic impedance), over-optimistic insertion loss but with proper phase : this is maybe because the current model does not compute dielectric loss, not investigated yet
- green CPW with strip thickness and limited substrate height (not implemented yet, but it shows these parameters have significant influence)
- blue CPW with strip thickness, limited substrate height and metal backing (not implemented yet)

[image]

Vinc0110, 2022-05-21:
If I understand https://github.com/scikit-rf/scikit-rf/pull/665#issuecomment-1133628660 correctly, this issue is not yet resolved.

jhillairet, 2022-05-21:
indeed. Was closed automatically.

mhuser, 2022-05-22:
Promising results, but still some works to achieve characteristic impedance agreement:
[image]

mhuser, 2022-05-22:
Got it !
[image]

mhuser, 2022-05-24:
I think #676 is ready, but there is still the question if we shall do a comparison vs ADS [CPW](https://edadocs.software.keysight.com/pages/viewpage.action?pageId=5928535) and [CPWG](https://edadocs.software.keysight.com/pages/viewpage.action?pageId=5928544).
It would not be a surprise if the results deviate a bit from Qucs because the method to account for thickness seems to make a less general agreement than for microstrip models.

mhuser, 2022-05-24:
@JAnderson419 would you be interested in this quest ?
I could also bear if we do not make the comparison 😉 
[CPW](https://edadocs.software.keysight.com/pages/viewpage.action?pageId=5928535) w = 3mm, s = 0.3mm, t = 35um, h = 1.55mm, t = 35um, ep_r = 4.5 (frequency invariant), tand = 0.018, rho = 1.7e-8 ohm/m, 0.2GHz - 20 GHz, 100pts
This one could be run once more with t = 0 if we want to compare without thickness correction
[image]

[CPWG](https://edadocs.software.keysight.com/pages/viewpage.action?pageId=5928544) w = 1.6mm, s = 0.3mm, t = 35um, h = 1.55mm, t = 35um, ep_r = 4.5 (frequency invariant), tand = 0.018, rho = 1.7e-8 ohm/m, 0.2GHz - 20 GHz, 100pts
[image]

JAnderson419, 2022-06-01:
My apologies @mhuser - I was out sick last week and missed this when catching up on emails. I'm happy to help. Thank you for all the work you are putting into these comparisons between tools! 

[CPW_ADS.zip](https://github.com/scikit-rf/scikit-rf/files/8815040/CPW_ADS.zip)

mhuser, 2022-06-01:
@JAnderson419 First of all, I hope you are getting better or that you will get well soon!
Thank you so much for these precious simulation data !

mhuser, 2022-06-02:
@JAnderson419 Could you run a last simulation with CPWG and all parameters keep equals but metal thickness = 0 ?
This will help to validate the base computation with metal backing.
Thank you a lot for the already generated precious data !

JAnderson419, 2022-06-03:
Here you are.

[CPWG_ads_t0um.txt](https://github.com/scikit-rf/scikit-rf/files/8832626/CPWG_ads_t0um.txt)

mhuser, 2022-06-04:
> Here you are.

Thank you a lot @JAnderson419 !

mhuser, 2022-06-04:
[image]

Above is a comparison ADS / QUCS for coplanar waveguide with/without ground backing and zero thickness. The solid curves are s-parameters generated by these simulators while dots are simulated with `skrf.media CPW` in ads resp. qucs mode.

The computation of static impedance and effective relative permittivity are the same for the two simulators for zero thickness.

The difference between these simulators are:

- ADS does not compute frequency dispersion of characteristic impedance and effective permittivity, QUCS does.
- QUCS keep relative permittivity of the substrate real and constant while ADS use a complex value with tand.

`skrf` generated curves (dots) shows a good agreement when these differences are applied (`compatibility_mode` set to 'ads' or 'qucs' for comparison sake)

mhuser, 2022-06-04:
For `skrf` default behaviour, my proposition is to pick QUCS behaviour with dispersion with the addition of the complex effective permittivity.

mhuser, 2022-06-04:
[image]

For coplanar waveguide with/without conductor backing, `skrf ` has good agreement with QUCS

[image]

The agreement with ADS is poor for insertion loss magnitude, which mean the computed static impedance is slightly different.

QUCS is using an empirical formula from Gupta, Garg, Bahl, Bhartia, _Microstrip Lines and Slotlines_, 2nd ed.Artech House, Inc., 1996. ADS claim to use a correction from Cohn, _Thickness Corrections for Capacitive obstacles and Strip Conductors_, IRE Trans. on Microwave Theory and Techniques, Vol. MTT-8, November 1960, pp. 638-644.

However, by reading last paper, there is no ready-to use thickness correction inside. After stepping back, reverse-engineering ADS thickness correction would need simulations with various thickness and... is probably not useful because it does not sounds like a standard and widely used solution.

By default, my proposition is to just keep the same thickness correction as QUCS.

mhuser, 2022-06-04:
[image]

There is a good agreement on insertion loss of skrf media model and a CPWG mesurement I did in the past. I will put that in an example notebook for CPW media.

The 180° jumps in the phase seems caused by deembedding algorithm but I did not investigate more yet.

closed 2022-06-08 by mhuser

mhuser, 2022-06-08:
Fixed by #676 that add thickness to CPW and more, and #684 for phase jumps in 2xthru deembedding.

PR #676 (2022-06-08): Refactor CPW media to fix #667
Add strip thickness, substrate height, dielectric loss and metal backing to `Media.CPW` to fix #667