scikit-rf/scikit-rf #670: Circuits should not rely only on the network name to detect a port

luar123, 2022-05-20:
To define a external port in circuits Circuit.Port() can be used and the docu states `Passing the frequency and name is mandatory. Port name must include the word ‘port’ inside. (ex: ‘Port1’ or ‘port_3’)`. Unfortunately all functions rely only on this string to determine if a network is a port or not, e.g. port_indexes():
```
if 'port' in str.lower(ntw.name):
                port_indexes.append(idx_cnx)
```
I use the circuit class to work with S-parameters from touchstone files and recently I loaded a file like `module123_sim_export.s123p` which led to a network.name containing 'port' and therefore the resulting circuit was completely wrong.

I think, using the network name as an identifier for ports is probably not a good idea at all, but I understand that it is a simple solution, because no additional parameter or similar is needed. If you want to keep it like this, maybe add some more metrics/restriction, e.g. only one-ports can be ports or the ports name has to start with `Port_`instead just containing `port`somewhere in the name.

Ttl, 2022-05-21:
#671 should fix this by not using name to check if network is a port.

jhillairet, 2022-05-21:
Indeed. When making the API, I had not imagined this scenario...

luar123, 2022-05-21:
Awesome, thank you! That was fast :-)

PR #671 (2022-05-22): Mark Circuit Ports with attribute instead of name
Fixes #670
closed 2022-05-22 by Ttl