scikit-rf/scikit-rf #679: skrf.network.Network.from_z Example is no longer working

dwaardenburg, 2022-05-24:
In the example on importing network parameters from the impedance matrix [https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.Network.from_z.html#skrf.network.Network.from_z](url) my notebook is throwing me an error. This is the code as I run it:

`import skrf as rf
import numpy as np
f = rf.Frequency(start=1, stop=2, npoints=4)  # 4 frequency points
z = np.random.rand(len(f),2,2) + np.random.rand(len(f),2,2)*1j  # 2-port z-matrix: shape=(4,2,2)
ntw = rf.Network.from_z(z, f=f)`

With the error:
VisibleDeprecationWarning: Creating an ndarray from nested sequences exceeding the maximum number of dimensions of 32 is deprecated. If you mean to do this, you must specify 'dtype=object' when creating the ndarray.
  temp_freq.f = npy.array(f) * temp_freq.multiplier
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
File c:\ProgramData\Anaconda3\envs\metal\lib\site-packages\skrf\network.py:498, in Network.from_z(cls, z, *args, **kw)
    [473](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/network.py?line=472) r"""
    [474](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/network.py?line=473) Create a Network from its Z-parameters
    [475](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/network.py?line=474) 
   (...)
    [495](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/network.py?line=494) 
    [496](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/network.py?line=495) """
    [497](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/network.py?line=496) s = npy.zeros(shape=z.shape)
--> [498](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/network.py?line=497) me = cls(s=s, *args, **kw)
    [499](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/network.py?line=498) me.z = z
    [500](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/network.py?line=499) return me

File c:\ProgramData\Anaconda3\envs\metal\lib\site-packages\skrf\network.py:468, in Network.__init__(self, file, name, params, comments, f_unit, s_def, **kwargs)
    [466](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/network.py?line=465) for attr in PRIMARY_PROPERTIES + ['frequency', 'f', 'z0', 'noise', 'noise_freq']:
    [467](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/network.py?line=466)     if attr in kwargs:
--> [468](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/network.py?line=467)         self.__setattr__(attr, kwargs[attr])

File c:\ProgramData\Anaconda3\envs\metal\lib\site-packages\skrf\network.py:1326, in Network.f(self, f)
   [1323](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/network.py?line=1322) @f.setter
   [1324](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/network.py?line=1323) def f(self, f: NumberLike) -> None:
   [1325](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/network.py?line=1324)     tmpUnit = self.frequency.unit
-> [1326](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/network.py?line=1325)     self.frequency = Frequency.from_f(f, unit=tmpUnit)

File c:\ProgramData\Anaconda3\envs\metal\lib\site-packages\skrf\frequency.py:274, in Frequency.from_f(cls, f, *args, **kwargs)
    [272](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/frequency.py?line=271)     f = [f]
    [273](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/frequency.py?line=272) temp_freq =  cls(0,0,0,*args, **kwargs)
--> [274](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/frequency.py?line=273) temp_freq.f = npy.array(f) * temp_freq.multiplier
    [275](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/frequency.py?line=274) temp_freq.check_monotonic_increasing()
    [277](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/frequency.py?line=276) return temp_freq

File c:\ProgramData\Anaconda3\envs\metal\lib\site-packages\skrf\frequency.py:495, in Frequency.f(self, new_f)
    [484](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/frequency.py?line=483) """
    [485](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/frequency.py?line=484) Sets the frequency object by passing a vector in Hz.
    [486](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/frequency.py?line=485) 
   (...)
    [490](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/frequency.py?line=489)     If frequency points are not monotonously increasing
    [491](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/frequency.py?line=490) """
    [492](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/frequency.py?line=491) self._f = npy.array(new_f)
    [494](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/frequency.py?line=493) if npy.allclose(    self._f,
--> [495](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/frequency.py?line=494)                     linspace(self._f[0], self._f[-1], len(self._f))):
    [496](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/frequency.py?line=495)     self.sweep_type = 'lin'
    [497](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/frequency.py?line=496) elif self._f[0] and npy.allclose(  self._f,
    [498](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/skrf/frequency.py?line=497)                     geomspace(self._f[0], self._f[-1], len(self._f))):

File <__array_function__ internals>:180, in linspace(*args, **kwargs)

File c:\ProgramData\Anaconda3\envs\metal\lib\site-packages\numpy\core\function_base.py:134, in linspace(start, stop, num, endpoint, retstep, dtype, axis)
    [131](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/numpy/core/function_base.py?line=130) if dtype is None:
    [132](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/numpy/core/function_base.py?line=131)     dtype = dt
--> [134](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/numpy/core/function_base.py?line=133) delta = stop - start
    [135](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/numpy/core/function_base.py?line=134) y = _nx.arange(0, num, dtype=dt).reshape((-1,) + (1,) * ndim(delta))
    [136](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/numpy/core/function_base.py?line=135) # In-place multiplication y *= delta/div is faster, but prevents the multiplicant
    [137](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/numpy/core/function_base.py?line=136) # from overriding what class is produced, and thus prevents, e.g. use of Quantities,
    [138](file:///c%3A/ProgramData/Anaconda3/envs/metal/lib/site-packages/numpy/core/function_base.py?line=137) # see gh-7142. Hence, we multiply in place only for standard scalar types.

TypeError: unsupported operand type(s) for -: 'Frequency' and 'Frequency'

dwaardenburg, 2022-05-24:
Sorry for my bad formatting, this is my first issue here.

FranzForstmayr, 2022-05-24:
This is either a bug in the documentation or in the code. 
For a quick workaround use:

```python
ntw = rf.Network.from_z(z, frequency=f)
```

Ttl, 2022-05-24:
This should be already fixed in master in #657. The same PR also added an option to initialize the network with `z` parameters: `ntw = rf.Network(z=z, f=f)`.

Ttl, 2022-05-24:
That change was only made 2 weeks ago and is still only in the development version. If you are using release version of skrf it will not work.