scikit-rf/scikit-rf #684: 180° phase jumps with IEEEP370_SE_NZC_2xThru deembedding

mhuser, 2022-06-04:
While using IEEEP370_SE_NZC_2xThru deembedding to extract side1 and side2 of a 100mm CPWG line measurement, 180° phase jumps appeared in insertion loss, inducing corresponding phase error in deembedding results:
[image]

These are caused by the computation of `e01` and `e10` using `sqrt`. The results of `sqrt` could be either positive or negative and the correct solution shall be picked.

This is done here:
https://github.com/mhuser/scikit-rf/blob/b7d026cbf513924e07c97274aab311d1e54f454f/skrf/calibration/deembedding.py#L1152-L1161
and here:
https://github.com/mhuser/scikit-rf/blob/b7d026cbf513924e07c97274aab311d1e54f454f/skrf/calibration/deembedding.py#L1163-L1172

The test assume the sign invert at each positive slope change of the phase. This is correct, but calibration error and measurement noise can cause positive spikes of small amplitudes that trigger the sign inversion wrongly, this is what happens on above plot:
[image]

mhuser, 2022-06-04:
@denzchoe 
Maybe could we use standard deviation `std` to compute a threshold, e. g. diff shall be greater than `mean + 3 * std `?

Ttl, 2022-06-05:
One method would be checking which root choice is closer to the previous frequency point:
```python
        k = 1
        test = k * np.sqrt(s21r * (1 - e111 * e112))
        e01 = zeros((n), dtype = complex)
        for i, value in enumerate(test):
            e01[i] = np.sqrt(s21r[i] * (1 - e111[i] * e112[i]))
            if i > 0:
                if npy.abs(-test[i] - test[i-1]) < npy.abs(test[i] - test[i-1]):
                    e01[i] *= -1
        # calc e10
        k = 1
        test = k * np.sqrt(s12r * (1 - e111 * e112))
        e10 = zeros((n), dtype = complex)
        for i, value in enumerate(test):
            e10[i] = np.sqrt(s12r[i] * (1 - e111[i] * e112[i]))
            if i > 0:
                if npy.abs(-test[i] - test[i-1]) < npy.abs(test[i] - test[i-1]):
                    e10[i] *= -1
```

Ttl, 2022-06-05:
Here's an even better method for determining e01 and e10:

```python
        # calc e01 and e10
        e10e01 = 0.5 * (s12r + s21r) * (1 - e111 * e112)
        e01 = np.sqrt(e10e01)
        e10 = zeros(n, dtype = complex)
        for i in range(n):
            if i > 0:
                if npy.abs(-e01[i] - e01[i-1]) < npy.abs(e01[i] - e01[i-1]):
                    e01[i] *= -1
            e10[i] = e10e01[i] / e01[i]
```

There is only one root choice since `e01 * e10` product is known from 2xthru measurement. This avoid choosing incompatible roots for e10 and e01 that can happen with the current code. This method works with both #676 notebook and IEEEP370 De-embedding notebook where noise is added to the measurements with `s2xthru.add_noise_polar(0.001, 1)`.

mhuser, 2022-06-05:
Looks great !

mhuser, 2022-06-05:
@Ttl 
I confirm this works fine !

mhuser, 2022-06-05:
@Ttl are you ok if I create a PR with this and add a test case that will fail with current code but pass with your modification ?

Residual phase of s2xthru.add_noise_polar(0.001, 1) current version, phase jumps everywhere !
[image]

With @Ttl solution, all phase jumps are gone !
[image]

Ttl, 2022-06-05:
Yes, that's fine.

PR #687 (2022-06-08): fix #684 phase jumps in 2xthru deembedding
- Remove 180° phase jumps in IEEEP370_SE_NZC_2xThru in presence of phase noise thanks to @Ttl elegant solution
- add a corresponding test case
- add paper reference for deembedding algorithm

This fixes #684
commit ceb33218 (2022-06-08): Merge pull request #687 from mhuser/fix-#684

fix #684 phase jumps in 2xthru deembedding
closed 2022-06-08 by mhuser