scikit-rf/scikit-rf #685: SOLT Standard Documentation Rendering Problem in readthedocs.io

biergaizi, 2022-06-05:
I just noticed my documentation contribution (#663) has been merged. However, I found [its rendering by readthedocs.io](https://scikit-rf.readthedocs.io/en/latest/examples/metrology/SOLT%20Calibration%20Standards%20Creation.html) is incorrect. 

In this sentence:

> It’s given as a third-degree polynomial with four coefficients, <span>$</span> y(x) = a_0 + a_1 x + a_2 x^2 + a_3 x^3 <span>$</span>.

The formula is not rendered correctly, the naked TeX is exposed.

And in the References, this item:

> [3]: S-Parameters for Signal Integrity, Peter J. Pupalaikis. Page 481. 

has completely disappeared. I think this is a serious problem.

However, [the GitHub rendering](https://github.com/biergaizi/scikit-rf/blob/7f6df99331ebfb5dc9fbe0be88d7e1973ed46a42/doc/source/examples/metrology/SOLT%20Calibration%20Standards%20Creation.ipynb) of the same documentation doesn't show these problems, so it's likely a bug in the Read the Docs rendering. Does anyone have experience working with Read The Docs? Is there a workaround availabte to these problems?

Ttl, 2022-06-05:
I made a PR to fix this in #686. There are some differences in how html is made compared to how it works in the notebook. Html flies can be generated locally with `make html` in doc folder.

closed 2022-06-05 by jhillairet

jhillairet, 2022-06-05:
Fixed by @Ttl : https://scikit-rf.readthedocs.io/en/latest/examples/metrology/SOLT%20Calibration%20Standards%20Creation.html

PR #686 (2022-06-05): Fix SOLT cal kit example HTML rendering
Fixes #685.

Remove line break in the middle of equation. Escape markdown in References section.
commit b583b482 (2022-06-05): Fix SOLT cal kit example HTML rendering

Fixes #685