scikit-rf/scikit-rf #688: readthedocs.io not reflecting the latest API changes (v0.22.1)

denzchoe, 2022-06-08:
As per title, 
Specifically on new deembedding Classes implemented from v0.22.1 and above

jhillairet, 2022-06-08:
What is missing ? The documentation has been compiled and updateded 2 days ago.

denzchoe, 2022-06-08:
The new P370 Deembedding classes are not reflected in readthedocs

<br>

[image]

jhillairet, 2022-06-08:
Ha OK.

It is because it has not been added into the docstring header. I'll do it now.

For info, the version "0.22.1" doesn't appear as a specific version in RTD as I didn't create a specific version for this minor release. However, the ["latest"](https://scikit-rf.readthedocs.io/en/latest/) version of the doc should reflect the current master branch.

denzchoe, 2022-06-08:
Are you holding it back to be reflected on RTD until more testing needs to be done? Or just accidental omission?

jhillairet, 2022-06-08:
> Or just accidental omission?

Yeah, we just forgot to add them into the docstring header so sphinx did not render them...

closed 2022-06-08 by jhillairet

mhuser, 2022-06-08:
Oops, I missed to fill this docstring. Thank you very much for fixing that !

jhillairet, 2022-06-08:
It should be ok now. Please check.

mhuser, 2022-06-08:
Confirmation. The four IEEEP370 methods do appears now for latest version on readthedocs.

PR #689 (2022-06-08): Adding the IEEEP370 deembedding methods to the docstring
fix #688