scikit-rf/scikit-rf #704: Drop non monotonic increasing does not work, if Network's z0 is accessed

FranzForstmayr, 2022-06-13:
As the access of network's `z0` silently writes the value `drop_non_monotonic_increasing` is not working any more.

So this is fine:
```python
import skrf as rf

netw = rf.Network('corrupt_file.s2p')
netw.drop_non_monotonic_increasing()
print(netw.z0.shape)
```
But this not:
```python
import skrf as rf

netw = rf.Network('corrupt_file.s2p')
print(netw.z0.shape)
netw.drop_non_monotonic_increasing()
print(netw.z0.shape)
```

I discovered this while writing #703, so I suggest to keep this issue open and to discuss in the discussion thread about the underlying problem of the `z0` storage.

Ttl, 2022-06-13:
I don't think `drop_non_monotonic_increasing()` ever worked for Networks with correctly shaped z0 (2 dimensions) because `npy.delete` doesn't specify axis. It also shouldn't directly read and write to `_z0` and instead use getter and setter. When getter is used the z0 shape is correct whenever it can be determined. It shouldn't matter how it's stored internally.

Ttl, 2022-06-13:
One case where the dimensions of internal `_z0` matter is when `s` is reassigned with a different shape array. If internal `_z0` is a scalar then the assignment is fine. However if `_z0` has a different number of ports or frequencies than the new s array, assignment also succeeds since consistency of z0 and s shapes is not enforced when assigning to s. Wrong things can then happen if calculations are performed that need z0 array since it's now incorrect shape.

FranzForstmayr, 2022-06-14:
> I don't think `drop_non_monotonic_increasing()` ever worked for Networks with correctly shaped z0 (2 dimensions) because `npy.delete` doesn't specify axis.

Probably yes, I used it mainly with data from different sources and the unit test uses a simple scalar `z0`

PR #705 (2022-06-14): drop_non_monotonic_increasing fix for 2D z0 shape
fixes #704