scikit-rf/scikit-rf #718: Towards a version 1.0.0

jhillairet, 2022-06-19:
### Discussed in https://github.com/scikit-rf/scikit-rf/discussions/692

According to recommended [semantic versioning](https://semver.org/):

> How do I know when to release 1.0.0?
> 
> If your software is being used in production, it should probably already be 1.0.0. If you have a stable API on which users have come to depend, you should be 1.0.0. If you’re worrying a lot about backwards compatibility, you should probably already be 1.0.0.

scikit-rf should probably already be released as 1.0.0. In addition, I've met some potential users who said to me that looking at `scikit-rf`, they saw the version number as being `0.XY` and deduced that the code was highly unstable, then drop and didn't look it further.

So I think scikit-rf is (soon to be) ready for a release 1.0.0.

This issue aims to track the desired improvements we want before releasing 1.0.0.

FranzForstmayr, 2022-11-03:
Should we move the qtapps in a separate repository? Would be easier to maintain probably?

dvincentwest, 2022-11-04:
@FranzForstmayr I was just reviewing the commit history for qtapps, and it doesn't see a ton of action.  In addition I know there is much needed work underway to revamp the virtual instrument code, in part to support a different GUI framework that uses skrf.  Those changes will likely break some behavior in the qtapps, and without active maintenance, it probably doesn't make sense to keep qtapps around in the main repo.

jhillairet, 2024-01-22:
I guess we are ready now for a 1.0 release, isn't it?

cafeclimber, 2024-01-22:
I think the apps directory still needs to be removed or moved to another repo, Network2 removed, and touchstone 2 support finalized before then at the least

mhuser, 2024-01-22:
We sounds now very close.

closed 2024-04-23 by jhillairet