scikit-rf/scikit-rf #721: true_divide errors

jrmagers, 2022-06-24:
I've noticed loading certain Touchstone files sometimes results in some `true_divide` errors/warnings:

```python

import skrf
ntwk = skrf.Network("dut.s2p")
```

Warnings/Errors:
```
c:\venv\sandbox\lib\site-packages\skrf\network.py:2089: RuntimeWarning: divide by zero encountered in true_divide
  y_opt = 1./(self.z0[0, 0] * (1. + gamma_opt)/(1. - gamma_opt))
c:\venv\sandbox\lib\site-packages\skrf\network.py:2089: RuntimeWarning: invalid value encountered in true_divide
  y_opt = 1./(self.z0[0, 0] * (1. + gamma_opt)/(1. - gamma_opt))
c:\venv\sandbox\lib\site-packages\skrf\network.py:2012: RuntimeWarning: divide by zero encountered in true_divide
  y_opt = 1./(self.z0[0, 0] * (1. + gamma_opt)/(1. - gamma_opt))
c:\venv\sandbox\lib\site-packages\skrf\network.py:2012: RuntimeWarning: invalid value encountered in true_divide
  y_opt = 1./(self.z0[0, 0] * (1. + gamma_opt)/(1. - gamma_opt))
```

Attached is "dut.s2p":
[dut.zip](https://github.com/scikit-rf/scikit-rf/files/8981702/dut.zip)

Vinc0110, 2022-06-26:
Your file includes problematic numbers for the noise parameters in lines 653-790, for example line 653: `2.21 0 1 0 2e-20`

The third value is the source reflection coefficient, which equals one. I guess that is `gamma_opt` from the equation that throws the warning: dividing by 1-1=0 is a problem, of course.

Removing lines 653-790 from your file fixes the issue.

jrmagers, 2022-06-26:
Good catch -- and this circuit is passive so noise parameters don't make any sense. I think it's fair to say this is not a bug in scikit-rf, just a poorly structured Touchstone file.

closed 2022-06-26 by jrmagers