scikit-rf/scikit-rf #726: Off-by-One Error in IEEE P370 De-embedding If the Input Has a DC point

biergaizi, 2022-06-29:
When attempting to run `IEEEP370_SE_NZC_2xThru()` for data de-embedding with a DC point with the following code:

```
import sys
import skrf
from skrf.calibration import IEEEP370_SE_NZC_2xThru

thru = skrf.Network(sys.argv[1])
thru = thru.extrapolate_to_dc(kind='linear')

de_embed = IEEEP370_SE_NZC_2xThru(
    dummy_2xthru=thru
)
```

It attempts to remove the DC point:

        if not self.use_z_instead_ifft:
            # strip DC point if one exists
            if(f[0] == 0):
                warnings.warn(
                    "DC point detected. An interpolated DC point will be included in the errorboxes.",
                    RuntimeWarning
                    )
                flag_DC = True 
                fold = f
                f = f[1:]
                s = s[:, :, 1:]

Then it fails later due to an off-by-one problem. 

```
  File "/home/user/.local/lib/python3.9/site-packages/scikit_rf-0.23.0-py3.9.egg/skrf/calibration/deembedding.py", line 1037, in __init__
    self.s_side1, self.s_side2 = self.split2xthru(self.s2xthru)
  File "/home/user/.local/lib/python3.9/site-packages/scikit_rf-0.23.0-py3.9.egg/skrf/calibration/deembedding.py", line 1176, in split2xthru
    dcs11 = self.DC(s11,f)
  File "/home/user/.local/lib/python3.9/site-packages/scikit_rf-0.23.0-py3.9.egg/skrf/calibration/deembedding.py", line 1115, in DC
    h1 = self.makeStep(fftshift(irfft(concatenate(([DCpoint], Hr * s)), axis=0), axes=0))
ValueError: operands could not be broadcast together with shapes (1840,) (1841,)
```

FranzForstmayr, 2022-06-29:
Shouldn't the last line be like this? 

```python
s = s[1:]
```
The s-parameters are usually stored like fxpxp.

biergaizi, 2022-06-29:
A similar problem exists in `IEEEP370_SE_ZC_2xThru` when the input has a DC point.

Reproducer:
```
import sys
import skrf
from skrf.calibration import IEEEP370_SE_ZC_2xThru

freq_list = list(range(10, 3000, 10))
freq = skrf.Frequency.from_f(freq_list, unit='MHz')
thru = skrf.Network(sys.argv[1])
thru.interpolate_self(freq)
thru = thru.extrapolate_to_dc(kind='linear')

dut = skrf.Network(sys.argv[2])
dut.interpolate_self(freq)
dut = dut.extrapolate_to_dc(kind='linear')

de_embed = IEEEP370_SE_ZC_2xThru(
    dummy_2xthru=thru,
    dummy_fix_dut_fix=dut,
)
```

which crashes with a different matrix error.

```
/home/user/.local/lib/python3.9/site-packages/scikit_rf-0.23.0-py3.9.egg/skrf/calibration/deembedding.py:2032: RuntimeWarning: DC point detected. The included DC point will not be used during extraction.
  warnings.warn(
Traceback (most recent call last):
[...]
  File "/home/user/.local/lib/python3.9/site-packages/scikit_rf-0.23.0-py3.9.egg/skrf/calibration/deembedding.py", line 1689, in __init__
    self.s_side1, self.s_side2 = self.split2xthru(self.s2xthru,
  File "/home/user/.local/lib/python3.9/site-packages/scikit_rf-0.23.0-py3.9.egg/skrf/calibration/deembedding.py", line 2039, in split2xthru
    sfix_dut_fix = Network(frequency = Frequency.from_f(f), s = s)
  File "/home/user/.local/lib/python3.9/site-packages/scikit_rf-0.23.0-py3.9.egg/skrf/network.py", line 480, in __init__
    self.__setattr__(attr, kwargs[attr])
  File "/home/user/.local/lib/python3.9/site-packages/scikit_rf-0.23.0-py3.9.egg/skrf/network.py", line 958, in s
    self._s = fix_param_shape(s)
  File "/home/user/.local/lib/python3.9/site-packages/scikit_rf-0.23.0-py3.9.egg/skrf/network.py", line 6725, in fix_param_shape
    raise ValueError('Input matrix must be square')
ValueError: Input matrix must be square
```

mhuser, 2022-07-01:
Should be fixed by #728 and also implement the interpolation to an harmonic sweep as a bonus. (Harmonic sweep is required to compute step response and the results will be better if the original data has not to be interpolated).

closed 2022-07-16 by jhillairet