scikit-rf/scikit-rf #731: inconsistent behavior when passing a network name as kwarg

joergschoebel, 2022-07-06:
Defining a name of a network sometimes works and sometimes it does not.

Two examples where passing a name works **correctly**:

Looking into the code: Media.inductor (similar: capacitor, resistor and many others) starts its definition using a match, to which the kwargs are passed. Then the match is modified to yield the result, so the name is correctly in the result.

Media.attenuator explicitly sets the name.

Examples where passing a name **does not work**:

Media.line defines a result element which includes the name (passed in kwargs). 
`result = self.match(nports=2, s_def='traveling', **kwargs)`

Then, the result element is cascaded with other elements: line uses a thru at both "ends" of the line. 
`result = self.thru(s_def='traveling')**result**self.thru(s_def='traveling')`

Network.cascade calls Network.connect, which uses a copy of the **first** element to build the result, but the thru does not carry the intended name, so the name is lost. 

Media.shunt_capacitor and shunt_inductor define a circuit element: a capacitor connected to a short, which includes the name. But note that kwargs are not passed to shunt() itself.
`return self.shunt(self.capacitor(C=C, **kwargs) ** self.short())`

Inside shunt(), the shorted capacitor is connected with an unnamed tee (because shunt() never receives any kwargs from shunt_capacitor() ). Again, the first element in connect() defines the name of the result, so the name is lost.
`return connect(self.tee(**kwargs), 1, ntwk, 0)`

Maybe it would help to include the kwargs when calling shunt(), then they will be passed to tee() and further to splitter() and match().
`return self.shunt(self.capacitor(C=C, **kwargs) ** self.short(), **kwargs)`

But I cannot oversee if that would have any unwanted side effects.

joergschoebel, 2022-07-06:
This issue has actually a more severe implication: Circuit obviously expects each network to have a unique name. I just received this error message:
```
File "C:\Program Files\Python310\lib\site-packages\skrf\circuit.py", line 179, in __init__
    raise AttributeError('All Networks must have a name. Faulty network:', ntw)
AttributeError: ('All Networks must have a name. Faulty network:', 2-Port Network: '',  500.0-1800.0 MHz, 67 pts, z0=[50.+0.j 50.+0.j])
```

A very basic workaround would be...
Put somewhere in the code:
```
def named(netwk, name):
    netwk.name = name
    return netwk
```
And then define each critical network by
```
shunt_cap = named(lineMedium.shunt_capacitor(1e-12), 'cap')
line = named(lineMedium.line(10, unit='mm', z0=50, embed=True), 'line10mm')
```

jhillairet, 2022-07-07:
Indeed. 

> ```
> shunt_cap = named(lineMedium.shunt_capacitor(1e-12), 'cap')
> line = named(lineMedium.line(10, unit='mm', z0=50, embed=True), 'line10mm')
> ```

Eventually the following is often used:

```
shunt_cap = lineMedium.shunt_capacitor(1e-12)
shunt_cap.name = 'cap'
```

but yes I agree that giving the name as a parameter during the creation of the Network should work in any case.

mhuser, 2022-07-07:
I think this could be fixed if I picking some line from #727 using `renormalize` instead of cascading `media.thru()`. Thus the line will keep the name form the `match` set trough the `**kwargs`. For the shunts, passing the `**kwargs` could probably do it.

mhuser, 2022-07-07:
With #732
```Python
import skrf as rf
from skrf.media import DefinedGammaZ0

f = rf.Frequency(1, 4, 5, 'GHz')
m = DefinedGammaZ0(frequency = f, z0 = 50, Z0 = 48)
l = m.line(80, 'deg', name = 'test_line', embed=True, z0=m.Z0)
print(l.name)
l2 = m.line(80, 'deg', name = 'test_line2')
print(l2.name)
s = m.shunt_inductor(10e-6, name = 'test_shunt_inductor')
print(s.name)
```
Output
```
test_line
test_line2
test_shunt_inductor
```

PR #732 (2022-07-16): fix `media.line` and `media.shunt_xxx` does not keep their network names
This PR attempt to fix #731 

While creating a `media.line` with `embed == True`, the name of the network, if passed into `**kwargs` was not kept due to the `media.thru()` cascading on both side.

This does not fix the faulty `embed` behaviour regarding the characteristic impedances (see #651 for reference to this issue).

`media.shunt_xxx` methods suffer from the same issue because the `**kwargs` is not passed to the `shunt` method. This PR fixes that.
closed 2022-07-16 by mhuser