scikit-rf/scikit-rf #734: time_gate with multiplication in time domain

j-bau, 2022-07-18:
For time gate skrf currently only supports convolution in frequency domain. In some cases it is more useful to multiply the gate in time domain.

The modification would be placed in time.py within the function time_gate and could be accessed by a new variable to switch between fd and td.

Code as optional path starting from line 300 .. 312 in scikit-rf/skrf/time.py
```
out = ntwk.copy()
s_td = np.fft.ifft(out.s[:,0,0])
s_td_g = s_td * np.fft.fftshift(gate)
s_fd_g = np.fft.fft(s_td_g)
out.s[:,0,0] = s_fd_g
```

Vinc0110, 2022-07-19:
Maybe the `mode` parameter could be used to set the desired behavior. For example `mode='time'` to directly multiply the window in the time domain.

jhillairet, 2022-07-19:
For clarification, the code under discussion is:
https://github.com/scikit-rf/scikit-rf/blob/7f4eee82c19ce3fe2e5f79472a904bad15e86bcb/skrf/time.py#L302-L312

I assume the [convolve1d ](https://docs.scipy.org/doc/scipy/reference/generated/scipy.ndimage.convolve1d.html) function from Numpy is already optimized. @j-bau , did you run into cases where multiplication in time domain was faster than the convolution in frequency domain?

j-bau, 2022-07-19:
Multiplication in time domain is indeed about 25x faster. I tested it versus my code proposal.

- _FD=True is for convolution in frequency domain
- _FD=False is for multiplication in time domain
- Based on Touchstone file with 5001 points

```
In [1]: %timeit -n 50 -r 7 time_gate(ntw.s21, window=('kaiser', 6), center=4, span=2, boundary='wrap', _FD=True)
71.1 ms ± 160 µs per loop (mean ± std. dev. of 7 runs, 50 loops each)

In [2]: %timeit -n 50 -r 7 time_gate(ntw.s21, window=('kaiser', 6), center=4, span=2, boundary='wrap', _FD=False)
2.74 ms ± 33.1 µs per loop (mean ± std. dev. of 7 runs, 50 loops each)
```
But this wasn't my main point to rise here. In some cases convolution leads to other results than multiplication. In my case the time domain approach delivered a more clear signal to continue working with.

Vinc0110, 2022-07-19:
> In some cases convolution leads to other results than multiplication. In my case the time domain approach delivered a more clear signal to continue working with.

Interesting. Is it a numerical issue? In theory, I think, both approaches should be equivalent. Or am I missing something else?

jhillairet, 2022-07-20:
> > In some cases convolution leads to other results than multiplication. In my case the time domain approach delivered a more clear signal to continue working with.
> 
> Interesting. Is it a numerical issue? In theory, I think, both approaches should be equivalent. Or am I missing something else?

I'm also interested. Could you provide more information?

Vinc0110, 2022-07-21:
Found two problems in the implementation of the frequency-domain gating:

1.  the convolution with `mode=reflect` caused significant boundary effects. Using `mode=wrap` works better

2. the frequency transform of the window was normalized in a strange way. Not sure what was going on there, but I think I found a much simpler and straight forward solution.

With the code below, both approaches now yield similar (identical?) results. I'll do some more testing and then create a PR to fix this issue during the weekend.

[image]

```python
from skrf.media.coaxial import Coaxial
import skrf as rf
import matplotlib.pyplot as plt
import numpy as np

from scipy.ndimage import convolve1d
from scipy import signal

# settings
freq = rf.Frequency(0, 20, 4001,'GHz')
window_type = ('tukey', 0.66)
#window_type = ('kaiser', 6)
t_center = (1+0.5)*2+0.1
t_span = 1

# create network
coax_media = Coaxial(frequency=freq, z0=50)
coax_line_1 = coax_media.line(1, 'ns', z0=65)
coax_line_2 = coax_media.line(0.5, 'ns', z0=50)
coax_line_3 = coax_media.line(0.1, 'ns', z0=55)
coax_line_4 = coax_media.line(0.5, 'ns', z0=50)
coax_line_5 = coax_media.line(1, 'ns', z0=65)
ntw_total = coax_line_1 ** coax_line_2 ** coax_line_3 ** coax_line_4 ** coax_line_5

# find start/stop gate indecies
t = ntw_total.frequency.t
t_start = t_center - 0.5 * t_span
t_stop = t_center + 0.5 * t_span
start_idx = (np.abs(t - t_start * 1e-9)).argmin()
stop_idx = (np.abs(t - t_stop * 1e-9)).argmin()

# create window
window_width = abs(stop_idx - start_idx)
window = signal.get_window(window_type, window_width)

# create the gate by padding the window with zeros
gate = np.zeros_like(t)
gate[start_idx:stop_idx] = window

# time-domain gating
ntw_gated_td = ntw_total.s11.copy()
s_td = np.fft.ifftshift(np.fft.ifft(ntw_gated_td.s[:, 0, 0]))
s_td_g = s_td * gate
ntw_gated_td.s[:,0,0] = np.fft.fft(np.fft.fftshift(s_td_g))

# frequency-domain gating
ntw_gated_fd = ntw_total.s11.copy()
kernel = np.fft.ifftshift(np.fft.fft(np.fft.fftshift(gate), norm='forward'))
ntw_gated_fd.s[:, 0, 0] = convolve1d(ntw_gated_fd.s[:, 0, 0], kernel, mode='wrap')

# plotting
fig, ax = plt.subplots(4, 1)

ntw_total.s11.plot_s_time_mag(label='input data', ax=ax[0])

ntw_gated_td.plot_s_time_mag(label='gated in TD', ax=ax[1])
ax[1].plot(ntw_total.frequency.t_ns, gate * 0.05, '--')

ntw_gated_fd.plot_s_time_mag(label='gated in FD', ax=ax[2])
ax[2].plot(ntw_total.frequency.t_ns, gate * 0.05, '--')

ntw_gated_td.plot_s_db(label='gated in TD', ax=ax[3])
ntw_gated_fd.plot_s_db(label='gated in FD', ax=ax[3])

ax[0].set_xlim(0, 5)
ax[1].set_xlim(0, 5)
ax[2].set_xlim(0, 5)

plt.tight_layout()
plt.show()

```

Vinc0110, 2022-07-21:
Provided that both methods are indeed identical, I think we should only implement the faster one (likely the time-domain multiplication) and remove the other from the code.

j-bau, 2022-07-22:
Very nice. Thanks for your help!

arsenovic, 2022-07-22:
great job!

Vinc0110, 2022-07-22:
We have a winner! As expected, the time-domain multiplication is much faster than the frequency-domain convolution. And both results are nearly identical, so I think the choice is a no-brainer. 

In the results below, I think the gating only works for $N>1000$ samples, hence the jump in the rms errors:

**window=('tukey', 0.66)**:
[image]

**window=('kaiser', 6)**:
[image]

closed 2022-09-02 by FranzForstmayr