scikit-rf/scikit-rf #743: Circuit does not handle complex port impedance using Power waves

wktanghk, 2022-08-21:
Hi, 

I ran into a problem when using scikit-rf Circuit. I cannot seem to plot the S-parameter correctly when the ports have complex impedance.

In the test case below, I simply connect port1 (5+j50) to port2 (50ohm). But the resulting network does not seem to be based on Power waves as expected. Instead, it seems to be based on Traveling (or perhpas pseudo) waves. As a result, S21 >0 dB. I tried to explicitly renormalize it to Power wave, but that still does not seem to give the expected results (such as I would get in ADS, and also based on hand calculation). I'm not sure what I'm missing. 

Any help would be very much appreciated!

import skrf as rf
from skrf import Network, Frequency, Circuit
import skrf.plotting
import numpy as np
import matplotlib
from pylab import *
rf.stylely()

freq = skrf.frequency.Frequency(start=28, stop=28, npoints=1, unit='ghz',sweep_type='lin')
## Simple test case: connect a (5+50j)Ω Port to a 50Ω port
P1 = Circuit.Port(freq, name='port01', z0=5+50j)
P2 = Circuit.Port(freq, name='port02', z0=50)

cnx_50 = [] # list of all connections
cnx_50.append([(P1,0),(P2,0)])
ckt_50 = rf.Circuit(cnx_50)
nw = ckt_50.network # This is different than ADS (which is based on power waves).
# With scikit-rf, S11 = -S22, so the results are actually based on 'traveling waves' [despite nw.s_def returns 'power']. Most importantly, S21 > 0dB
# On the other hand, by looking at S11 & S22, ADS gives pwr wave as expected. And S21 < 0dB

# This section Try to explicitly convert to a power wave, but failed
nw_pwr = nw.copy()
nw_pwr.renormalize(nw.z0[0],s_def='power') #This is still different than ADS. Actually no idea what this is by looking at S11 & S22
print('nw.s is :' + str(nw.s))
print('nw_pwr.s is :' + str(nw_pwr.s))

Vinc0110, 2022-08-21:
This is a duplicate of the question on the mailing list: https://groups.google.com/g/scikit-rf/c/Be4JVe7eX14
@wktanghk this should not be a problem with the most recent version of scikit-rf, as stated by Henrik. Please check and then close this issue.

wktanghk, 2022-08-21:
@Vinc0110 Thank you so much for your quick response! Yes, I verfied that with v0.23.0, this issue is now resolved. Thanks!

closed 2022-08-21 by wktanghk