scikit-rf/scikit-rf #749: Cal kit fitting

tuxgvf, 2022-09-02:
Hi all,
Is it possible to add a cal kit fitting tools like this one
https://www.qsl.net/in3otd/electronics/VNA_calkit/calkit.html
Thanks
Tux

jhillairet, 2022-09-06:
Dear @tuxgvf

There is a recent page made by @biergaizi on SOLT Calibration Standards Creation : https://scikit-rf.readthedocs.io/en/latest/examples/metrology/SOLT%20Calibration%20Standards%20Creation.html 

I understand that your would like to fit some experimental S-param values to the mathematical model, am I right?

Vinc0110, 2022-10-21:
I'm closing this issue for now. Please write and reopen this if there are open questions.

closed 2022-10-21 by Vinc0110

biergaizi, 2022-10-21:
I'm not sure if this feature request should be closed. I think it's an interesting feature for inclusion on the long-term roadmap of this project. Cal kit fitting is legitimately a more difficult problem than modeling an existing kit since it involves numerical optimization. I remember seeing one example from a METAS presentation and it's not that simple. It would definitely be helpful the scikit-rf has a built-in feature to perform this task.

tuxgvf, 2022-10-22:
I agree with biergaizi, an example of feature is this one https://www.qsl.net/in3otd/electronics/VNA_calkit/calkit.html it's an octave script. I don't know how accurate it is. But the idea is to help fitting S-parameters measurement to a polynomial model when toi have no Idea about your fringing capacitor and delay for an open for example.  It's the "reverse engineering " of this https://scikit-rf.readthedocs.io/en/latest/examples/metrology/SOLT%20Calibration%20Standards%20Creation.html .

Hope this clarify my Idea.