scikit-rf/scikit-rf #753: Touchstone cannot parse Example 5 from Touchstone spec

NAThompson, 2022-09-11:
The [Touchstone spec](http://www.ibis.org/touchstone_ver2.0/touchstone_ver2_0.pdf) gives a few examples of certain representative touchstone files. Consider example 5, on page 11:

```
! 4-port S-parameter data
! Default impedance is overridden by the [Reference] keyword arguments
! Data cannot be represented using 1.0 syntax
[Version] 2.0
# GHz S MA R 50
[Number of Ports] 4
[Number of Frequencies] 1
[Reference] 50 75 0.01 0.01
[Matrix Format] Full
[Network Data]
5.00000 0.60 161.24 0.40 -42.20 0.42 -66.58 0.53 -79.34 !row 1
0.40 -42.20 0.60 161.20 0.53 -79.34 0.42 -66.58 !row 2
0.42 -66.58 0.53 -79.34 0.60 161.24 0.40 -42.20 !row 3
0.53 -79.34 0.42 -66.58 0.40 -42.20 0.60 161.24 !row 4
```

Now attempt to parse this:

```python
>>> from skrf.io.touchstone import Touchstone
>>> ts = Touchstone('example_5.ts')
Traceback (most recent call last):
  File "~/lib/python3.9/site-packages/skrf/io/touchstone.py", line 135, in __init__
    self.load_file(fid)
  File "~/lib/python3.9/site-packages/skrf/io/touchstone.py", line 280, in load_file
    values.extend([ float(v) for v in line.split() ])
  File "~/lib/python3.9/site-packages/skrf/io/touchstone.py", line 280, in <listcomp>
    values.extend([ float(v) for v in line.split() ])
ValueError: could not convert string to float: '[matrix'
```

jhillairet, 2022-09-11:
That's right, Touchstone format V2 is not yet supported. It's a good reminder that a support of the V2 should be great before releasing a version 1.0

Vinc0110, 2023-03-14:
Another user on the email list was stumbling across the missing touchstone v2 support: https://groups.google.com/g/scikit-rf/c/79ZOyQV8pxc

I think a UserWarning would be a good idea in such cases, so users know what the problem is. Of course, we should also start working on adding the v2 support.

@jhillairet Do you know if anyone has already started working on it?

jhillairet, 2023-03-14:
I'd like to support it before releasing v1.0.

As far as I know, nobody is working on it. It's in my to-do list, but if
someone wants to tackle it ;)

cafeclimber, 2023-03-14:
I can try to take a look after the vi module 😂

jhillairet, 2023-05-19:
> If there is no hurry then I can take over this and touchstone V2.0 issue. As a new contributor :)

Sure, with pleasure!

Matjaz-B, 2023-05-19:
Great!

jhillairet, 2023-10-27:
@Matjaz-B , did you work on it by the way?

Matjaz-B, 2023-12-22:
> @Matjaz-B , did you work on it by the way?

Sorry for late response.

I started working on it, but then I had to postpone it.
I'll try to finish it in January.

FranzForstmayr, 2024-01-21:
Fixed with #993

closed 2024-01-21 by FranzForstmayr