scikit-rf/scikit-rf #754: Possible mistake in the 'SOLT Calibration Standards Creation' example

miek, 2022-09-14:
I have a self-built cal kit and I've been working on fitting Keysight's model to measured S-parameters, so that I can plug the coefficients into my E5062A VNA's cal kit definitions.

I've been using the code from the SOLT Calibration Standards Creation example and it's been extremely helpful (thank you @biergaizi!), but I noticed that it wasn't matching the VNA's internal model when `offset_z0` was set to anything other than 50.

In the example, when creating the mediums `z0` is set to 50:

https://github.com/scikit-rf/scikit-rf/blob/dde299c2f9e8c38f90665db6d90a22a6b6d831f1/doc/source/examples/metrology/SOLT%20Calibration%20Standards%20Creation.ipynb?short_path=ad9db82#L321-L337

However, in the summary at the end `z0` is set to `offset_z0`:

https://github.com/scikit-rf/scikit-rf/blob/dde299c2f9e8c38f90665db6d90a22a6b6d831f1/doc/source/examples/metrology/SOLT%20Calibration%20Standards%20Creation.ipynb?short_path=ad9db82#L553

I'm still getting my head around all the theory, but I *think* the latter code is a mistake and `z0` should also be set to 50 here. If I make that change, then the model matches what I see from the VNA very well.

FWIW, my test procedure looks like this: I do a least_squares fit to find suitable coefficients to match the model to my measured S-parameters, program those into the VNA, and do a response cal on that particular standard. I leave the standard connected, take a measurement, and then compare it to my stored S-params.

biergaizi, 2022-09-22:
Thanks for reporting.

When I wrote the documentation, I tested it on several different calibration standards, but all were 50 Ω standards. So it's plausible that non-50 Ω standards still have problems. 

> I'm still getting my head around all the theory, but I think the latter code is a mistake and z0 should also be set to 50 here. If I make that change, then the model matches what I see from the VNA very well.

I think it's not a problem of the theory, but just a small implementation detail about reference impedance and impedance renormalization. Because more than one impedance is involved here, it can get confusing:  the first impedance is the physical port impedance of the VNA, the second is the impedance of the lossless transmission line.

Let's say the calibration standard has an offset impedance of 75 Ω. If we follow the tutorial and use

    medium = DefinedGammaZ0(frequency=freq, z0=offset_z0, Z0=zc, gamma=gamma_l)

The generated S-parameters is a measurement of a 75-ohm standard by a 75-ohm VNA. I think this behavior is a reasonable default.

But if we use

    medium = DefinedGammaZ0(frequency=freq, z0=50, Z0=zc, gamma=gamma_l)

The generated S-parameters is a measurement of a 75-ohm standard by a 50-ohm VNA. Both results are mathematically equivalent, but the numbers are different... So perhaps you forgot to check the reference impedance before fitting your parameters?

To make the case clearer, consider the following code.

```
import skrf
from skrf.media import DefinedGammaZ0
import numpy as np

def keysight_calkit_offset_line(freq, offset_delay, offset_loss, offset_z0, force_50_port=False):
    if offset_delay or offset_loss:
        alpha_l = (offset_loss * offset_delay) / (2 * offset_z0)
        alpha_l *= np.sqrt(freq.f / 1e9)
        beta_l = 2 * np.pi * freq.f * offset_delay + alpha_l
        zc = offset_z0 + (1 - 1j) * (offset_loss / (4 * np.pi * freq.f)) * np.sqrt(freq.f / 1e9)
        gamma_l = alpha_l + beta_l * 1j

        if force_50_port:
            medium = DefinedGammaZ0(frequency=freq, z0=force_50_port, Z0=zc, gamma=gamma_l)
        else:
            medium = DefinedGammaZ0(frequency=freq, z0=offset_z0, Z0=zc, gamma=gamma_l)
        offset_line = medium.line(d=1, unit='m', z0=medium.Z0, embed=True)
        return medium, offset_line
    else:
        medium = DefinedGammaZ0(frequency=freq, Z0=offset_z0)
        line = medium.line(d=0)
        return medium, line

freq = skrf.Frequency(1, 9000, 1001, "MHz")
medium_75, offset_75 = keysight_calkit_offset_line(
    freq,
    offset_delay=29.242e-12, offset_loss=2.2e9, offset_z0=75,
    force_50_port=False
)
medium_50, offset_50 = keysight_calkit_offset_line(
    freq,
    offset_delay=29.242e-12, offset_loss=2.2e9, offset_z0=75,
    force_50_port=True
)
```

Now, `open_50` would be the data from a 50-ohm VNA:

```
>>> open_50
1-Port Network: '',  1.0-9000.0 MHz, 1001 pts, z0=[50.+0.j]
```

and `open_75` would be the data from a 75-ohm VNA:

```
>>> open_75
1-Port Network: '',  1.0-9000.0 MHz, 1001 pts, z0=[75.+0.j]
```

And the numbers are different.

```
>>> open_50.s
array([[[ 0.99999996-2.76034796e-04j]],

       [[ 0.99999619-2.76005328e-03j]],

       [[ 0.99998625-5.24403440e-03j]],

       ...,

       [[-0.6768273 +7.30024144e-01j]],

       [[-0.67312012+7.33443957e-01j]],

       [[-0.66939995+7.36841109e-01j]]])
>>> open_75.s
array([[[ 0.99999991-4.14052184e-04j]],

       [[ 0.99999143-4.14007006e-03j]],

       [[ 0.99996906-7.86598400e-03j]],

       ...,

       [[-0.84106373+5.34827595e-01j]],

       [[-0.83907103+5.37941757e-01j]],

       [[-0.83706681+5.41048495e-01j]]])
```

But they are mathematical equivalent once the port impedance is renormalized. For example, we may renormalize open_75 to 50 Ω, and the result would be the measurement displayed on a 50 Ω VNA.

    >>> open_75.renormalize(50)
    >>> open_50 == open_75
    True

Thus, my conclusion is that it's not a bug, just a confusion of port impedance. You probably fitted the S-parameters with the wrong port impedance.

If you still have problems, or if you have any idea on how to make the documentation more easy to understand, please leave a comment.

biergaizi, 2022-09-22:
> For example, in the [85032B kit](https://www.keysight.com/gb/en/assets/9922-01521/technical-specifications/Calibration-Kit-Definitions.pdf), the short standards have Offset Z0 set to 50.209 Ω or 49.992 Ω. In these cases, the result should still be normalised to the VNA port impedance and not to offset_z0.

Interesting. I'll see how can I update the documentation.

closed 2022-10-29 by jhillairet

AceYoung938, 2022-12-13:
`import skrf as rf
import pylab
from skrf.calibration import SOLT
from skrf.media import DefinedGammaZ0
import numpy as np

def keysight_calkit_offset_line(freq, offset_delay, offset_loss, offset_z0, port_z0):
    if offset_delay or offset_loss:
        alpha_l = (offset_loss * offset_delay) / (2 * offset_z0)
        alpha_l *= np.sqrt(freq.f / 1e9)
        beta_l = 2 * np.pi * freq.f * offset_delay + alpha_l  
        zc = offset_z0 + (1 - 1j) * (offset_loss / (4 * np.pi * freq.f)) * np.sqrt(freq.f / 1e9) 
        gamma_l = alpha_l + beta_l * 1j 
        medium = DefinedGammaZ0(frequency=freq, z0=port_z0, Z0=zc, gamma=gamma_l) 
        offset_line = medium.line(d=1, unit='m', z0=medium.Z0, embed=True) 
        return medium, offset_line
    else:
        medium = DefinedGammaZ0(frequency=freq, Z0=offset_z0)
        line = medium.line(d=0)
        return medium, line

#
def keysight_calkit_open(freq, offset_delay, offset_loss, c0, c1, c2, c3, offset_z0=50, port_z0=50):
    medium, line = keysight_calkit_offset_line(freq, offset_delay, offset_loss, offset_z0, port_z0)
    # Capacitance is defined with respect to the port impedance offset_z0, not the lossy
    # line impedance. In scikit-rf, the return values of `shunt_capacitor()` and `medium.open()`
    # methods are (correctly) referenced to the port impedance.
    if c0 or c1 or c2 or c3:
        poly = np.poly1d([c3, c2, c1, c0])
        capacitance = medium.shunt_capacitor(poly(freq.f)) ** medium.open()
    else:
        capacitance = medium.open() 
    return line ** capacitance

def keysight_calkit_short(freq, offset_delay, offset_loss, l0, l1, l2, l3, offset_z0=50, port_z0=50):
    # Inductance is defined with respect to the port impedance offset_z0, not the lossy
    # line impedance. In scikit-rf, the return values of `inductor()` and `medium.short()`
    # methods are (correctly) referenced to the port impedance.
    medium, line = keysight_calkit_offset_line(freq, offset_delay, offset_loss, offset_z0, port_z0)
    if l0 or l1 or l2 or l3:
        poly = np.poly1d([l3, l2, l1, l0])
        inductance = medium.inductor(poly(freq.f)) ** medium.short()
    else:
        inductance = medium.short()
    return line ** inductance

def keysight_calkit_load(freq, offset_delay=0, offset_loss=0, offset_z0=50, port_z0=50):
    medium, line = keysight_calkit_offset_line(freq, offset_delay, offset_loss, offset_z0, port_z0)
    load = medium.match()
    return line ** load

def keysight_calkit_thru(freq, offset_delay=0, offset_loss=3.554e9, offset_z0=50, port_z0=50):
    medium, line = keysight_calkit_offset_line(freq, offset_delay, offset_loss, offset_z0, port_z0)
    thru = medium.thru()
    return line ** thru

freq = rf.Frequency(1, 9000, 1001, "MHz")
open_std = keysight_calkit_open(
    freq,
    offset_delay=29.242e-12, offset_loss=2.2e9,
    c0=49.43e-15, c1=-310.1e-27, c2=23.17e-36, c3=-0.1597e-45
)
short_std = keysight_calkit_short(
    freq,
    offset_delay=31.785e-12, offset_loss=2.36e9,
    l0=2.077e-12, l1=-108.5e-24, l2=2.171e-33, l3=-0.01e-42
)
load_std = keysight_calkit_load(freq)
thru_std = keysight_calkit_thru(freq)

open_std.write_touchstone(filename='open_std')
short_std.write_touchstone(filename='short_std')
load_std.write_touchstone(filename='load_std')
thru_std.write_touchstone(filename='thru_std')

short = rf.Network('./short_std.s1p')
shorts = rf.two_port_reflect(short, short)
open = rf.Network('./open_std.s1p')
opens = rf.two_port_reflect(open, open)
load = rf.Network('./load_std.s1p')
loads = rf.two_port_reflect(load, load)

my_ideals = [
    shorts,
    opens,
    loads,
    rf.Network('D:\PyCharm\Python_QT5\VNA\thru_std.s2p'),
]

# a list of Network types, holding 'measured' responses
my_measured = [
    rf.Network('D:\PyCharm\Python_QT5\VNA\snp_file_2\SHORT.s2p'),
    rf.Network('D:\PyCharm\Python_QT5\VNA\snp_file_2\OPEN.s2p'),
    rf.Network('D:\PyCharm\Python_QT5\VNA\snp_file_2\LOAD.s2p'),
    # short_measured,
    # open_measured,
    # load_measured,
    rf.Network('D:\PyCharm\Python_QT5\VNA\snp_file_2\THUR.s2p'),
]

## create a SOLT instance
cal = SOLT(
    ideals=my_ideals,
    measured = my_measured,
    # isolation = rf.Network('D:\PyCharm\Python_QT5\VNA\snp_file_2\LOAD.s2p'),
)

## run, and apply calibration to a DUT
cal.run()
# apply it to a dut
pylab.figure(1)
pylab.title('measured')
dut = rf.Network('D:\PyCharm\Python_QT5\VNA\snp_file_2\DUT2.s2p')
dut.plot_s_db()
dut_caled = cal.apply_cal(dut)

# plot results
pylab.figure(2)
pylab.title('calibrated')
dut_caled.plot_s_db()

# save results
dut_caled.write_touchstone(filename='dut1-real')

pylab.figure(3)
pylab.title('real')
dut_REAL = rf.Network('D:\PyCharm\Python_QT5\VNA\snp_file_2\DUT2_REAL.s2p')
dut_REAL.plot_s_db()
pylab.show()`

hello!
I create a SOLT Calibration Standard and use in SOLT Calibration Code, but it's error?
OSError: [Errno 22] Invalid argument: 'D:\\PyCharm\\Python_QT5\\VNA\thru_std.s2p'

commit 3d593ef4 (2022-10-29): Merge pull request #779 from biergaizi/solt-revise

SOLT.ipynb: distinguish offset and port impedances, close #754.
commit 6fc8b9dc (2022-10-29): SOLT.ipynb: distinguish offset and port impedances, close #754.

Currently, in example code snippets, if a specific offset impedance
is used, the VNA port impedance is also set to this matching value.
The original justification was that, this way, specifying a 75-ohm
offset impedance will create S-parameters as measured by a 75-ohm VNA,
which is a reasonable default.

However, sometimes an offset impedance value slightly different from
the port impedance is used to model imperfections in the standard.
For example, a 50-ohm VNA standard may have a 50.209-ohm or 49.992-ohm
offset impedance. Sett [... cut]