scikit-rf/scikit-rf #757: Time-domain gating: Frequency-domain window required?

Vinc0110, 2022-09-22:
This is a branch from the discussion on the mailing list: https://groups.google.com/g/scikit-rf/c/IoH8hNV58sI

Possibly related to issue #734.

The problem is with the roll-off in the edges of the frequency-domain results after gating (at f_min and f_max). A perfectly straight line is provided by PNA transforms. They probably apply a frequency-domain window before the gating. 

Discussion: Is this required or advised to do by default?

The theoretical background is covered in an application note by Agilent/Keysight: 
- [new version; login required](https://www.keysight.com/de/de/assets/7018-01451/application-notes/5989-5723.pdf)
- [old version; no login required](https://www.eeweb.com/wp-content/uploads/articles-app-notes-files-time-domain-analysis-using-a-network-analyzer-1304976108.pdf)

Example code based on the data by Roger in the mailing list:

```python
from skrf.media.coaxial import Coaxial
import skrf
import matplotlib.pyplot as plt
import numpy as np

# reflection delays in nanoseconds
t1 = 758.725e-6
t2 = 82.759e-3
t3 = 416.759e-3
t4 = 508.759e-3

# window settings
freq = skrf.Frequency(0.05, 33, 1000, 'GHz')
print(freq.f)
window_type = ('tukey', 0.5)
t_center = t2
t_span = 0.25   # 150 ps

# create network
media = Coaxial(frequency=freq, z0=50)
coax_line_1 = media.line(0.5 * t2, 'ns', z0=50, embed=True)
coax_line_2 = media.line(0.5 * (t3 - t2), 'ns', z0=25, embed=True)
coax_line_3 = media.line(0.5 * t2, 'ns', z0=50, embed=True)
nw_total = coax_line_1 ** coax_line_2 ** coax_line_3

# apply gate
nw_gated = skrf.time_gate(nw_total.s11, window=window_type, center=t_center, span=t_span)

# plotting
fig, ax = plt.subplots(3, 1)

nw_total.s11.plot_s_time(ax=ax[0], label='Original')
nw_gated.s11.plot_s_time(ax=ax[0], label='Gated')
ax[0].set_xlim(-0.5, 1.5)
ax[0].legend(loc='upper right')

nw_total.s11.plot_z_time_step(ax=ax[1], label='Original')
nw_gated.s11.plot_z_time_step(ax=ax[1], label='Gated')
ax[1].set_xlim(-0.5, 1.5)

nw_total.s11.plot_s_mag(ax=ax[2], label='Original')
nw_gated.s11.plot_s_mag(ax=ax[2], label='Gated')
ax[2].set_ylim(0, 1)

plt.tight_layout()
plt.show()
```

**With `freq = skrf.Frequency(0.05, 33, 1000, 'GHz')`**:
[image]

**With `freq = skrf.Frequency(0.05, 330, 1000, 'GHz')`**:
[image]

Vinc0110, 2022-09-23:
In case my sloppy writing did not make clear what the problem is:

Roger measured a mismatch (50 Ohm -  25 Ohm - 50 Ohm) with a Keysight PNA and used the integrated function to gate the first reflection. It showed the expected reflection coefficient of 0.33 at all frequencies (50 MHz to 33 GHz):
[image]

When I replicate the network in scikit-rf and then gate the first reflection, it only gives the constant reflection coefficient of 0.33 in the center of the frequency range and rolls off at the lower and upper frequency limit. The width and roll-off depends on the frequency span, see my results above (1st with 50 MHz to 33 GHz, 2nd with 50 MHz to 330 GHz).

Based on the application note by Agilent, it seems they apply an option frequency-domain window to the measurements before the gating to mitigate this behavior. Hence the difference in the PNA results compared to the scikit-rf results (the roll-off).

Questions to be disussed:
- It this roll-off behavior normal or is it a bug in the scikit-rf code?
- If normal, should we also offer an optional frequency-domain window before the gating, just like they seem to do on the PNA? Or should we leave this to the user?

Vinc0110, 2022-09-25:
Ok, things have been really confusing, but I think I sorted it out. Various different topics got mixed, all described very well in the application note by Agilent linked above: 
- **frequency-domain truncation and windowing**: the measured frequency span would ideally be infinite, but it is obviously limited in real-world applications. This causes widening of peaks, seomtimes even large ripples in the time-domain responses after the IDFT/IFFT. Applying a (non-rectangular) window function to the frequency-domain data helps mitigating this effect. scikit-rf already has a feature for this: [Network.windowed()](https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.Network.windowed.html). This is fine for (visual) analysis of the time-domain response, but results in total garbage if that response gets transformed back into frequency domain (e.g. after time gating).
- **time-domain low-pass mode**: taking the *IDFT/IFFT on positive and negative frequency samples* with the negative part being a complex-conjugate copy of the positive part (hermitian response), including the dc component. This results in a real-valued time-domain signal. `irfft()` and `rfft()` from `numpy.fft` do exactly this.
- **time-domain band-pass mode**: taking the *IDFT/IFFT only on the positive frequency samples*, generally without the dc component. This results in a complex-valued time-domain signal. The regular `ifft()` and `fft()` from `numpy.fft` do exactly this.

The current implementation of `time_gate()` performs time-domain **band-pass** mode. What we see is actually in line with the statements in Agilent's application note linked above:

> The IFT is applied from minus one-half of the frequency span to plus one-half of the
span. This windows both sides of the data which increases the impulse width and
reduces the effective bandwidth. The fact that the response is centered around the
VNA center frequency has the effect of multiplying the normal time domain response
by a “modulation” function producing a sinusoidal wave on top of the normal response.
This is apparent in bandpass mode, real or imaginary formats, but is eliminated in the
log mag or lin mag format. In contrast, the center of the windowing function in the low-
pass mode is the DC term, or the first point of the data set. Compared to the low-pass
mode, for the same frequency span and number of points, bandpass mode has twice
the impulse width, which may obscure closely spaced responses.

Sorry for all the plots, but it's really interesting to compare different setups. As you will see, the overall frequency span of the measurement and the (relative?) span of the window has a big impact. The type of window is not too important and only influences the ripple.

---

**Setup 1: `freq = skrf.Frequency(0, 330, 1001, 'GHz')`**

**Plot 1.1**: comparing the time-domain responses using low-pass mode, band-pass mode, and the current skrf implementation (normal and windowed):
[image]

**Plot 1.2**: Gating the first reflection using low-pass and band-pass transforms with a Hann window:
[image]

**Plot 1.3**: Gating the first reflection using low-pass and band-pass transforms with a Kaiser window (beta=10):
[image]

**Plot 1.4**: Gating the first reflection using low-pass and band-pass transforms with a Tukey window (alpha=0.6):
[image]

---

**Setup 2: `freq = skrf.Frequency(0, 33, 1001, 'GHz')`**

**Plot 2.1**: comparing the time-domain responses using low-pass mode, band-pass mode, and the current skrf implementation (normal and windowed):
[image]

**Plot 2.2**: Gating the first reflection using low-pass and band-pass transforms with a Hann window:
[image]

**Plot 1.3**: Gating the first reflection using low-pass and band-pass transforms with a Kaiser window (beta=10):
[image]

**Plot 1.4**: Gating the first reflection using low-pass and band-pass transforms with a Tukey window (alpha=0.6):
[image]

---

**Setup 3: `freq = skrf.Frequency(0, 30, 1001, 'GHz')`**

**Plot 3.1**: comparing the time-domain responses using low-pass mode, band-pass mode, and the current skrf implementation (normal and windowed):
[image]

**Plot 3.2**: Gating the first reflection using low-pass and band-pass transforms with a Hann window:
[image]

**Plot 3.3**: Gating the first reflection using low-pass and band-pass transforms with a Kaiser window (beta=10):
[image]

**Plot 3.4**: Gating the first reflection using low-pass and band-pass transforms with a Tukey window (alpha=0.6):
[image]

jhillairet, 2022-09-26:
Great summary! That would be a super nice new example in the documentation! ;)

Vinc0110, 2022-09-26:
Thanks!
Yes, that is probably an useful example for the docs. I'll so that soon. 

But first, I think, I should create a PR and add the low-pass transform along with the existing band-pass implementation. I used my own routines for the examples above, which are not yet available in scikit-rf. 
This is not only relevant for [time_gate()](https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.time.time_gate.html), but also for various `{s, z, y, a, h, t}_time` representations. I'm not sure if the mode should be selected automatically based on whether or not the response includes a dc and therefore enables a low-pass transform. So far, I have not yet looked into this in more detail.

Otherwise, the existing `mode` parameter of [time_gate()](https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.time.time_gate.html) could be used to select the type of transform, at least for this function. It's current intended for the time gate to switch from a band-pass to a band-stop behavior.

RA226, 2022-09-28:
I can confirm the simulation of the measured data looks good. I am assuming the version of time_gate used here is the one working in the time domain and multiplying the window with the Fourier transformed network?  Did you also test it with the old convolution method? I did not see the roll off with convolution.

RA226, 2022-10-01:
**Keysight reference**
This document is excellent and I have studied it many times!!!
**Low-pass and band-pass**
You are absolutley correct to state that where you can you should use low-pass. I am always thinking in terms of band-pass because I work at such high frequencies.  For me the mismatched line is unusual and something I am working on with NPL in the UK and it is in fact their data that you are using with their approval.
**Convolution or FTT**
They should be mathematically equivalent but to get the equivalence we have to treat the data correctly.
_With Roll-off_
1.  Bandpass FFT with no window causes a roll-off
2. Bandpass Convoltion with Constant or Nearest causes almost identical roll off. This probably depends on exactly where the data stops but it is true in this case.

_With no roll-off_
1. Bandpass FFT windowed in frequency domain and then remove the window after timegating will remove the roll-off. 
2. Bandpass Convolution with Reflect , Mirror, Wrap will also remove the roll-off.

It is easier to visualise why the convolution needs data beyond the last point because as the kernel starts to move past that point the signal will drop-off unless you pad out the data using Reflect or similar. We should always be careful to think about what the final points really mean in this case.

In the FFT case if we do not window the data will  have a sharp transition at the edge as it will most likely not be zero. The window forces the data at the edge to go to zero. Then you need to remove it when you transform back.

**Windowing in Frequency domain**
If you window in the frequency domain using  Network.windowed before Fourier transforming to the time domain you have to divide the network by the window when you transform back to the frequency domain. So if you used the Network.windowed method with all the defaults you would have to divide by the equivalent window as follows:
```
Window_full = signal.kaiser(NPoints, beta=6)
S = 0.5 * S_trans_windowed / Window_full
# Needs the factor of 2 to scale the amplitude correctly
```

I have tested this in the past and it works fine.

**How to proceed**
My view would be:
1. Adding a low-pass option is a good idea
2. Use FFT and have it windowed by default. The inverse window could be provided as another method of Network.
4.  Be able as an option to switch off windowing
5. There maybe some cases where convolution provides advantage. In my tests with 15mm silica in freespace (75-110 GHz) the FFT method gave some ringing which is worse at the edge of the band but this was not seen in convolution as shown below.

[image]

Vinc0110, 2022-10-03:
Thanks for the detailed reply!
I'll follow the above suggestions and will create a PR for this in the next couple of days.
@RA226 It would be great if you could review the code once the PR is created. I'll send you a request.

Vinc0110, 2022-10-16:
A finally created the PR #770 for this issue, please have a look.

The results for my above example are not very different. The roll-off remains, regardless of frequency-domain windowing. Also, convolution and FFT yield identical results, which I still think is a good sign.

Top plot: without frequency-domain windowing, i.e. `fft_window=None`.
Bottom plot: with `fft_window='hann'`.
The theoretical value is $\Gamma = \frac{25\Omega - 50\Omega}{25\Omega + 50\Omega} = -0.33$.

[image]

@RA226, is this maybe just a bad example and the roll-off comes from a totally different artefact? Your example with the S21 on 15mm Silica uses frequency data from 75 GHz to 110 GHz. It would be interesting to test the code on that. Can you provide the s2p file?

RA226, 2022-10-17:
@Vinc0110  Here are two .s2p files one with noise and one without noise
[Silica_s2p.zip](https://github.com/scikit-rf/scikit-rf/files/9802343/Silica_s2p.zip)

I will test the code against some of my data as soon as I can.

Vinc0110, 2022-10-18:
Dear Roger, thank you for the files. 

I must say, your Silica data and your respective gating results are both confusing me. Due to the limited span, there is basically only one single peak in the impulse response (at around 100 ps). In its skirt hides a tiny second peak at around 290 ps. Which one to you want to isolate? In your plot I cannot really see any difference between the raw S21 and the gated version.

Here is what I get for the S21 without noise:

1) Gating at `t_center = 0.1 ns, t_span = 0.25 ns`:
[image]

2) Gating at `t_center = 0.29 ns, t_span = 0.25 ns`:
[image]

Once again, convolution and fft are identical if no frequency-domain window is used (orange and green lines). Using a Hann window seems to make it worse (red lines).

Regarding the roll-off in the gated frequency-domain responses: Could this be due to the abrupt edges of the gating window? This is best seen in the dB plots in the center. The window rolls down from 1 to 0 within the course of only a few samples, i.e. it's not very smooth when the resolution is low due to limited frequency span.

Vinc0110, 2022-10-18:
Oh, almost forgot: If you look at the code I used for this (PR #770), you will see I do not have the factor of 2 when removing the frequency-domain window. Including it gave me incorrect scaling. I also fail to understand why that should be required. Am I missing something important here or am I doing it totally wrong?

RA226, 2022-10-19:
Dear Vince,
I fear I only gave you half the information you needed. In my work I am not using time gating in the way that you might normally use it. This sample sits in a system which is 4m long which produces many artefacts due to standing waves caused by reflections at interfaces. There is therefore good seperation of the sample signal and the system in the time domain and I use gating to remove the system artefacts.  That said this is synthesised data with random noise added  so its not exactly the same.  The two peaks you refer to are both from the sample and they are in fact the Fabry Perot fringes from the slab of material.

So I used Centre = 0.097 ns and span = 8 ns in bandpass mode. I would like to make a detailed comparison between what I did previously and what you have now but I am pushed for time at the moment. I should be able to get around to it in a week or two. Alternatively I could provide  you with the code I used to do this work.  I am not python expert but it might help you and be a quicker way to proceed.

Concerning the factor of two I would need to do teh comparison to comment properly.

Regards

Roger

mhuser, 2022-11-03:
If there is a right or wrong result with windowing and gating, I do not know.

In my current understanding, using the crude lowpass (with dc and harmonic sweep) method implies a high cutoff frequency (with roll-off) and crude bandpass low and high cutoff frequencies (also with roll-off).

Windowing (without un-windowing) will attenuate frequency truncation (Nyquist frequency and aliasing) and change the shape without removing the cutoff. This seems compatible with the results of this issue and also with the old Agilent document, p. 18.

The VNA vendors want to do the gating process "transparent" and conceal these cutoff frequencies somehow. The end-users want to see the results "with some of the frequency data removed" but without the artifacts inserted by the gating process.

One great asset of #770 is that the user can try/choose the method that best fit his specific case 😄

closed 2022-11-06 by mhuser