scikit-rf/scikit-rf #762: Network.plot_s_polar() fails

NAThompson, 2022-10-03:
```
>>> from skrf.network import Network
>>> nwrk = Network("foo.s4p")
>>> nwrk.plot_s_polar()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "$VIRTUAL_ENV/lib/python3.9/site-packages/skrf/plotting.py", line 972, in plot_prop_polar
    plot_complex_polar(
  File "$VIRTUAL_ENV/lib/python3.9/site-packages/skrf/plotting.py", line 543, in plot_complex_polar
    plot_polar(theta=theta, r=r, x_label=x_label, y_label=y_label,
  File "$VIRTUAL_ENV/lib/python3.9/site-packages/skrf/plotting.py", line 444, in plot_polar
    ax = plt.gca(polar=True)
TypeError: gca() got an unexpected keyword argument 'polar'
```

Vinc0110, 2022-10-04:
I can confirm this problem, but only for matplotlib version 3.6. It works fine with older versions.

It seems there was a change in `matplotlib.pyplot.gca()` from matplotlib version 3.5 to 3.6:

- in v3.5 it was [`gca(**kwargs)`](https://matplotlib.org/3.5.3/api/_as_gen/matplotlib.pyplot.gca.html). The `**kwargs` are passed to `add_subplot()` and `polar=True` is a valid option.
- in v3.6 (stable) it is [`gca()`](https://matplotlib.org/stable/api/_as_gen/matplotlib.pyplot.gca.html) without any `**kwargs`.

Vinc0110, 2022-10-04:
The change is also documented in the API changes:
https://matplotlib.org/stable/api/prev_api_changes/api_changes_3.6.0.html#arguments

> Keyword arguments to `gca()`; there is no replacement.

jhillairet, 2022-10-05:
@Vinc0110 , are you working on a fix?

FranzForstmayr, 2022-10-05:
I don't use the polar plots on a regular basis, but did we oversee an Deprecation?

Vinc0110, 2022-10-05:
I have been thinking about a solution, but I have not yet started anything.

> did we oversee an Deprecation?

There was no Deprecation Warning when I tested this yesterday using `gca(polar=True)` with matplotlib 3.5

jhillairet, 2022-10-05:
Shouldn't 
https://github.com/scikit-rf/scikit-rf/blob/cde0f639188b8955acd6227e240409419db861ac/skrf/plotting.py#L445
be just transformed in
```python
        fig, ax = plt.subplots(subplot_kw={'projection': 'polar'})
```
?

FranzForstmayr, 2022-10-05:
I don't think so, but there's a discussion about that topic.

https://github.com/matplotlib/matplotlib/issues/20770

Vinc0110, 2022-10-05:
No, we cannot simply create a new Axes. With `gca()`, a new Axes is only create if there is no existing one available for plotting. That's also the problem discussed in that issue in the matplotlib repository: https://github.com/matplotlib/matplotlib/issues/20770#issuecomment-890494483

> The reason I can't use plt.gca() for step one is it will create a new Axes with a non-polar projection, and it's not then possible to change the projection after creating the Axes. https://github.com/matplotlib/matplotlib/pull/20773 seems to fix this.

I don't see how that problem is fixed in matplotlib 3.6. They later removed all parameters alltogether:
https://github.com/matplotlib/matplotlib/blob/76a5711d4bd104218106262a7946128ff57ada93/lib/matplotlib/figure.py#L1604-L1615

```python
    def gca(self):
        """
        Get the current Axes.

        If there is currently no Axes on this Figure, a new one is created
        using `.Figure.add_subplot`.  (To test whether there is currently an
        Axes on a Figure, check whether ``figure.axes`` is empty.  To test
        whether there is currently a Figure on the pyplot figure stack, check
        whether `.pyplot.get_fignums()` is empty.)
        """
        ax = self._axstack.current()
        return ax if ax is not None else self.add_subplot()
```

Vinc0110, 2022-10-05:
This seems to work with my above example:
```python
# inside scikit-rf
if not mplt.get_fignums() or not mplt.gcf().axes or mplt.gca().name != 'polar':
    ax = mplt.figure().add_subplot(projection='polar')
else:
    ax = mplt.gca()
r1 = [2, 2.5, 3, 1.5]
phi1 = [0.1, 1.5, 4, 6]
ax.plot(phi1, r1)
# /inside
```

closed 2022-10-11 by FranzForstmayr