scikit-rf/scikit-rf #780: IEEE370 Deembedding scikit vs matlab

Zwelckovich, 2022-10-21:
Hello,
i compared the fixtures are produced by python(scikit) and the official matlab scripts.
They are slightly different.
S21:
[image]
S11:
[image]

But if you compare the residual plots by deembedding the s2xthru with the fixtures the differences in the s11 and s21 are much more obvious (blue: matlab scripts / yellow: AICC tool / red: scikit-rf)
S21:
[image]
S11:
[image]

Are you doing something different in the python scripts in comparison to the matlab scripts?

Best regards
Sven

mhuser, 2022-10-21:
Hello Sven,
Which method did you use and what where the parameters ?
The S11 residuals seems much better with scikit-rf.
Best regards,
Matthias

mhuser, 2022-10-21:
It could be interesting to compare step responses like in [this example](https://scikit-rf.readthedocs.io/en/latest/examples/networktheory/IEEEP370%20Deembedding.html) 

There are no intentionnal differences with the matlab implementation/standard description, but translation in Python imply there are some small changes. However, that should not impact the results that much. It is also possible you used a parameter combination that has not been thoroughly tested.

mhuser, 2022-10-21:
It came to me there is one difference between scikit-rf andcthe matlab script. The numbering scheme within scikit-rf follow the IEEEP370 standard, this is FIX-2 has port 2 connected to dut. FIX-2 ports are swapped in the matlab script (that not follow IEEEP370 standard recommendation, surpisingly).

Zwelckovich, 2022-10-21:
Yeah, thats the solution. If i dont flip the FIX-2 i have the same results.

Zwelckovich, 2022-10-22:
So will there be an update? Because from my perspective using the fixture calculation from scikit and the deembed function from scikit you are getting wrong result. Because if your orientation of the port 2 is to the dut the deembedding function is wrong:
[image]

mhuser, 2022-10-22:
I am afraid the function is correct. `.flipped()` turn the FIX-2 so that the cascading works.
The only difference in the matlab script is that this flip does not exists. With matlab no swap is required.
We had a dicussion and choose to follow the IEEEP370 recommanded numbering scheme.

denzchoe, 2022-10-22:
View previous discussion here.
https://github.com/scikit-rf/scikit-rf/pull/694#pullrequestreview-1002014974

mhuser, 2022-10-22:
And you can check [in the matlab code](https://opensource.ieee.org/elec-char/ieee-370/-/blob/master/TG1/IEEEP3702xThru.m#L199)  that FIX-2 is already swapped at creation:

```Matlab
    % S-parameters are setup correctly
    if and(flag_DC == 0, flag_df == 0)

        fixture_model_1r = zeros(2,2,n);
        fixture_model_1r(1,1,:) = e001;
        fixture_model_1r(2,1,:) = e01;
        fixture_model_1r(1,2,:) = e01;
        fixture_model_1r(2,2,:) = e111;

        fixture_model_2r = zeros(2,2,n);
        fixture_model_2r(2,2,:) = e002;
        fixture_model_2r(1,2,:) = e10;
        fixture_model_2r(2,1,:) = e10;
        fixture_model_2r(1,1,:) = e112;
```

mhuser, 2022-10-22:
Hello Sven,
Whatever the outcome is, I have to say this: The feedback of users cross-checking our implementation against matlab and aicc tool is very precious to us!
It help us to understand the needs, find problems or gain confidence in the functionnalities.
Matthias

Vinc0110, 2022-10-22:
[The tutorial](https://scikit-rf.readthedocs.io/en/latest/examples/networktheory/IEEEP370%20Deembedding.html) already includes figures at the top, that explain the arrangements and the port numbering.

As this is obviously still a pitfall, how about another warning or hint in the API docs, such as [IEEEP370_SE_NZC_2xThru](https://scikit-rf.readthedocs.io/en/latest/api/calibration/generated/skrf.calibration.deembedding.IEEEP370_SE_NZC_2xThru.html) and the others?

mhuser, 2022-10-22:
> As this is obviously still a pitfall, how about another warning or hint in the API docs, such as [IEEEP370_SE_NZC_2xThru](https://scikit-rf.readthedocs.io/en/latest/api/calibration/generated/skrf.calibration.deembedding.IEEEP370_SE_NZC_2xThru.html) and the others?

This sounds good!

jhillairet, 2022-10-22:
Yes, a warning explaining that scikit-rf *do follow* the IEEE recommendation may help future users.

Zwelckovich, 2022-10-22:
So i have done some comparison of the residual of scikit, matlab and aicc. I also have a version of the matlab and scikit residuals with a flipped fix2 and not flipped. 

AICC settings:
[image]

Matlab script:
[image]

Python script with comparison of the touchstone files and deembedding:
[image]

S11 plot of all:
[image]

S11 plot of the three curves which are nearly the the same:
[image]

If you dont flip the fix2 of the matlab fixture its totally out of range. The flipped fix2 of scikit is also off. The question is which deembedding is the right one?

denzchoe, 2022-10-23:
@Zwelckovich,

#### Suggestion:
My suggestion at this point is to let us have access to your code and generated S2P files from MATLAB for comparison on our end as well. 

#### Further Comments:
However, from my understanding; AICC is supposed to be generated from the source code of reference, [TG1 Implementation in IEEE SA Open page](https://opensource.ieee.org/elec-char/ieee-370/-/tree/master/TG1). Scikit-rf IEEEP370 library also referenced TG1 page. Some minor tweaks and improvement were implemented on Scikit (to follow [Official Documented Standards](https://standards.ieee.org/standard/370-2020.html), and to allow more control in [causality prior de-embedding](https://github.com/scikit-rf/scikit-rf/pull/708))

What version of AICC are you using?

mhuser, 2022-10-23:
The better deembeddings are these with the smallest residuals (matlab_not_flipped and scikit_flipped). Ideally after deembedding the fixture from 2xthru it should go towards perfect matching (very low return loss) and zero insertion loss, what both matlab and scikit-rf agree. Strange is your AICC results 😓 

The three last curves are close because of equally bad, in such a way return loss is quite high. Comparing insertion loss and s11 step responses would be more meaningfull.

Could you consider sharing your 2xthru, the two matlab fixtures and the version of the tools you used with us?

Zwelckovich, 2022-10-24:
AICC Version: ZcorrectedDeembGUI_v1p0r0 Executable

IEEEP3702xThru Version: Revision: v17 (latest from git repo)

Touchstone Files:
[touchstone_files.zip](https://github.com/scikit-rf/scikit-rf/files/9849505/touchstone_files.zip)

Zwelckovich, 2022-10-24:
In the meantime i did the same with the microstrip segments from the example

In addition I added the forward transmission and the time step plots
[image]
[image]
[image]
[image]
[image]
[image]

mhuser, 2022-10-24:
Hello @Zwelckovich 
Thank you for the dataset and the interesting comparison plots. I will give them a try.
I do not have access to Matlab, but I see you included the fixtures generated by the script, so it's all right.

A hint to put code that I can copy-paste instead picture 😉 :
\```Python
Nice formated code here
\```

mhuser, 2022-10-24:
@Zwelckovich I have some results with your data! My test script is attached.
My AICC tool seems to be broken, I will have to reinstall it.
## Matlab
[image]

## scikit-rf
[image]

## residuals comparison
[image]

- The FIX-2 time step domain agreement is poor for both matlab and scikit-rf
- At low frequency, the return loss is positive for at least FIX-2 (both tools)
- Matlab fixtures exhibit big phase jumps due to issue #684 this has been fixed in scikit-rf but is still a problem with matlab
- When looking at residuals, Matlab is going outside the ±1° limits recommended from Standard due to the phase jumps
- Both tools keep within the ±0.1 dB residual limit recommended from standard and the agreement is very decent

mhuser, 2022-10-24:
The answer, is then there is two main difference in scikit-rf implementation: FIX-2 is flipped relatively to Matlab and there is a better root solution choice that avoid phase jumps in fixtures

mhuser, 2022-10-24:
Bonus, a comparison of matlab vs scikit-rf fixtures:
## FIX-1
[image]
## FIX-2
[image]

Some small differences in lower frequencies are visible in addition to the phase jumps that are removed within scikit-rf. I have no real explanation for these yet (maybe the point used to renormalize the impedance is shifted from one sample ?), but they look acceptable to me.
It is the very first time we are allowed to compare to matlab implementation directly, thanks to @Zwelckovich! Before, we had to indirectly compare against the results of the AICC tool.

Test script:
[tests_for_scikit-rf_issue_780.zip](https://github.com/scikit-rf/scikit-rf/files/9853065/tests_for_scikit-rf_issue_780.zip)

Zwelckovich, 2022-10-25:
Thanks for your help @mhuser . Was a pleasure to discuss the problem with you. If you need any data of the matlab implementation  in the future just ask me.

mhuser, 2022-10-25:
You are welcome @Zwelckovich !
This is the warning I am looking to add in the docstring:
```Python
'''
    Note
    ----
    numbering diagram::
        
         FIX-1    DUT      FIX-2
         +----+   +----+   +----+
        -|1  2|---|1  2|---|2  1|-
         +----+   +----+   +----+
         
    Warning for matlab users
    ------------------------
    There are two differences compared to the original matlab implementation:
        - FIX-2 is flipped (see diagram above)
        - A more robust root choice solution is used that avoids the apparition
          of 180° phase jumps in the fixtures in certain circumstances
'''
```
Do you see something else we could enhance in IEEEP370 deembedding?

It still bothers me somehow that the results exhibit some differences but I have re-read the code without finding major difference others than the libraries used behind the hood for fft and interpolation that are python and not matlab implementations (but should give equivalent results).

mhuser, 2022-10-25:
The readthedocs page can be reviewed [here](https://scikit-rf--781.org.readthedocs.build/en/781/api/calibration/generated/skrf.calibration.deembedding.IEEEP370_SE_NZC_2xThru.html#skrf.calibration.deembedding.IEEEP370_SE_NZC_2xThru)

mhuser, 2022-10-25:
I finally found the legendary agreement with matlab:
[image]
[image]
[image]

The current scikit-rf differ for two reasons:

1. In this example, the frequency is not a harmonic sweep (the frequency step is not the first frequency). For this reason, the data has to be interpolated to a harmonic sweep for the fft/ifft computations with dc point. The results are then interpolated back to the originaly frequency. Matlab use `interp1d('spline')` which does not exists in scikit. `kind = 'linear'` is used instead but this change the results. If `kind = 'cubic'` is rather used, it leads to the same results compared to matlab but I fear it could fail if extrapolation is required depending the frequency vector. Here I would recommend the use a harmonic sweep instead if possible (50 Mhz to 66.95 GHz with 50 MHz steps, 1339 points).
2. The solution to avoid the 180° phase jumps affect the results, but I think they are probably better than the matlab implementation

mhuser, 2022-10-25:
Huum, seems extrapolation is working with `kind = 'cubic'` too

mhuser, 2022-10-26:
I have packed everything together in #781 and tadaa now the agreement with matlab is very accurate even with s11 and s22 but with the phase jumps removed 🎉!
[image]

[image]

[image]

@denzchoe maybe it will be worth giving a try with your last deembedding challenge, although it may not change a lot.

jhillairet, 2022-10-26:
Great work @mhuser! It is especially satisfying when it works even better than with Matlab ;)

arsenovic, 2022-10-26:
this was a fun thread to watch. great detective work!

mhuser, 2022-10-26:
And everything happened thanks to @Zwelckovich endeavours to use scikit-rf IEEEP370 deembedding! He did not give up his deembedding challenge!

closed 2022-10-27 by mhuser

Zwelckovich, 2022-11-02:
I implemented the e01 and e10 calculation from the nzc scikit deembedding in the original matlab source. So if you are interested you can download my version of the NZC-IEEE370 fix calculation without the 180° jumps (like in scikit).
[IEEEP3702xThru_scikit.zip](https://github.com/scikit-rf/scikit-rf/files/9919484/IEEEP3702xThru_scikit.zip)

mhuser, 2022-11-03:
> I implemented the e01 and e10 calculation from the nzc scikit deembedding in the original matlab source. So if you are interested you can download my version of the NZC-IEEE370 fix calculation [...]

Thank you @Zwelckovich !

@denzchoe do you think there is a way/interest/possibility to present the phase-jump removal to IEEEP370 Matlab authors? The interest of this issue is to present a data case where they occur.