scikit-rf/scikit-rf #783: Code linting and testing / Tests are failing for every python versions

mhuser, 2022-10-25:
I fear I am blinded or bewildered somehow, but every PR I do as of now is failing at code linting and testing for any python version. This is the case even if the commit only changes a comment.

jhillairet, 2022-10-26:
(@FranzForstmayr)

I found that:

1) tox expects `[tool:pytest]` instead of `[pytest]` to run properly. 

Then,

2) When running `python -m pytest`, pytest collects tests in directories which it is not supposed to (despite the `norecursedirs` option in `tox.ini`). Seems like the problem comes from pytest in this case (?). I found that the following options in `pyproject.toml`

```
[tool.pytest.ini_options]
norecursedirs = [
    "skrf/vi", 
    "skrf/src", 
    "qtapps", 
    "doc/sphinxext", 
    "doc/source/examples/instrumentcontrol/"
]
```

NB: I've removed `qtapps` because on my machine it can't found the sip module, despite it is installed...

[I have a PR ready,](https://github.com/jhillairet/scikit-rf/tree/CI/fix_tests) unless you want to look at it.

FranzForstmayr, 2022-10-26:
For reference:
https://github.com/jhillairet/scikit-rf/commit/2f3a8fbbd2593bd37cebcee84d655b17c6ab0ea9#commitcomment-87999261

FranzForstmayr, 2022-10-26:
Bug report for pytest-cov
https://github.com/pytest-dev/pytest-cov/issues/561

FranzForstmayr, 2022-10-26:
> (@FranzForstmayr)
> 
> I found that:
> 
> 1. tox expects `[tool:pytest]` instead of `[pytest]` to run properly.

That's the difference from `pyproject.toml` and `tox.ini`. We have our config in `tox.ini` which had the correct syntax.
However, I moved the config to `pyproject.toml`

FranzForstmayr, 2022-10-26:
> ```
> [tool.pytest.ini_options]
> norecursedirs = [
>     "skrf/vi", 
>     "skrf/src", 
>     "qtapps", 
>     "doc/sphinxext", 
>     "doc/source/examples/instrumentcontrol/"
> ]
> ```

This was already in `tox.ini`, but it seems that `pytest` uses the configuration from `pyproject.toml` if available.

FranzForstmayr, 2022-10-26:
Fixed in #776

closed 2022-10-26 by FranzForstmayr

mhuser, 2022-10-27:
Thank you a lot for the lightning fast fix!