scikit-rf/scikit-rf #786: Can't create a Circuit with copies of networks

fffrune, 2022-11-04:
I apologize if this is an ignorant comment, I'm new to this project.

I was trying to create a Circuit with multiple copies of a network.  But I get...
`  File "C:\data\school\grad\research\env\lib\site-packages\skrf\circuit.py", line 194, in __init__
    raise AttributeError('A (network, port) node appears twice in the connection description.')
AttributeError: A (network, port) node appears twice in the connection description.`

But I'm not using the *same* network.  I am making a copy (multiple attempts, even).  Here is a simple example that produces the error (you can provide your own s-parameter file):
`lineFile = '8um_ms_line.s2p'
tline0 = skrf.Network(lineFile)
tline1 = tline0.copy()
# Or could also be (with the same result)
#tline1 = copy.deepcopy(tline0)
# Or could also be (with the same result)
#tline1 = skrf.Network(lineFile)
freqs = tline0.frequency
port0 = skrf.Circuit.Port(frequency=freqs, name='p0')
port1 = skrf.Circuit.Port(frequency=freqs, name='p1')
lineTeeLineConnections = [[(port0, 0),  (tline0, 0)],
                          [(tline0, 1), (tline1, 0)],
                          [(tline1, 1), (port1, 0)]
                          ]
print('Line values equivalent? ', tline0 == tline1)
print('Line IDs equivalent? ', id(tline0) == id(tline1))
lineTeeLine = skrf.Circuit(lineTeeLineConnections) # This fails!`

As I implied in the example above, it looks like the Circuit logic is testing for value equality, when it should be testing for reference equality.  Does that make sense?

fffrune, 2022-11-04:
Ahh, it's the network's `name` that gets compared.  I hadn't considered that.  Easy.  Thanks!

fffrune, 2022-11-04:
It might be helpful to indicate that in the error message, or in the documentation about `Circuit`s.

jhillairet, 2022-11-04:
> It might be helpful to indicate that in the error message, or in the documentation about `Circuit`s.

Yes, you're right, the documentation could be more explicit about the fact Circuit expect a distinct name for each Network. I'll do that.