scikit-rf/scikit-rf #787: Frequency comparison failure due to units

fffrune, 2022-11-04:
I am trying to create a Circuit and all of my networks have the same frequency range, but one is defined in units of Hz, and the other in GHz, causing an error:

tline freqs: 1000000000.0-40000000000.0 Hz, 391 pts
stubs freqs: 1.0-40.0 GHz, 391 pts
Traceback (most recent call last):
  File "subnetwork_growth.py", line 234, in <module>
    SLL_circuit = skrf.Circuit(SLL_connections)
  File "C:\data\school\grad\research\env\lib\site-packages\skrf\circuit.py", line 187, in __init__
    raise AttributeError('All Networks must have same frequencies')
AttributeError: All Networks must have same frequencies

I imagine this should be a simple fix.

fffrune, 2022-11-04:
Nevermind, something else is wrong here.  You can ignore this (I don't know how to delete the issue?).  I'm exporting S-parameters from different applications, the frequency start and stop are the same, but there is something subtly wrong here.

tline freqs: 1.0-40.0 GHz, 391 pts
stubs freqs: 1.0-40.0 GHz, 391 pts
Traceback (most recent call last):
  File "subnetwork_growth.py", line 234, in <module>
    SLL_circuit = skrf.Circuit(SLL_connections)
  File "C:\data\school\grad\research\env\lib\site-packages\skrf\circuit.py", line 187, in __init__
    raise AttributeError('All Networks must have same frequencies')
AttributeError: All Networks must have same frequencies

When I compare tline.f == stubs.f I get...
[ True  True  True  True  True  True  True  True  True  True  True  True
  True  True  True  True  True  True  True  True  True  True  True  True
  True  True  True  True  True  True  True False  True False  True  True
 False  True False  True  True False  True False  True  True False  True
 False  True  True False  True False False False False False False False
  True False False False False False False False False False  True False
 False False False False False False False False  True False False False
 False False False False False False  True  True  True  True  True  True
  True  True  True  True  True  True  True  True  True  True  True  True
  True  True  True  True  True  True  True  True  True  True  True  True
  True  True  True  True  True  True  True  True  True  True  True  True
  True  True  True  True  True  True  True  True  True  True  True  True
  True  True  True  True  True  True  True  True  True  True  True  True
  True  True  True  True  True  True  True  True  True  True  True  True
  True  True  True  True  True  True  True  True  True  True  True  True
  True  True  True  True  True  True  True  True  True  True  True  True
  True  True  True  True  True  True  True  True  True  True  True  True
  True  True  True  True  True  True  True  True  True  True False False
 False False False False  True False False False False False False False
 False False  True False False False False False False False False False
  True False False False False False False False False False  True False
 False False False False False False False False  True False False False
 False False False False False False  True False False False False False
 False False False False  True False False False False False False False
 False False  True False False False False False False False False False
  True False False False False False False False False False  True False
 False False False False False False False False  True False False False
 False False False False False False  True False False False False False
 False False False False  True False False False False False False False
 False False  True False False False False False False False False False
  True False False False False False False False False False  True False
 False False False False False False False False  True False False False
 False False False False False False  True]

So it's not an issue with skrf!

closed 2022-11-04 by fffrune

jhillairet, 2022-11-04:
Note that in such a case, scikit-rf can be a great help. You can interpolate a Network using the frequency array used in another Network. See for example [Network.interpolate](https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.Network.interpolate.html) and example [here](https://scikit-rf.readthedocs.io/en/latest/tutorials/Networks.html#Interpolation-and-Concatenation).

I'm moving the issue as a solved question in the discussion part.