scikit-rf/scikit-rf #791: API docs on ReadTheDocs not configured correctly

Vinc0110, 2022-11-05:
The [API docs on RTD](https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.frequency.Frequency.html) are not right. The left columns are way too narrow:

[image]

FranzForstmayr, 2022-11-08:
[image]

Looks fine on my machine

Vinc0110, 2022-11-08:
Believe it or not, it's also fine on my machine now...

closed 2022-11-08 by Vinc0110

jhillairet, 2022-11-09:
> Believe it or not, it's also fine on my machine now...

I saw it too when you post the issue. I guess they have solved a problem on a server-side...