scikit-rf/scikit-rf #795: Missing file ind.s2p in tutorial

eendebakpt, 2022-11-13:
The tutorial at https://scikit-rf.readthedocs.io/en/latest/tutorials/Deembedding.html loads the file `data/ind.s2p` in this block of code
````
# Look at the raw inductor measurement with parasitics included
# From S11/S22, it is clear that it is not a pure inductance.

import skrf as rf
import numpy as np
import matplotlib.pyplot as plt

raw_ind = rf.Network('data/ind.s2p')
raw_ind.plot_s_smith()
```
The data `data/ind.s2p` does not seem to be included in the scikit-rf package. Can the file be added to the distibution, or the documentation updated with a link to the location where the data can be found?

JAnderson419, 2022-11-13:
Hello Pieter,

Can you check again and see if readthedocs is still giving you an error? It loads the inductor fine when I open the tutorial on the link you sent, so I am wondering if something was down momentarily? 

[image]

eendebakpt, 2022-11-13:
@JAnderson419 The readthedocs renders fine, but I am trying to run the code on my local machine. I copy-paste the code to my IDE, but the data file is not available.

JAnderson419, 2022-11-13:
Ah. I understand now what you are trying to do. The inductor data file (and other tutorial files) are in a [subfolder](
https://github.com/scikit-rf/scikit-rf/tree/master/doc/source/tutorials/data) of the tutorials directory, so the relative link is broken when you copy/paste the code snippet into your IDE. For now, you can download the data folder from Github to run the tutorial.

I agree this is not ideal for a tutorial where people may want to follow along. It looks like most of the doc/source/tutorials/data folder is a copy of the skrf/data folder - perhaps @vik-s can confirm this. We should probably move the new data from the doc/source/tutorials/data folder to the skrf/data folder and change the import in the example code so that it is easier for users to run without downloading extra data. I can look into this a bit more tomorrow probably.

vik-s, 2022-11-14:
@JAnderson419 yes, copying the files over to skrf/data folder should take care of the import issues. It looks like the same files are available in two separate locations in the repository?

jhillairet, 2022-11-14:
> We should probably move the new data from the `doc/source/tutorials/data` folder to the `skrf/data` folder and change the import in the example code so that it is easier for users to run without downloading extra data. I can look into this a bit more tomorrow probably.

Eventually, but by doing that, the tutorial example will have to look data in (`../../../skrf/data`) which is not great as well if you want to replicate it. 

Another possibilities:
- Documenting where to download the file (and/or like making a html link to the file in the documentation)
- Referring the `skrf/data` directory using Python internals, like `os.path.dirname(skrf.__file__)`, which is also a bit cumbersome

Note this issue also appear (with much more files) in the documentation examples, where files are stored in the various directories contained in `skrf/doc/source/examples`

jhillairet, 2022-12-14:
Does anyone have other idea about how to ease user's experience?

vik-s, 2022-12-16:
@jhillairet Perhaps it's better to use a inductor component from within scikit-rf instead of an s2p file?

PR #861 (2023-03-29): Remove duplicate files in documentation
fix #795
- All duplicate files in doc/source/tutorials/data wrt skrf/data have been removed, except `ring slot.s2p` and `ro*.s1p` which are used in the introductive docs. We could have use some `os.path` things to point to the files located in `skrf/data`, but it would obfuscate a lot the introductive documentation, which should be as simple as possible.
- the files used in the deembedding tutorial are now moved into `skrf/data`