scikit-rf/scikit-rf #796: Matplotlib dependency should not be optional

luar123, 2022-11-15:
When installing scikit-rf with pip: `pip install scikit-rf` the optional dependency matplotlib is not installed. However without matplotlib `import skrf` or even `from skrf import Network` will fail:

```  File "...\lib\site-packages\skrf\__init__.py", line 14, in <module>
    from . import media
  File "...\lib\site-packages\skrf\media\__init__.py", line 40, in <module>
    from .freespace import Freespace
  File "...\lib\site-packages\skrf\media\freespace.py", line 19, in <module>
    from ..data import materials
  File "...\lib\site-packages\skrf\data\__init__.py", line 28, in <module>
    from ..io.general import read
  File "...\lib\site-packages\skrf\io\__init__.py", line 29, in <module>
    from .general import *
  File "...\lib\site-packages\skrf\io\general.py", line 81, in <module>
    from ..calibration.calibration import Calibration
  File "...\lib\site-packages\skrf\calibration\__init__.py", line 20, in <module>
    from . import deembedding
  File "...\lib\site-packages\skrf\calibration\deembedding.py", line 64, in <module>
    import matplotlib.pyplot as plt
ModuleNotFoundError: No module named 'matplotlib'
```

This was no issue in 0.23.1.

Vinc0110, 2022-11-15:
This behavior is certainly not intended. The change probably came with PR #785, but I don't know what exactly went wrong.

luar123, 2022-11-15:
Actually, I think #785 fixes it, but it is not included in 0.24.0. Maybe just need to release 0.24.1?

jhillairet, 2022-11-15:
> Actually, I think #785 fixes it, but it is not included in 0.24.0. Maybe just need to release 0.24.1?

Sure, why not.

FranzForstmayr, 2022-11-16:
We can release a 0.24.1, but for @luar123 I simply recommend to install matplotlib via pip.

luar123, 2022-11-16:
Thanks! Sure installing matplotlib directly is working fine. I just use a script to setup virtual environments automatically and that suddenly failed. I have a workaround so for me it is not super important.

closed 2022-11-16 by luar123

jhillairet, 2022-11-16:
> Thanks! Sure installing matplotlib directly is working fine. I just use a script to setup virtual environments automatically and that suddenly failed. I have a workaround so for me it is not super important.

Anyway, thanks for reporting. The new minor version will be released soon.