scikit-rf/scikit-rf #797: Inconsistent results for two ways of making ground connection in scikit-rf circuit, for simulating a lumped LC hybrid circuit.

linhz0hz, 2022-11-15:
I am trying to simulate a lumped LC hybrid circuit using the Circuit module of scikit-rf. There are several ground-connected capacitors, and there are a few ways to connect them. I can connect all of them together and to ground 
(cnx=[...,[(C1,1),(C2,1),(C3,1),(gnd,0)]]) or each to ground individually (
cnx=[...,
[(C1,1),(gnd,0)],
[(C2,1),(gnd,0)],
[(C3,1),(gnd,0)],
]).
 Conceptually the result should be the same, but when I simulate it they are not. It seems that the only correct method is connecting all ground-connected pins in one statement only. Why? 

If it is not clear I can provide more details later

jhillairet, 2022-11-15:
Yeah, indeed. 

The reason is that a `Ground` should be used only once in the connection list. So:
- when connecting N components to a single Ground like in  `(cnx=[...,[(C1,1),(C2,1),(C3,1),(gnd,0)]])`, under the hood Circuit deals properly with this multiple connections by creating the adequate N to 1 Network. So it should be OK. 
- Another possibility is to create as many Ground (with different names) as necessary:
```python
cnx=[...,
[(C1,1),(gnd1,0)],
[(C2,1),(gnd2,0)],
[(C3,1),(gnd3,0)],
]).
```

However, in the case where the same Network is used multiple time in the connection list, like in: 
```python
cnx=[...,
[(C1,1),(gnd,0)],
[(C2,1),(gnd,0)],
[(C3,1),(gnd,0)],
]).
```
I'm afraid only one of the capacitor is correctly grounded, and I'm not sure about the status of the other.

linhz0hz, 2022-11-16:
I see. This is not an problem of ground node only, it is the same for any node of any network. Maybe I missed it in the documentation, or should that be emphasized in the documentation?

jhillairet, 2022-11-16:
> Maybe I missed it in the documentation, or should that be emphasized in the documentation?

You're right, I'll highlight this in the tutorial part of the documentation. It's also related to #789.

PR #798 (2022-11-17): Add warnings in the Circuit tutorial
Resolves #797
closed 2022-11-17 by jhillairet