scikit-rf/scikit-rf #803: mTRL Comparison skrf vs NIST/TUG mTRL (Ziad Hatab)

Zwelckovich, 2022-11-22:
Hello,
using the skrf mTRL and the mTRL from Ziad Hatab (https://github.com/ZiadHatab/multiline-trl-calibration) i have different results. The skrf mTRL are not in the same range as the TUG mTRL. 
[image]

I attached the source files and the touchstone files in the zip file.
[skrf_issue.zip](https://github.com/scikit-rf/scikit-rf/files/10065975/skrf_issue.zip)

Vinc0110, 2022-11-22:
@ZiadHatab

ZiadHatab, 2022-11-22:
@Zwelckovich 

Hi, the mTRL in skrf is working fine. The problem is that you used a non-zero thru length (37.25mm). In my code, after calibration, the reference is set in the middle of the thru. With skrf, the reference plane is automatically shifted backward half the length of the thru you specify.

To get the same result as skrf, you can use the below function in my implementation before applying the calibration to the dut. I already have a section in my repo discussing this. Also regarding skrf, they already mention this in their doc https://scikit-rf.readthedocs.io/en/latest/api/calibration/generated/skrf.calibration.calibration.NISTMultilineTRL.html

```python
cal.shift_plane(-line_lengths[0]/2)
```

[image]

jhillairet, 2022-11-24:
OK, Thank you @ZiadHatab. 

I'll close the issue and will transform it as a discussion. Please feel free to continue the discussion as needed.