scikit-rf/scikit-rf #804: touchstone exported from hfss has power wave definition not only traveling wave.

Ekamezo, 2022-11-22:
skrf version 0.23.* can not read or write touchstone format with power wave.
[https://github.com/scikit-rf/scikit-rf/commit/c64db00b8d3ebf26b34c555693a5a8e77e5b4f0e](url)

hfss exports touchstone format with traveling and/or power definition.
skrf must be able to read and export with those definitions as before. And also the default s_def should be power in order to unify.

Ekamezo, 2022-11-22:
definitions can be changed here
[image]

we always use this definition to design complex port impedance device, waveguide etc.
touchstone format with power exported from hfss is very useful to design. I think we should keep the function.

jhillairet, 2022-11-22:
HFSS uses (was using at least) power wave formulation per default and scikit-rf supports both conventions (see [this example](https://scikit-rf.readthedocs.io/en/latest/examples/networktheory/Working%20with%20Complex%20Characteristic%20Impedances.html)).

I don't understand why you run into problems with the Touchstone format. Could you make an example?

Ekamezo, 2022-11-22:
sample.s20p is exported from hfss with power wave definition.

test 1
ntwk = rf.Network('./sample.s20p')
ntwk.write_touchstone(filename='./sample_export', write_z0=True)
--> s is not changed, but def. traveling is not correct. (2 data lines are overlapped.)
[image]

test 2
ntwk = rf.Network('./sample.s20p')
ntwk.s_def = 'power'
ntwk.write_touchstone(filename='./sample_export', write_z0=True)
--> s is changed and raise warning.
UserWarning: Network s_def will be changed from power to traveling to be compatible with HFSS. To silence this warning manually use Network.renormalize to change the s_def.
[image]

Ekamezo, 2022-11-22:
Thank you for your reply.
Yes, scikit-rf supports both. However, s_def became "traveling" when it read the exported touchstone file from hfss.
And, s_def became "traveling" when it was exported by write_touchstone().

ntwk = rf.Network('./sample.s1p') # sample.s1p is exported from hfss with power wave definition
print(ntwk.s_def)
--> traveling

I often renormalize port impedance with frequency dependent complex impedance to design antenna , so I would like to read and write it with power. It was skrf's strength to be able to do that.

Ekamezo, 2022-11-22:
sample.s1p is exported from hfss with power wave definition.

test 1
ntwk = rf.Network('./sample.s1p')
ntwk.write_touchstone(filename='./sample_export', write_z0=True)
--> s is not changed, but traveling definition is not correct. (2 data lines are overlapped.)
[image]

test 2
ntwk = rf.Network('./sample.s1p')
ntwk.s_def = 'power'
ntwk.write_touchstone(filename='./sample_export', write_z0=True)
--> s is changed and raise warning.
UserWarning: Network s_def will be changed from power to traveling to be compatible with HFSS. To silence this warning manually use Network.renormalize to change the s_def.
[image]

To solve this problem, reading and/or writing touchstone with power is needed.

jhillairet, 2022-11-22:
Interesting. This is probably not intended. 

Could you share the first lines of these files for further testing?

Ekamezo, 2022-11-22:
Datas are small (npoints=2, nports=1), I will share you all of lines including s-parameters.

>Original data exported from hfss with power definition (sample.s1p)
! Touchstone file exported from HFSS
!        File:           
!        Generated:      
!        Design:         HFSSDesign1
!        Project:        
!        Setup:          Setup1
!        Solution:       Sweep1
!
! Variables:
!        XX = 3
!        YY = 1
!
!Data is not renormalized
""# GHz S RI     <-- don't need "" (# makes 'GHz S RI' large font.)
! Terminal data exported
! Port[1] = 1
29.5 -0.35405022547854376 -0.25487435607686554
! Port Impedance 30.000000000000 -10.00000000000000 
37.0 -0.24739933815826692 0.6386340944068366
! Port Impedance 20.000000000000 10.00000000000000 

>Test 1 (sample_export.s1p)
! Touchstone file exported from HFSS
!        File:           
!        Generated:      
!        Design:         HFSSDesign1
!        Project:        
!        Setup:          Setup1
!        Solution:       Sweep1
! Variables:
!        XX = 3
!        YY = 1
!Data is not renormalized
!
!Created with skrf (http://scikit-rf.org).
!Data is not renormalized
""# GHz S RI R
! Port[1] = 1
!freq ReS11 ImS11
29.5 -0.35405022547854376 -0.25487435607686554
! Port Impedance 30.00000000000000 -10.00000000000000 
37.0 -0.24739933815826692 0.6386340944068366
! Port Impedance 20.00000000000000 10.00000000000000 

>Test 2 (sample_export.s1p)
! Touchstone file exported from HFSS
!        File:           
!        Generated:      
!        Design:         HFSSDesign1
!        Project:        
!        Setup:          Setup1
!        Solution:       Sweep1
! Variables:
!        XX = 3
!        YY = 1
!Data is not renormalized
!
!Created with skrf (http://scikit-rf.org).
!Data is not renormalized
# GHz S RI R
! Port[1] = 1
!freq ReS11 ImS11
29.5 -0.43900834417083234 0.19647571908264905
! Port Impedance 30.00000000000000 -10.00000000000000 
37.0 -0.5667163853616851 0.014934425327703131
! Port Impedance 20.00000000000000 10.00000000000000

Ekamezo, 2022-11-22:
>read problem
#672  skrf/network.py
[image]
This last column forces s_def to be 'traveling' regardless of its original definition.

>write problem
#672 skrf/network.py  write_touchstone()
[image]
This last column changes s_def to 'traveling' every time.

Ttl, 2022-11-23:
The `s_def` change to `traveling` was made because that is the default export setting in HFSS. At the time I wasn't even aware that it could be changed. It doesn't seem like there is any comment about which definition is used so just changing the `s_def` to be `power` would break the other case instead. I don't see how we could know which definition should be used?

Writing `power` definition file with port impedances could be allowed.

Ekamezo, 2022-11-27:
It seems the pseudo wave s-parameters is also exported from hfss when port impedance is renormalized.
About detail information of this, please refer to the "chapter 28 - HFSS Technical Notes - Excitations" in HFSS help manual.

The exported touchstone file doesn't have `s_def`, so that `s_def` shoud be determined by user when user import file.
How about using option to decide `s_def` regardless of whether it is output from hfss as before?

ex)
 ntwk = rf.Network("sample.s1p", s_def='traveling')
 ntwk = rf.Network("sample.s1p", s_def='pseudo')
 ntwk = rf.Network("sample.s1p", s_def='power')
And the default for `s_def `is still `S_DEF_DEFAULT` for consistency

Ekamezo, 2022-11-27:
When write touchstone file which has complex port impedance and power/pseudo `s_def`, skrf changes s_def to traveling.
Could I write touchstone file without changing `s_def`?

jhillairet, 2022-11-28:
> How about using option to decide `s_def` regardless of whether it is output from hfss as before?

```python
ntwk = rf.Network("sample.s1p", s_def='traveling') 
ntwk = rf.Network("sample.s1p", s_def='pseudo') 
ntwk = rf.Network("sample.s1p", s_def='power') 
```
well, this is already the case. `Network` constructor has a `s_def` parameter:
https://github.com/scikit-rf/scikit-rf/blob/9a2c0fd04ff1e66d9e081ea445b06567e2112d5b/skrf/network.py#L346-L349

jhillairet, 2022-12-05:
@Ekamezo , in which context and workflow do you need to use complex characteristic impedance?

Ekamezo, 2022-12-06:
Complex impedance is not only characteristic impedance that can be obtained by solving 2-dimensional eigen modes in transmission lines and waveguides. Some of the complex port impedance, they occur on RFID port, RFIC port and mismatched lines (not characteristic impedance of the lines). In these case, pseudo wave and/or power wave is often used to avoid complex calculation of power flow.
And I directly would like to see and store s-parameters as 'power' def, because power flow can be easily calculated by power def gamma and spara .
As you know,
reflection power  = |s11|^2 when s11's def  = 'power',
reflection power != |s11|^2 when s11's def = 'traveling'.
So we can see reflection power from gamma(s11) even though z0 has complex.

jhillairet, 2022-12-14:
> In these case, pseudo wave and/or power wave is often used to avoid complex calculation of power flow.

yes I agree

> And I directly would like to see and store s-parameters as 'power' def, because power flow can be easily calculated by power def gamma and spara .

@Ekamezo , would you have a sketch of a simple design I could reproduce? In particular, I'd to test with the latest versions of HFSS.

Ekamezo, 2022-12-24:
[image]

I recommend to use complex antenna input impedance point. If you set or renormalize port impedance to conjugate of antenna input impedance, s11(gamma) of power wave will become zero.

jhillairet, 2023-01-15:
> In the latest version,` print(ntwk.s_def)` returns 'traveling' every lines.

I agree this is not what one could expect, as `s_def` is set by default to `power` in scikit-rf, unless specified differently by the user. 

However, As said by @Ttl, the HFSS uses the _Generalized_ (=`travelling`) definition internally, and it is the default definition. As nothing in the Touchstone file can help us for deducing what definition has been used, eventually we could throw a warning saying that the definition has been changed following the HFSS default. Indeed, this is written in the HFSS documentation:

> __S Parameters Tab__
>The S Parameters tab offers choices for S parameter definition for post processing as Generalized or Power. Power S parameters are calculated for the reporter, S parameter export, and the Matrix display. Changing the setting between Generalized or Power updates reports and exported data, but not the solve status.
>
>Generalized S parameters are based on the relationship between scattered and incident fields; hence, it is possible to have S parameters greater than one if there is a strong imaginary component of the reference impedance. While correct, some circuit designers may find this counter-intuitive and prefer instead to use Power S parameters. Power S parameters are based on the transfer and reflection of power, which must be less than unity for any passive device.

And this definition can be changed when one right-clicks on a design -> "Design Settings...":
[image]

In the present situation, if I generate two Touchstone files generated by HFSS with the Generalized and Power definitions, scikit-rf correctly reads both files (assuming 'travelling' if both cases). It also correctly exports Touchstone files having the same results inside that the initial HFSS files (files attached):
 
```python
#%%
ntwk_hfss_Generalized = rf.Network('HFSS_Sref_Generalized.s2p')
ntwk_hfss_Generalized.write_touchstone('HFSS_Sref_Generalized_from_skrf.s2p', write_z0=True, form='MA')
print(ntwk_hfss_Generalized.s_def)

#%%
ntwk_hfss_Power = rf.Network('HFSS_Sref_Power.s2p', s_def='power')
ntwk_hfss_Power.write_touchstone('HFSS_Sref_Power_from_skrf.s2p', write_z0=True, form='MA')
print(ntwk_hfss_Power.s_def)  # traveling
```
Problems could arise if connecting these network to other stuff. If you need the 'power' definition, then I suggest changing the definition after having read the Touchstone file:

```python
ntwk_hfss_Power.s_def='power'
print(ntwk_hfss_Power.s_def)  # power
```

[skrf_HFSS_s_def.zip](https://github.com/scikit-rf/scikit-rf/files/10420781/skrf_HFSS_s_def.zip)

Ekamezo, 2023-02-11:
Given that there is a possibility of incorrect s_def, wouldn't it be better to discontinue line 2102 ?
https://github.com/scikit-rf/scikit-rf/blob/9a2c0fd04ff1e66d9e081ea445b06567e2112d5b/skrf/network.py#L2099-L2104

Instead, it may be better to use option as usual.
```
ntwk_hfss = rf.Network('ntwk_hfss.s2p')  # s_def=default
print(ntwk_hfss.s_def) # power

ntwk_hfss = rf.Network('ntwk_hfss.s2p', s_def='power')
print(ntwk_hfss.s_def) # power

ntwk_hfss = rf.Network('ntwk_hfss.s2p', s_def='traveling')
print(ntwk_hfss.s_def) # traveling
```

jhillairet, 2023-02-12:
Yes, I agree it should be possible to select the preferred convention

However, I'm not sure about the default one. The default should match HFSS
default as well (i.e.traveling?).

Ekamezo, 2023-02-16:
Indeed, this is a difficult decision. While traveling waves are commonly used in full-wave analysis tools like HFSS, power waves are often used in circuit network calculations. 
The current default in scikit-rf is power, and it is possible that the developers originally intended it for circuit network calculations.

As already noted, when loading a touchstone file outputted from HFSS, the user needs to select the definition since it is not defined in the data. Therefore, it seems unnecessary to change the default of scikit-rf to traveling wave to match the default of HFSS.

Also, I have one concern. I am not sure if the Generalized definition in HFSS is always traveling wave. I am concerned that the Generalized definition in HFSS may not always be traveling wave but also may be pseudo wave. The traveling wave definition may apply only when using a waveport and not renormalizing ?

Ekamezo, 2023-02-16:
As previously mentioned, there is also an issue where writing data to a file forces the use of traveling wave.
Except for cases where the imaginary part is small, many designers may prefer to use the power wave when there is a complex port impedance. In such cases, it may be desirable to output the data in the power wave definition.
https://github.com/scikit-rf/scikit-rf/blob/9a2c0fd04ff1e66d9e081ea445b06567e2112d5b/skrf/network.py#L2248-L2250

Shouldn't we stop forcing a specific definition when exporting (or importing) data?

jhillairet, 2023-02-18:
> Shouldn't we stop forcing a specific definition when exporting (or importing) data?

in fact, it's a bit more subtle than that. 

skrf, as most (if not all) circuit solvers,  uses the power wave definition by default. Hence, if `s_def` is currently not explicitly passed, the power wave definition will be assumed
https://github.com/scikit-rf/scikit-rf/blob/da8ef584e77ba71f94d50b7884ad124156b2b492/skrf/network.py#L349-L351
 
If we remove:
https://github.com/scikit-rf/scikit-rf/blob/da8ef584e77ba71f94d50b7884ad124156b2b492/skrf/network.py#L2106-L2107
then reading an HFSS-generated Touchstone file will also assumes the power wave definition  by default. Which is _not_ the default setting in HFSS. 
So it means many users could be impacted with their codes producing different results, assuming their s-parameters produced with HFSS have been produced (erroneously) with the power wave definition (instead of the default "Generalized")...   

However, I found a workaround which should work:
- if the `s_def` parameter is not passed (ie `s_def=None`), the assumption of 'power' is delayed *after* that the data (touchstone) have been read. 
- During the Touchstone reading, `s_def` is set to the default HFSS setting only if `s_def` has not been passed. Otherwise if uses the passed `s_def`  

So, the following will work:
```python
ntwk_orig = rf.Network('hfss_file.s1p')
self.assertEqual(ntwk_orig.s_def, S_DEF_HFSS_DEFAULT)
        
ntwk_orig = rf.Network('hfss_file.s1p', s_def='power')
self.assertEqual(ntwk_orig.s_def, 'power')
```
where `S_DEF_HFSS_DEFAULT` could be 'traveling' or 'pseudo' (i'm not sure about the meaning of "Generalized" in HFSS)

jhillairet, 2023-02-18:
Anyway, we still have a problem for exporting files, due to:
https://github.com/scikit-rf/scikit-rf/blob/da8ef584e77ba71f94d50b7884ad124156b2b492/skrf/network.py#L2253-L2255

- keeping these lines, a user dealing with HFSS with the 'power' option will have different results...
- removing these lines, there is the possibility that user code gave different results...

Samuelopez-ansys, 2023-03-21:
Hi @jhillairet , @Ekamezo ,

This is a very interesting topic, I would encourage @Ekamezo to reach Ansys support through the [Ansys customer portal](https://support.ansys.com/), I think it is the most efficient way to get this information.

I will try to investigate it.

jhillairet, 2023-03-29:
I came accross a side-effect, for a HFSS-exported S-parameters (not renormalized) with complex z0. In such case, `ntwk.y` and `rf.s2y(ntwk.s, ntwk.z0)` does not give the same result, as `s2y` assumes by default the `power` definition. Imposing `s_def='travelling'` in `s2y` leads to the expected result (at least the same than found in HFSS).

PR #869 (2023-04-01): Fix reading/writing HFSS Touchstone files using the 'power' S-param definition
Fixes #804 

- If the `s_def` parameter is not explicitly passed (ie `s_def=None`), the assumption of `power` is delayed _after_ have read the touchstone data.
- If HFSS-generated, `s_def` is set to the default HFSS setting only if `s_def` has not been passed explicitly. Otherwise if uses the passed `s_def`

The following will work:

```python
ntwk_orig = rf.Network('hfss_file.s1p') # HFSS generated file
self.assertEqual(ntwk_orig.s_def, S_DEF_HFSS_DEFAULT)
        
ntwk_orig = rf.Network('hfss_file.s1p', s_def='pow [... cut]
closed 2023-04-01 by jhillairet