scikit-rf/scikit-rf #812: ```'SOLT Calibration Standards Creation' Problem

AceYoung938, 2022-12-13:
```
`import skrf as rf
import pylab
from skrf.calibration import SOLT
from skrf.media import DefinedGammaZ0
import numpy as np

def keysight_calkit_offset_line(freq, offset_delay, offset_loss, offset_z0, port_z0):
    if offset_delay or offset_loss:
        alpha_l = (offset_loss * offset_delay) / (2 * offset_z0)
        alpha_l *= np.sqrt(freq.f / 1e9)
        beta_l = 2 * np.pi * freq.f * offset_delay + alpha_l  
        zc = offset_z0 + (1 - 1j) * (offset_loss / (4 * np.pi * freq.f)) * np.sqrt(freq.f / 1e9) 
        gamma_l = alpha_l + beta_l * 1j 
        medium = DefinedGammaZ0(frequency=freq, z0=port_z0, Z0=zc, gamma=gamma_l) 
        offset_line = medium.line(d=1, unit='m', z0=medium.Z0, embed=True) 
        return medium, offset_line
    else:
        medium = DefinedGammaZ0(frequency=freq, Z0=offset_z0)
        line = medium.line(d=0)
        return medium, line

#
def keysight_calkit_open(freq, offset_delay, offset_loss, c0, c1, c2, c3, offset_z0=50, port_z0=50):
    medium, line = keysight_calkit_offset_line(freq, offset_delay, offset_loss, offset_z0, port_z0)
    # Capacitance is defined with respect to the port impedance offset_z0, not the lossy
    # line impedance. In scikit-rf, the return values of `shunt_capacitor()` and `medium.open()`
    # methods are (correctly) referenced to the port impedance.
    if c0 or c1 or c2 or c3:
        poly = np.poly1d([c3, c2, c1, c0])
        capacitance = medium.shunt_capacitor(poly(freq.f)) ** medium.open()
    else:
        capacitance = medium.open() 
    return line ** capacitance

def keysight_calkit_short(freq, offset_delay, offset_loss, l0, l1, l2, l3, offset_z0=50, port_z0=50):
    # Inductance is defined with respect to the port impedance offset_z0, not the lossy
    # line impedance. In scikit-rf, the return values of `inductor()` and `medium.short()`
    # methods are (correctly) referenced to the port impedance.
    medium, line = keysight_calkit_offset_line(freq, offset_delay, offset_loss, offset_z0, port_z0)
    if l0 or l1 or l2 or l3:
        poly = np.poly1d([l3, l2, l1, l0])
        inductance = medium.inductor(poly(freq.f)) ** medium.short()
    else:
        inductance = medium.short()
    return line ** inductance

def keysight_calkit_load(freq, offset_delay=0, offset_loss=0, offset_z0=50, port_z0=50):
    medium, line = keysight_calkit_offset_line(freq, offset_delay, offset_loss, offset_z0, port_z0)
    load = medium.match()
    return line ** load

def keysight_calkit_thru(freq, offset_delay=0, offset_loss=3.554e9, offset_z0=50, port_z0=50):
    medium, line = keysight_calkit_offset_line(freq, offset_delay, offset_loss, offset_z0, port_z0)
    thru = medium.thru()
    return line ** thru

freq = rf.Frequency(1, 9000, 1001, "MHz")
open_std = keysight_calkit_open(
    freq,
    offset_delay=29.242e-12, offset_loss=2.2e9,
    c0=49.43e-15, c1=-310.1e-27, c2=23.17e-36, c3=-0.1597e-45
)
short_std = keysight_calkit_short(
    freq,
    offset_delay=31.785e-12, offset_loss=2.36e9,
    l0=2.077e-12, l1=-108.5e-24, l2=2.171e-33, l3=-0.01e-42
)
load_std = keysight_calkit_load(freq)
thru_std = keysight_calkit_thru(freq)

open_std.write_touchstone(filename='open_std')
short_std.write_touchstone(filename='short_std')
load_std.write_touchstone(filename='load_std')
thru_std.write_touchstone(filename='thru_std')

short = rf.Network('./short_std.s1p')
shorts = rf.two_port_reflect(short, short)
open = rf.Network('./open_std.s1p')
opens = rf.two_port_reflect(open, open)
load = rf.Network('./load_std.s1p')
loads = rf.two_port_reflect(load, load)

my_ideals = [
    shorts,
    opens,
    loads,
    rf.Network('D:\PyCharm\Python_QT5\VNA\thru_std.s2p'),
]

# a list of Network types, holding 'measured' responses
my_measured = [
    rf.Network('D:\PyCharm\Python_QT5\VNA\snp_file_2\SHORT.s2p'),
    rf.Network('D:\PyCharm\Python_QT5\VNA\snp_file_2\OPEN.s2p'),
    rf.Network('D:\PyCharm\Python_QT5\VNA\snp_file_2\LOAD.s2p'),
    # short_measured,
    # open_measured,
    # load_measured,
    rf.Network('D:\PyCharm\Python_QT5\VNA\snp_file_2\THUR.s2p'),
]

## create a SOLT instance
cal = SOLT(
    ideals=my_ideals,
    measured = my_measured,
    # isolation = rf.Network('D:\PyCharm\Python_QT5\VNA\snp_file_2\LOAD.s2p'),
)

## run, and apply calibration to a DUT
cal.run()
# apply it to a dut
pylab.figure(1)
pylab.title('measured')
dut = rf.Network('D:\PyCharm\Python_QT5\VNA\snp_file_2\DUT2.s2p')
dut.plot_s_db()
dut_caled = cal.apply_cal(dut)

# plot results
pylab.figure(2)
pylab.title('calibrated')
dut_caled.plot_s_db()

# save results
dut_caled.write_touchstone(filename='dut1-real')

pylab.figure(3)
pylab.title('real')
dut_REAL = rf.Network('D:\PyCharm\Python_QT5\VNA\snp_file_2\DUT2_REAL.s2p')
dut_REAL.plot_s_db()
pylab.show()`
```

hello!
I create a SOLT Calibration Standard and use in SOLT Calibration Code, but it's error?
OSError: [Errno 22] Invalid argument: 'D:\\PyCharm\\Python_QT5\\VNA\thru_std.s2p'

_Originally posted by @AceYoung938 in https://github.com/scikit-rf/scikit-rf/issues/754#issuecomment-1347876711_

AceYoung938, 2022-12-13:
[image]

AceYoung938, 2022-12-13:
I don't know, Maybe it's my code problem, I don't know how to solve .
My teacher asked me to make the  VNA's software with 'scikit-rf'.

AceYoung938, 2022-12-13:
[image]
Why is this 'deal_medium.short()',  I think it should be 'deal_medium.open()'

jhillairet, 2022-12-14:
> Why is this 'ideal_medium.short()', I think it should be 'ideal_medium.open()'

Using a shunt capacitor in series with an open could be used as well and it would be equivalent of a series capacitor shorted. See to the sentence just below:
`For the open standard, a series medium.shunt_capacitor() terminated by a medium.open() could have also been used to get the same result. `
@biergaizi , can you confirm?

AceYoung938, 2022-12-14:
After create the standards, I want to use it in the SOLT Calibration. I don't know how to do it. So i try to use the standards to generate‘s1p’ file and use 'rf.two_port_reflect(open, open)' (API) to generate 's2p' file.  [image] 
Then, i use the 's2p' file to make the list of ideal Network’s and calibrate the network. But run to this code 
[image]
it have a error and Prompt invalid argument. I don't know why it have a error, maybe My method is wrong。
My English is not good, I don’t know if I have expressed it clearly

jhillairet, 2022-12-14:
Is it a .s2p file coming from an example or from you ? If it comes from you, can you post the file (not as an image) ?

AceYoung938, 2022-12-14:
The thru_std.s2p file is come from standards.  I use the example of 'SOLT Calibration Standards Creation' and get the 'thru_std'.
Then i use this code--"thru_std.write_touchstone()" and get this s2p file.  
[thru_std.txt](https://github.com/scikit-rf/scikit-rf/files/10229027/thru_std.txt)

AceYoung938, 2022-12-14:
Because github cannot upload the S2P file, I modified its suffix

jhillairet, 2022-12-14:
OK, so first you should not overwrite the Python function `open` with the following line :

`open = rf.Network('./open_std.s1p')`

this is a bad practice that may cause you some trouble later eventually. Use another variable name, like `open_` or `my_open`. 

For your problem, It comes from the path you used as :

`rf.Network('D:\PyCharm\Python_QT5\VNA\thru_std.s2p')`

which does not work in Python, since Python uses `\` to denote special characters. You should use instead:

`rf.Network('D:\\PyCharm\\Python_QT5\\VNA\\thru_std.s2p')`
or
`rf.Network(r'D:\PyCharm\Python_QT5\VNA\thru_std.s2p')`

There are even better way of setting a path in Python, for example using the [pathlib](https://docs.python.org/3/library/pathlib.html#operators) module. For more info see for example https://stackoverflow.com/questions/2953834/windows-path-in-python

AceYoung938, 2022-12-15:
Thank you for your answer，give you ❤️❤️❤️

closed 2022-12-15 by jhillairet