scikit-rf/scikit-rf #815: OnePort calibration leads to very sparse corrected Network when using different frequency ranges

cactorium, 2022-12-17:
So I'm using the OnePort to do one port calibration of VNA data, and it looks like the output Network tends to have very few points when different frequency range settings are used.

For example, if I were to do something like this:
```
cal = OnePort(measured=[...], ideals=[...]) # networks with frequency range like 1 MHz-2 GHz, 5000 pts
cal.run()
net = skrf.Network("test.s1p") # network with frequency range like 1MHz-10MHz, 1000 pts
net_corrected = cal.apply_cal()
```

`net_corrected` would only have one point at 1 MHz. I've traced this to `OnePort`, where it cascades the correction network with the given network. The issue is that the cascade operation causes the output network to only have the points where there's an exact match between the frequencies collected to do the calibration and the frequencies used in the measurement, which doesn't tend to happen very often if the frequency ranges aren't exactly the same.

Would it be possible to change `OnePort` to interpolate the correction network to match the input frequency range? I understand the point of only using the exact matching points to ensure accuracy in the general case of network cascades, but it seems very limiting to have the calibration only work as expected when the frequency ranges are precisely matched. I can write a pull request if this seems like a reasonable change

Vinc0110, 2022-12-18:
A warning would probably be a good idea in such cases.
Still, why don't you just provide both networks with identical frequency vectors, as expected by the calibration routine?

If you have measured your calibration at different frequencies than your data to be calibrated (not ideal), you can always interpolate either or both of them:
https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.Network.interpolate.html

cactorium, 2022-12-18:
It's for cases where I'm zooming into a particular frequency range to examine the S parameters more closely, or where I'm using a coarser frequency grid to reduce the scanning time.

Yeah, the issue is that there's no way to get the network from `OnePort` to interpolate it to match the input network's frequency range, because it's calculated and cascaded internally. I'm writing a small library to work with a MiniVNA Tiny, and I want the output network to have the frequency range the user requested, so the output network's range preferably should match the input network range. Ideally the calibration network needs to be interpolated into the input network range, but the network is never exposed so I can't perform that interpolation. I could interpolate to match the calibration's frequencies and then interpolate back into the input network's frequency range, but that would add a lot more error than necessary

I'm basically proposing to add an interpolation step at around line 1099 of https://github.com/scikit-rf/scikit-rf/blob/master/skrf/calibration/calibration.py . It shouldn't affect measurements where the frequency ranges match, it should work the same as it does now, but it'd expand where it'd be usable to basically any frequency range

jhillairet, 2023-01-15:
OK, I do understand the need.

> I'm basically proposing to add an interpolation step at around line 1099 of https://github.com/scikit-rf/scikit-rf/blob/master/skrf/calibration/calibration.py . It shouldn't affect measurements [...]

Did you test yourself? Could you make a PR?

cactorium, 2023-02-04:
Sorry for the delay, I finally got around to it

closed 2023-02-09 by cactorium