scikit-rf/scikit-rf #817: pyinstaller bug

zhouzhuming, 2022-12-20:
when packaged to exe using pyinstaller,the skrf cannot cp to the c:\Temp\_MEIxxx, and the exe can't run

jhillairet, 2023-01-06:
Is it really related with scikit-rf? Or with pyinstaller? Could you give more details on the issue?

cafeclimber, 2023-02-14:
I've used pyinstaller on a project with no issues. You have to make sure your spec file is correct. Here's what I had for PyChamber that worked. Of course this might only be relevant in that situation but may prove useful. Given I was able to package an app with skrf successfully, I think this is not a skrf bug and can be closed

```
# -*- mode: python ; coding: utf-8 -*-
from PyInstaller.utils.hooks import collect_data_files
from PyInstaller.utils.hooks import copy_metadata

datas = []
datas += collect_data_files('skrf')
datas += collect_data_files('aquarel')
datas += copy_metadata('pychamber')

block_cipher = None

a = Analysis(
    ['pychamber/launch.py'],
    pathex=['pychamber'],
    binaries=[],
    datas=datas,
    hiddenimports=['pyvisa_py', 'pyqtgraph.console.template_pyqt5'],
    hookspath=[],
    hooksconfig={},
    runtime_hooks=[],
    excludes=[],
    win_no_prefer_redirects=False,
    win_private_assemblies=False,
    cipher=block_cipher,
    noarchive=False,
)
pyz = PYZ(a.pure, a.zipped_data, cipher=block_cipher)

splash = Splash(
    './resources/splash.png',
    binaries=a.binaries,
    datas=a.datas,
    text_pos=(10, 465),
    text_size=12,
    text_color='white'
)

exe = EXE(
    pyz,
    a.scripts,
    splash,
    splash.binaries,
    a.binaries,
    a.zipfiles,
    a.datas,
    [],
    name='PyChamber',
    icon='./resources/logo.png',
    debug=False,
    bootloader_ignore_signals=False,
    strip=False,
    upx=True,
    upx_exclude=[],
    runtime_tmpdir=None,
    console=False,
    disable_windowed_traceback=False,
    argv_emulation=False,
    target_arch=None,
    codesign_identity=None,
    entitlements_file=None,
)
```

jhillairet, 2023-03-20:
@zhouzhuming I'm closing this issue for now. Please reopen it if needed.

closed 2023-03-20 by jhillairet