scikit-rf/scikit-rf #825: Spikes in in single ended fix parameter using MM_NZC from IEEE370

Zwelckovich, 2023-01-17:
Hi,
i have strange results in the single ended s-parameters using the MM_NZC function from the IEEE370 class.
The last value is totally out of range:
[image]
I attached the files. If you want to deembed the thru you have to renumber it first:
`thru.renumber([0,2,1,3], [0, 1, 2, 3])`
[spara.zip](https://github.com/scikit-rf/scikit-rf/files/10434733/spara.zip)

mhuser, 2023-01-17:
Hi @Zwelckovich 
I did not test the data yet, but the last point looks indeed bad.
Your frequency sweep is multiples of the first frequency so there is no extrapolation to blame here.
Had you a chance to compare with the Matlab implementation?
This could narrow the search.

Zwelckovich, 2023-01-17:
Hi @mhuser,
i've looked into the matlab sources but the only script for mixed mode parameter is for the ZC (impedance corrected) method. To recode this i have to do the same thing as you guys did in python:
`#convert to mixed-modes
        mm_2xthru = se_2xthru.copy()
        mm_2xthru.se2gmm(p = 2)
        
        #extract common and differential mode and model fixtures for each
        sdd = subnetwork(mm_2xthru, [0, 1])
        scc = subnetwork(mm_2xthru, [2, 3])
        dm_dd  = IEEEP370_SE_NZC_2xThru(dummy_2xthru = sdd, z0 = self.z0 * 2,
                                use_z_instead_ifft = self.use_z_instead_ifft,
                                verbose = self.verbose,
                                forced_z0_line = self.forced_z0_line_dd)
        dm_cc  = IEEEP370_SE_NZC_2xThru(dummy_2xthru = scc, z0 = self.z0 / 2,
                                use_z_instead_ifft = self.use_z_instead_ifft,
                                verbose = self.verbose,
                                forced_z0_line = self.forced_z0_line_cc)
        
        #convert back to single-ended
        mm_side1 = concat_ports([dm_dd.s_side1, dm_cc.s_side1], port_order = 'first')
        se_side1 = mm_side1.copy()
        se_side1.gmm2se(p = 2)
        mm_side2 = concat_ports([dm_dd.s_side2, dm_cc.s_side2], port_order = 'first')
        se_side2 = mm_side2.copy()
        se_side2.gmm2se(p = 2)`

Zwelckovich, 2023-01-17:
In Matlab the peak is much smaller but also there:
[image]
[image]

mhuser, 2023-01-19:
> i've looked into the matlab sources but the only script for mixed mode parameter is for the ZC (impedance corrected) method. To recode this i have to do the same thing as you guys did in python:

Ooops, I did not notice the method was not available in Matlab !
I wonder if this is related to the algorithm or to the single/mixed modes conversions.

mhuser, 2023-01-19:
The final spike is already present in the common mode.
[image]

```Python
import skrf as rf
import matplotlib.pyplot as plt
import numpy as np
from skrf.calibration import IEEEP370_SE_NZC_2xThru
from skrf.media import MLine
rf.stylely()

se_2xthru = rf.Network('thru.s4p', name = 'se_2xthru')
se_2xthru.renumber([0,2,1,3], [0, 1, 2, 3])
mm_2xthru = se_2xthru.copy()
mm_2xthru.name = 'mm_2xthru'
mm_2xthru.se2gmm(p = 2)
sdd = mm_2xthru.subnetwork([0, 1])
scc = mm_2xthru.subnetwork([2, 3])

dm_dd = IEEEP370_SE_NZC_2xThru(dummy_2xthru = sdd, z0 = 100, name = 'sdd')
dm_cc = IEEEP370_SE_NZC_2xThru(dummy_2xthru = scc, z0 = 25, name = 'scc')

plt.figure()
mm_2xthru.plot_s_db(0, 1)
mm_2xthru.plot_s_db(3, 2)
plt.xlim((7.5e9, 8.2e9))

plt.figure()
dm_dd.s_side1.name = 'mm_fix_dd1'
dm_dd.s_side2.name = 'mm_fix_dd2'
dm_cc.s_side1.name = 'mm_fix_cc1'
dm_cc.s_side2.name = 'mm_fix_cc2'
dm_dd.s_side1.plot_s_db(0, 1)
dm_dd.s_side2.plot_s_db(0, 1)
dm_cc.s_side1.plot_s_db(0, 1)
dm_cc.s_side2.plot_s_db(0, 1)
plt.xlim((7.5e9, 8.2e9))
```

mhuser, 2023-01-19:
The worst is that if the two fixtures are cascaded, the spike disappear.
[image]

mhuser, 2023-01-19:
if the deembedding is called with `verbose = True`, the common mode seems very short. In that case, I am afraid the impedance cannot be properly determined (no uniform section of some samples at center). Maybe the impedance is a mix of the line impedance and the reflections at connectors.
[image]

Vinc0110, 2023-03-18:
How to proceed here? Is it a bug or is it just an issue with the data? Can we close this?

mhuser, 2023-03-22:
Here are the observations:

- Matlab implementation has the spike
- Scikit-rf has the spike
- The impedance section in the middle is too short to fit within the scope of IEEE370 paper

I would say this is not an implementation issue against the Matlab script but the input data are not compatible with the scope of the algorithm (yet I do not understand why the spike appears).

Example with 2xthru that has too short impedance section at middle point:
```Python
import skrf as rf
import matplotlib.pyplot as plt
import numpy as np
from scipy.constants import  c, pi
from skrf.calibration import IEEEP370_SE_NZC_2xThru
from skrf.media import DefinedGammaZ0
rf.stylely()

f = rf.Frequency(0.01, 8, 800, 'GHz')
beta  = 2 * pi * f.f / c # propagation in air
m = DefinedGammaZ0(frequency = f, z0 = 25, gamma = 1j * beta)

l1 =  m.line(150, 'ps', z0 = 27, embed = True)
l2 =  m.line(20, 'ps', z0 = 45, embed = True)

thru = l1 ** l2 ** l1
thru.name = '2xthru'

dm = IEEEP370_SE_NZC_2xThru(dummy_2xthru = thru, z0 = 25, name = 'FIX-FIX',
                            verbose = True)

plt.figure()
dm.s_side1.name = 'FIX-1'
dm.s_side2.name = 'FIX-2'
fixfix = dm.s_side1 ** dm.s_side1.flipped()
fixfix.name = 'FIX-FIX'

fixfix.plot_s_db(1, 0, marker = 's', markevery = 10, linestyle = 'none')
dm.s_side1.plot_s_db(0, 1)
dm.s_side2.plot_s_db(0, 1)
thru.plot_s_db(1, 0)
plt.xlim((0, 8.2e9))
```
[image]
The middle impedance is wrong (35 ohm instead of 45 ohm) this is because the section is too short with measured resolution and the lentgh of 2xthru is not an integer multiple of 8 GHz period.
[image]
Big spike at the end of FIX-1 and FIX-2, yet reconstruction of the FIX-FIX seems correct.

Now with a 2xthru with longer middle section and length close to multiples of 8 GHz period:
```Python
import skrf as rf
import matplotlib.pyplot as plt
import numpy as np
from scipy.constants import  c, pi
from skrf.calibration import IEEEP370_SE_NZC_2xThru
from skrf.media import DefinedGammaZ0
rf.stylely()

f = rf.Frequency(0.01, 8, 800, 'GHz')
beta  = 2 * pi * f.f / c # propagation in air
m = DefinedGammaZ0(frequency = f, z0 = 25, gamma = 1j * beta)

l1 =  m.line(125, 'ps', z0 = 27, embed = True)
l2 =  m.line(125/2, 'ps', z0 = 45, embed = True)

thru = l1 ** l2 ** l1
thru.name = '2xthru'

dm = IEEEP370_SE_NZC_2xThru(dummy_2xthru = thru, z0 = 25, name = 'FIX-FIX',
                            verbose = True)

plt.figure()
dm.s_side1.name = 'FIX-1'
dm.s_side2.name = 'FIX-2'
fixfix = dm.s_side1 ** dm.s_side1.flipped()
fixfix.name = 'FIX-FIX'

fixfix.plot_s_db(1, 0, marker = 's', markevery = 10, linestyle = 'none')
dm.s_side1.plot_s_db(0, 1)
dm.s_side2.plot_s_db(0, 1)
thru.plot_s_db(1, 0)
plt.xlim((0, 8.2e9))
```
[image]
Proper impedances displayed
[image]
No spike