scikit-rf/scikit-rf #831: LRRM calibration deviations to WinCal

cloidnerux, 2023-01-29:
I tried to evaluate the LRRM algorithm in scikit rf against the WinCal implementation. I used the measured responses of the ISS(FormFactor 138-356, Thru, Short, Open, Load) and measurements of some transmission lines where I had both the raw data and the calibrated results. However, it appears that there is a somewhat static difference in the transmission as apparent in the following image:
[image]

I reevaluated the influence of the ideals multiple times, played around with using just inductance estimation and inductance and capacitance evaluation, and found the later to be better. The ideals gave either completely wrong results or the ones seen in the figure.
Therefore, I suspect there is some issue either in how LRRM is implemented or in how the standards have to be provided.
The ideals are calculated using the given values for the ISS and following routines:

```
import numpy as np
import skrf as skrf
from skrf.calibration import SOLT, TwelveTerm, LRRM
from skrf.media import DefinedGammaZ0
import matplotlib.pyplot as plt
import sys

class ProbeCalibrationMeas:
    
    def __init__(self, folder, method='LRRM', inductanceEstimate=0, singlePorts=False):
        self.folder_path = folder
        self.data = skrf.read_all_networks(self.folder_path)
        print("Read " + str(len(self.data)) + " networks!")
        self.f = self.data[list(self.data.keys())[0]].f
        self.open_standards = [key for key, value in self.data.items() if 'open' in key.lower()]
        self.short_standards = [key for key, value in self.data.items() if 'short' in key.lower()]
        self.load_standards = [key for key, value in self.data.items() if 'load' in key.lower()]
        self.thru_standards = [key for key, value in self.data.items() if 'thru' in key.lower()]
        
        self.switch_terms = [key for key, value in self.data.items() if 'switch_term' in key.lower()]
        self.ideal_standards = [self.GetThruStandard(self.f, Z0=50, delay=0.5, refLoss=-0.55, refDelay=27, refFreq=40e9),
                                self.GetShortStandard(self.f, delay=0),
                                self.GetOpenStandard(self.f, delay=0),
                                self.GetLoadStandard(self.f, R=50, L=inductanceEstimate, delay=0)]
        
        if singlePorts:
            self.measured_standards = [self.data[self.thru_standards[0]],
                                       skrf.two_port_reflect(self.data[self.short_standards[0]], self.data[self.short_standards[1]]),
                                       skrf.two_port_reflect(self.data[self.open_standards[0]], self.data[self.open_standards[1]]),
                                       skrf.two_port_reflect(self.data[self.load_standards[0]], self.data[self.load_standards[1]])]
            
        else:
            self.measured_standards = [self.data[self.thru_standards[0]],
                                       self.data[self.short_standards[0]],
                                       self.data[self.open_standards[0]],
                                       self.data[self.load_standards[0]]]
        if method == 'LRRM':
            if(len(self.switch_terms) == 0):
                self.cal = LRRM(ideals = self.ideal_standards, measured = self.measured_standards, match_fit = 'lc',)
            else:
                st = self.data[self.switch_terms[0]]
                self.cal = LRRM(ideals = self.ideal_standards, measured = self.measured_standards, 
                                match_fit = 'lc',
                                switch_terms = [st.s21, st.s12])#is the order correct?
            self.cal.run()
            print('Inductance ' +str( np.mean(self.cal.solved_l) * 1e12) + 'pH')
        elif method == 'SOLT':
            if(len(self.switch_terms) == 0):
                self.cal = SOLT(ideals = self.ideal_standards, measured = self.measured_standards)
            else:
                st = self.data[self.switch_terms[0]]
                self.cal = SOLT(ideals = self.ideal_standards, measured = self.measured_standards, 
                                switch_terms = [st.s21, st.s12])
            self.cal.run()
        elif method == 'TwelveTerm':
            if(len(self.switch_terms) == 0):
                self.cal = TwelveTerm(ideals = self.ideal_standards, measured = self.measured_standards, n_thrus = 1, trans_thres=-10)
            else:
                st = self.data[self.switch_terms[0]]
                self.cal = TwelveTerm(ideals = self.ideal_standards, measured = self.measured_standards, 
                                      n_thrus = 1, trans_thres=-10, switch_terms = [st.s21, st.s12])
            self.cal.run()
[... 119 lines cut ...]
    
    #f - frequency vector in Hz
    #Z0 - reference impedance
    #delay - the delay in ps
    #refLoss - the reference loss in dB
    #refDelay  - the reference delay in ps
    #refFreq- the reference frequency in H
    #refLoss=-0.55, refDelay=27, refFreq=40e9
    def GetThruStandard(self, f, Z0=50, ZL=50, delay=0.5, refLoss=-0.55, refDelay=27, refFreq=40e9):
        if(refLoss > 0):
            print('Reference Loss is positive! should be negative or zero!')
        omega = 2*np.pi*f
        refLossMag = 10**(refLoss/20)
        gamma_l = refLossMag*(delay/refDelay)*np.sqrt(f/refFreq)+1j*(omega*delay*1e-12+refLossMag*(delay/refDelay)*np.sqrt(f/refFreq))
        s = np.zeros((f.shape[0], 2, 2), dtype=np.complex128)
        x = np.exp(-1*gamma_l)
        GAMMA = (ZL - Z0) / (ZL + Z0)
        s[:,1,0] = (1-GAMMA**2)*x/(1-x**2*GAMMA**2)
        s[:,0,1] = (1-GAMMA**2)*x/(1-x**2*GAMMA**2)
        s[:,0,0] = (1-x**2)*GAMMA/(1-x**2*GAMMA**2)
        s[:,1,1] = (1-x**2)*GAMMA/(1-x**2*GAMMA**2)
        # s[:,0,0] = 1e-15
        # s[:,1,1] = 1e-15
        return skrf.Network(frequency=skrf.Frequency.from_f(f, unit='hz'), s=s, name='ideal/thru', f_unit='Hz')
```

Ttl, 2023-01-30:
In your ideal thru standard generation code the functionality of `refLoss` doesn't match what is written in the comments. Generating thru with `delay = refDelay` does not result in `refLoss` at `refFreq`. Specifying a wrong thru loss should result in systematic error in transmission S-parameters just as you are seeing. Below is a thru generation code that works according to the comments. Does using it solve the issue?

```python
    #f - frequency vector in Hz
    #Z0 - reference impedance
    #delay - the delay in ps
    #refLoss - the reference loss in dB
    #refDelay  - the reference delay in ps
    #refFreq- the reference frequency in Hz
    #refLoss=-0.55, refDelay=27, refFreq=40e9
    def GetThruStandard(self, f, Z0=50, ZL=50, delay=0.5, refLoss=-0.55, refDelay=27, refFreq=40e9):
        if(refLoss > 0):
            print('Reference Loss is positive! should be negative or zero!')
        omega = 2*np.pi*f
        refLossMag = np.log(10**(refLoss/20))
        alpha_l = -refLossMag*(delay/refDelay)*np.sqrt(f/refFreq)
        gamma_l = alpha_l + 1j*(omega*delay*1e-12 + alpha_l)
        s = np.zeros((f.shape[0], 2, 2), dtype=np.complex128)
        x = np.exp(-1*gamma_l)
        GAMMA = (ZL - Z0) / (ZL + Z0)
        s[:,1,0] = (1-GAMMA**2)*x/(1-x**2*GAMMA**2)
        s[:,0,1] = (1-GAMMA**2)*x/(1-x**2*GAMMA**2)
        s[:,0,0] = (1-x**2)*GAMMA/(1-x**2*GAMMA**2)
        s[:,1,1] = (1-x**2)*GAMMA/(1-x**2*GAMMA**2)
        return skrf.Network(frequency=skrf.Frequency.from_f(f, unit='hz'), s=s, name='ideal/thru', f_unit='Hz')
```

cloidnerux, 2023-01-31:
Using your code for the thru yields in even higher deviation:
[image]
Why do you use np.log in calculating the linear loss magnitude?
`refLossMag = np.log(10**(refLoss/20))`

Using a negative sign with proper loss magnitude results in gain for the transmission line
`alpha_l = -refLossMag*(delay/refDelay)*np.sqrt(f/refFreq)`

Ttl, 2023-01-31:
The result of log is negative and minus sign is needed so that the line is lossy. Log is needed so that the loss is as specified after exp. In your original function due to exp the loss of the thru is not `refLoss` dB if `delay = refDelay` at `refFreq`. So either the calculation is incorrect or the thru loss shouldn't be `refLoss` dB if `delay = refDelay` at `refFreq` as I understood from the comments.

WinCal LRRM is not to my knowledge documented and skrf LRRM implementation is not claiming to be identical to WinCal LRRM. The paper skrf implementation is based is: https://www.mdpi.com/2076-3417/7/9/866. There are few implementation depended details in LRRM such as the load model (inductor, capacitor, both or something more complex). Fitting of the load model can also be done differently especially with more complex load models and different implementations can give different results if they don't fit the load values the same way.

skrf LRRM implementation is tested to give correct result with random inputs. Any mismatch with WinCal LRRM results from either different ideal definitions or different implementation details of the load model fitting. In any case I would first verify that the thru loss is specified correctly as the thru S-parameters need to be known exactly and that the load model is the same as what WinCal uses.

Ttl, 2023-01-31:
I guess https://ieeexplore.ieee.org/document/4734364 is the WinCal LRRM implementation. Or at least was in 2006, not sure if it has changed since then. `match_fit = 'l'` should in that case be the correct choice. The math in the skrf LRRM is different but assumptions seem the same. One difference I can see is that the paper says that inductance is chosen that achieves the minimum sum of ideal open magnitude minus calculated open magnitude, skrf implementation minimizes square of error instead. I'm not sure if this results in any significant difference.

Ttl, 2023-02-01:
Nice. However, open and short don't look very good and I don't think the calibration is very accurate. LRRM calibration assumes that open is lossless. Any errors in ideal definitions or measurements are seen in the solved open. If everything would be working perfectly the solved open should also be lossless which is not the case here.

`match_fit = 'lc'` is more sensitive to measurement errors and incorrect thru ideal definition. It also assumes that open is ideal capacitor. Due to two degrees of freedom in match fitting the result is likely worse if the assumptions don't hold and there is a reason why it's not the default option.